circuitikz/circuitikz #405: Wrong text anchor in MOSFET with body diode

crexyer, 2020-05-21:
In MOSFET component, I want to move the name of diode up. However, when I use text anchor (`s2.text`) to create a label manually, the label is wrong placed. Is there a way to fix it?

My CircuiTikZ version is 1.1.2.

MWE:

```tex
\documentclass{standalone}
\usepackage{circuitikz}
\begin{document}
\begin{circuitikz}
\draw (0,0) node[nigfete, bodydiode] (s1) {$\mathrm{D}_1$}
  (s1.G) node [above] {$\mathrm{S}_1$};
\draw (2.5,0) node[nigfete, bodydiode] (s2) {}
  (s2.text) node [above right] {$\mathrm{D}_2$}
  (s2.G) node [above] {$\mathrm{S}_2$};
\end{circuitikz}
\end{document}
```

[image]

The source code starting from line 1476 in `pgfcirctripoles.tex` is posted here:

```tex
\anchor{text}{
    \northeast
    \pgfmathsetlength{\pgf@circ@scaled@Rlen}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}
    \ifpgf@circuit@transisors@fixlabels
        \ifpgf@circuit@fet@bodydiode
            % try to put the text to the right of the flyback diode
            \pgfmathsetlength{\pgf@circ@res@other}{(
                \ctikzvalof{tripoles/#1/bodydiode distance}*
                \ctikzvalof{tripoles/#1/width} +
                \ctikzvalof{tripoles/#1/bodydiode scale}*
                \ctikzvalof{bipoles/diode/height}/2
                )*\pgf@circ@scaled@Rlen}
            \advance\pgf@x by \pgf@circ@res@other
        \fi
        % add a bit of space to avoid central (substrate) terminal if drawn
        \advance\pgf@x by 0.05\pgf@circ@scaled@Rlen\relax
        \pgf@y=\dimexpr.5\dp\pgfnodeparttextbox-.5\ht\pgfnodeparttextbox\relax
    \else
        \pgf@y=.7\pgf@y
        \pgf@x= \pgf@circ@scaled@Rlen
        \pgf@x=0.1\pgf@x
    \fi
}
```

Rmano, 2020-05-21:
I think you got confused about the `text` anchor. This is a magical anchor that is used internally to place the text given in the `{}` of a node. So iif a node hs no text, it's not well defined. Maybe you want this? (see page 77 in the manual or around there).

 ```latex
\documentclass{standalone}
\usepackage{circuitikz}
\begin{document}
\begin{circuitikz}
\draw (0,0) node[nigfete, bodydiode] (s1) {$\mathrm{D}_1$}
  (s1.G) node [above] {$\mathrm{S}_1$};
\draw (2.5,0) node[nigfete, bodydiode] (s2) {}
  (s2.body C out) node [right] {$\mathrm{D}_2$}
  (s2.G) node [above] {$\mathrm{S}_2$};
\end{circuitikz}
\end{document}
```

[image]

crexyer, 2020-05-22:
Thanks for your explanation. I added text to the `nigfete` node along with an additional
node using text anchor. As you said, text anchor should be well defined, but they don't coincide.

```tex
\documentclass{standalone}
\usepackage{circuitikz}
\begin{document}
\begin{circuitikz}
\draw (0,0) node[nigfete, bodydiode] (s1) {$\mathrm{D}_1$}
  (s1.G) node [above] {$\mathrm{S}_1$};
\draw (2.5,0) node[nigfete, bodydiode] (s2) {$\mathrm{D}_2$}
  (s2.text) node [right, red, inner sep=0pt, outer sep=0pt] {$\mathrm{D}_2$}
  (s2.G) node [above] {$\mathrm{S}_2$};
\end{circuitikz}
\hspace*{15pt}
\end{document}
```

[image]

However, if I comment out `\ifpgf@circuit@fet@bodydiode` in the anchor definition, it works well. Can you explain why?

```tex
\anchor{text}{
    \northeast
    \pgfmathsetlength{\pgf@circ@scaled@Rlen}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}
    \ifpgf@circuit@transisors@fixlabels
        % \ifpgf@circuit@fet@bodydiode % <-- comment out
            % try to put the text to the right of the flyback diode
            \pgfmathsetlength{\pgf@circ@res@other}{(
                \ctikzvalof{tripoles/#1/bodydiode distance}*
                \ctikzvalof{tripoles/#1/width} +
                \ctikzvalof{tripoles/#1/bodydiode scale}*
                \ctikzvalof{bipoles/diode/height}/2
                )*\pgf@circ@scaled@Rlen}
            \advance\pgf@x by \pgf@circ@res@other
        % \fi % <-- comment out
        % add a bit of space to avoid central (substrate) terminal if drawn
        \advance\pgf@x by 0.05\pgf@circ@scaled@Rlen\relax
        \pgf@y=\dimexpr.5\dp\pgfnodeparttextbox-.5\ht\pgfnodeparttextbox\relax
    \else
        \pgf@y=.7\pgf@y
        \pgf@x= \pgf@circ@scaled@Rlen
        \pgf@x=0.1\pgf@x
    \fi
}
```

```tex
\documentclass{standalone}
\usepackage{circuitikz}
\begin{document}
\begin{circuitikz}
\draw (0,0) node[nigfete, bodydiode] (s1) {$\mathrm{D}_1$}
  (s1.G) node [above] {$\mathrm{S}_1$};
\draw (2.5,0) node[nigfete, bodydiode] (s2) {$\mathrm{D}_2$}
  (s2.text) node [right, red, inner sep=0pt, outer sep=0pt] {$\mathrm{D}_2$}
  (s2.G) node [above] {$\mathrm{S}_2$};
\end{circuitikz}
\end{document}
```

[image]

Rmano, 2020-05-22:
I can look at it, but the matter is that you are not supposed to use `text` anchor *outside* the shape definition. When tick draw the node, it puts the text in a box so that you can measure it, and then compute the `text` anchor to know where the box is to be placed. After that, `text` is fair game. It's not a saved anchor, so the placement of the diode will compute things again and anything can happen.

Rmano, 2020-05-22:
And in your case, you have the red label because tikz knows that the MOS has a body diode only *into* the code of the shape. When you call again the `text` anchor you are outside of the shape group, so the `bodydiode` flag has got back to the default vale, and the text label is calculated with the wrong flag. That also explains why, if you remove the if you got the same...

Notice that if you add another MOS after that the text anchor could be off in a lot of funny ways. That anchor is private of the shape, you can't use it.

crexyer, 2020-05-22:
You are right. The anchor position is uncertain when used outside the shape. I misunderstood the difference between `\anchor` and `\savedanchor`. Thanks!

closed 2020-05-22 by crexyer