circuitikz/circuitikz #410: straightvoltages and arrow length

ArchangeGabriel, 2020-06-05:
Hi there,

I can’t find a way to change arrow length of voltages when using `straightvoltages`. They are often too short, especially on sources:

```
\documentclass{standalone}
\usepackage[straightvoltages,european]{circuitikz}
\begin{document}
\begin{circuitikz}
    \draw (0,0) to [european voltage source,i=$I$,v=${U=E}$] (3,0);
\end{circuitikz}
\end{document}
```
[test.pdf](https://github.com/circuitikz/circuitikz/files/4737535/test.pdf)

I’ve tried to fiddle with `voltage/distance from node`, but to no avail.

Rmano, 2020-06-05:
Unfortunately, it is fixed. I'll look into it. You can fake something with

```latex
\documentclass{standalone}
\usepackage[EFvoltages, european, straightvoltages]{circuitikz}
\begin{document}
\begin{circuitikz}
    \def\eq{=}
    \draw (0,0) to [european voltage source,i=$I$,] (3,0);
    \draw (0,.5) to[open, v^=$U\eq E$, voltage=straight] ++(3,0);
\end{circuitikz}
\end{document}
```
[image]

The need of `voltage=straight` in the `open` bipole is a bug :-(

Rmano, 2020-06-06:
Ah, no, it's not a bug. The problem is that `european` and `straightvoltages` option conflicts, and `european` wins.

ArchangeGabriel, 2020-06-06:
Thanks for the workaround. Regarding priority of `european` and `straightvoltages`, note that I had them in reverse order. ;)

Rmano, 2020-06-06:
The order of options when loading this package is not really significant (it is not respected). I do not want to change it for backward compatibility. 
This is one reason for which I prefer using styling (`\ctikzset{voltage=straight}`) for this kind of options; it's much more controllable and stable across versions. 
BTW, both  `european` and  `straight` have straight arrows for voltage in case of generators. I do not know why, but it is like this since, well, forever. I still feel that they are really cramped, and I'll try to adjust them a bit in one of the new versions.

Rmano, 2020-06-06:
> especially since there is no shortcut for `european voltage source`

Notice that you can define your shortcuts easily using Ti*k*Z styles:

```latex
 \documentclass{standalone}
\usepackage[EFvoltages, straightvoltages]{circuitikz}
\begin{document}
\begin{circuitikz}
    \def\eq{=}
    \tikzset{EVS/.style={european voltage source}}
    \tikzset{EVSbig/.style={european voltage source, sources/scale=1.5}}

    \draw (0,2) to [EVS,i=$I$,v=$U\eq E$] ++(3,0);
    \draw (0,0) to [EVSbig,i=$I$,] ++(3,0);
    \draw (0,.8) to[open, v^=$U\eq E$] ++(3,0);
\end{circuitikz}
\end{document}
```

Rmano, 2020-06-08:
Anyway ... the main problem you have is that the voltage generators have (for historical reasons) a different voltage label treatment than the other bipoles (which involves switching the side of where the label is --- don't ask). So you can simply ask to treat the component as a standard bipole:

```latex\documentclass{standalone}
\usepackage[EFvoltages, straightvoltages]{circuitikz}
\begin{document}
\begin{circuitikz}
    \def\eq{=}
    \draw (0,0) to [european voltage source, bipole/is voltage=false, i=$I$, v=$U\eq E$] (3,0);
\end{circuitikz}
\end{document}
```

[image]

...and you can put it into a style as shown above.

Rmano, 2020-06-16:
There was an error in the order of options. I changed it in commit https://github.com/circuitikz/circuitikz/pull/411/commits/7ffab765853eb1bed8c4c0273da8a5cca18331fe

ArchangeGabriel, 2020-06-16:
The `bipole/is voltage=false` trick solves the length, but it reverses the arrow direction too (at least with RPvoltages)… So that’s not good.

Good point for fixing the options order though. ;)

Rmano, 2020-06-16:
> The `bipole/is voltage=false` trick solves the length, but it reverses the arrow direction too (at least with RPvoltages)… So that’s not good.

Yes, but it's expected because now the bipole is treated as a passive element....

ArchangeGabriel, 2020-06-18:
Indeed, I’m just saying this is not a proper workaround. ;)

Rmano, 2020-06-18:
> Indeed, I’m just saying this is not a proper workaround. ;)

Well, you can just use `v<` then... ;-) . 

If I could go back on time, I will remove all "direction" versions, use a fixed default direction for voltages, currents, and flows (arrows in the same direction of path, independent on the device, full stop) and then let adjust it with `<` and `>`. 

Too late...

ArchangeGabriel, 2020-06-19:
OK, so using `v^>=` I finally got what I wanted. But I don’t understand in what way your PR solves the issue…

Rmano, 2020-06-20:
No, it really doesn't close the first issue, sorry --- reopened. 

The matter is that there are workarounds that are easily applied, and the time to implement an option to make arrows longer I think it's better applied to other things... for now. I will add  the workarounds to the manual, and in the meanwhile, I'll let this issue open to see if somebody shows up with a PR...

Rmano, 2020-06-30:
Now it's in the manual:

[image]

[image]
[image]

closed 2020-06-30 by Rmano