circuitikz/circuitikz #412: Tail-less components

matthuszagh, 2020-06-11:
I frequently find myself using `tlground` instead of `ground`. This is useful, for instance, when you want a resistor to appear halfway between a node and ground:

```latex
\begin{circuitikz}
\draw (0,0) to[R,*-] ++(0,-2) node[tlground]{};
\end{circuitikz}
```
gives

[image]

versus

```latex
\begin{circuitikz}
\draw (0,0) to[R,*-] ++(0,-2) node[ground]{};
\end{circuitikz}
```

which gives

[image]

I think tail-less varieties of other components would be useful too. An obvious example would be `vcc`. There is of course a workaround for this. I could do

```latex
\begin{circuitikz}
\draw (0,0) to[R,*-] ++(0,2) node[vcc,anchor=north]{};
\end{circuitikz}
```

But this opens up the possibility for issues if I want the connecting line to be smaller than the tail size. For instance, a bit of an exaggerated example is

```latex
\begin{circuitikz}
\draw (0,0) to[R,*-] ++(0,1) node[vcc,anchor=north]{};
\end{circuitikz}
```

[image]

The other components I frequently find myself wishing for tail-less varieties is transistors. These also seem to support workarounds, but they're a bit more work and harder to remember (i.e. `inner up`, `inner down`, `nobase`).

There may also be other components this applies to that I haven't though of.

Finally,  a minor point, but when I'm adding path-style components I find it easier to think about the distance between two nodes rather than the distance between two nodes minus twice the tail distance.

Is it possible to add tail-less varieties for more components? I.e. something like `node[tlnpn]` or `node[npn,tl]`, where the terminals then refer to the adjusted positions? Between the 2 syntaxes the 2nd makes a bit more sense to me, but the 1st has the advantage of being stylistically like `tlground`.

---

All snippets use

```latex
\usepackage[american]{circuitikz}
\ctikzset{resistors/scale=0.6}
```

Rmano, 2020-06-11:
Yes, I see this is a useful thing to have: if you notice, almost all the components I have added in the last four years have a kind-of-way to remove "tails" (or leads, or pins, whatever they are called). For example, in the  "new" multi-terminal BJT, you have: 

[image]

and similar options in chips, mux-demuxes, and new logic gates.

The problem is that this will be quite a huge job. You have to go through all shape definitions, and change them to remove tails, as well as adjusting the geographical anchors (otherwise a lot of things will start to fail, from the label/voltages positioning to the `fit` library).

matthuszagh, 2020-06-12:
That makes sense, thanks for the response.

Rmano, 2020-06-12:
An easier way --- and one that can be done in element by element --- is the following one. In the generic code we already have: 
```latex
\tikzset{all leads/.code={\pgfcirc@draw@output@leadstrue\pgfcirc@draw@input@leadstrue}}
\tikzset{no leads/.code={\pgfcirc@draw@output@leadsfalse\pgfcirc@draw@input@leadsfalse}}
```
that is used by the logic ports, but we can extend it. For example, let consider the following fake shape:
[image]
without changing any of the existing anchors, we can add the following anchors:
[image]
and the using the if above to suppress the drawing of the "tails":
```latex
\ifpgfcirc@draw@input@leads
   [whatever]
\fi
```

That will have the problem that you will need to use the "node" form of the shape to have it tailless (`to[...` automatically links to `left` and `right` anchors) and the `fit` will overestimate the size of the component when there are no tails. But it's easy to do, I think. So for example your `vcc` could became `node[vcc, no leads, anchor=bsouth]` or something like that.

matthuszagh, 2020-06-17:
Unfortunately, I'm having a little difficulty with this. I'm trying:

```latex
\tikzset{all leads/.code={\pgfcirc@draw@output@leadstrue\pgfcirc@draw@input@leadstrue}}
\tikzset{no
  leads/.code={\pgfcirc@draw@output@leadsfalse\pgfcirc@draw@input@leadsfalse}}
\begin{document}
\begin{circuitikz}
\draw (0,0) to[R,*-] ++(0,1.2) node[vcc, no leads, anchor=bsouth]{};
\end{circuitikz}
\end{document}
```

But I'm getting an undefined control sequence.

What is the if control structure? And what is the `[whatever]`? Thanks.

Rmano, 2020-06-17:
>  I'm having a little difficulty with this. I'm trying:

Uh, no, I'm sorry, I explained myself badly. What I mean is that `all leads` and `no leads` are already defined in current `circuitikz`, and that somebody could change the definition of `vcc` (and similar) to make use of these `\if...` and adding an appropriate anchor. 
The idea is going and editing `vcc` in `pgfcircmonopoles.tex` (or wherever)... I have no time now, but I'll try to do tha change to show how it could be done. But basically that was a "hook" to see if somebody started coding... ;-)

matthuszagh, 2020-06-26:
Ah I see, thanks for the clarification.

Rmano, 2020-06-26:
Here is an example: I marked the change to `vcc` code (I just copied the code in `pgfcircmonopoles.tex`). The problem is that the tailless thing works only for the legacy arrows, because the other arrows need a stem to detect the direction, and I have to think about how to do it. 
Notice that the tailless anchor is just at the top of the arrow --- you can change it if you want at least a minimum wire into it. 

```latex
\documentclass[12pt]{article}
\usepackage{tikz}
\usepackage[RPvoltages, american,siunitx]{circuitikz}
\makeatletter
%
% override original vcc here
%
\pgfdeclareshape{vcc}{
    \savedmacro{\ctikzclass}{\edef\ctikzclass{power supplies}}  % class of these components
    \saveddimen{\scaledRlen}{\pgfmathsetlength{\pgf@x}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}}
    \saveddimen{\scaledwidth}{% thanks to @Schrödinger's cat on https://tex.stackexchange.com/a/506249/38080
        \pgfgettransformentries{\tmpa}{\tmpb}{\tmpc}{\tmpd}{\tmp}{\tmp}%
        \pgfmathsetmacro{\gscale}{sqrt(abs(\tmpa*\tmpd-\tmpb*\tmpc))}% abs should not be needed
        \pgfmathsetlength{\pgf@x}{(\ctikzvalof{\ctikzclass/scale}*\gscale*\ctikzvalof{monopoles/vcc/width})*\pgf@circ@Rlen}%
    }
    \savedanchor{\northeast}{%
        \pgfmathsetlength{\pgf@circ@scaled@Rlen}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}
        \pgf@circ@res@step=\ctikzvalof{monopoles/vcc/width}\pgf@circ@scaled@Rlen
        \pgf@x=.5\pgf@circ@res@step
        \pgf@y=3\pgf@x%
    }
    \anchor{north}{\northeast\pgf@x=0cm\relax}
    \anchor{east}{\northeast\pgf@y=0.5\pgf@y\relax}
    \anchor{south}{\pgfpointorigin}
    \anchor{west}{\northeast\pgf@y=0.5\pgf@y\pgf@x=-\pgf@x}
    \anchor{north east}{\northeast}
    \anchor{north west}{\northeast\pgf@x=-\pgf@x}
    \anchor{south east}{\northeast\pgf@y=0pt\relax}
    \anchor{south west}{\northeast\pgf@y=0pt\pgf@x=-\pgf@x}
    \anchor{center}{\pgfpointorigin}
    \anchor{left}{\pgfpointorigin}
    \anchor{right}{\pgfpointorigin}
    %%% ADDED ANCHOR
    \anchor{tailless}{\northeast\pgf@x=0cm\relax} % this is the same as north
    \anchor{text}{
        \pgfmathsetlength{\pgf@circ@scaled@Rlen}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}
        \pgf@circ@res@step=\ctikzvalof{monopoles/vcc/width}\pgf@circ@scaled@Rlen
        \pgfpathmoveto{\pgfpoint{-.5\wd\pgfnodeparttextbox}{2\pgf@circ@res@step+2\ht\pgfnodeparttextbox}}
        \pgfpathmoveto{\pgfpoint{.5\wd\pgfnodeparttextbox}{2\pgf@circ@res@step+2\ht\pgfnodeparttextbox}}
        \pgf@x=0pt
        \pgf@y=2\pgf@circ@res@step
        \advance \pgf@y by 0.5\ht\pgfnodeparttextbox
        \pgf@x=-.5\wd\pgfnodeparttextbox
    }
    \behindforegroundpath{
        \pgf@circ@scaled@Rlen=\scaledRlen
        \pgf@circ@res@step=\ctikzvalof{monopoles/vcc/width}\pgf@circ@scaled@Rlen
        \pgfscope
            \edef\pgf@circ@temp{\ctikzvalof{monopoles/vcc/arrow}}\edef\pgf@temp{legacy}
            \ifx\pgf@temp\pgf@circ@temp
                \pgfstartlinewidth=\pgflinewidth
                \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}

                \pgfpathmoveto{\pgfpoint{-0.5\pgf@circ@res@step}{.8\pgf@circ@res@step}}
                \pgfpathlineto{\pgfpoint{0}{1.5\pgf@circ@res@step}}
                \pgfpathlineto{\pgfpoint{0.5\pgf@circ@res@step}{.8\pgf@circ@res@step}}
                \pgfusepath{draw}

                \pgfsetlinewidth{\pgfstartlinewidth}
            \else
            \pgfsetarrowsend{\pgf@circ@temp}
        \fi
        %%% CHANGED HERE
        \ifx\pgf@temp\pgf@circ@temp % if is legacy
            \ifpgfcirc@draw@input@leads
            % do the standard thing
                \pgfpathmoveto{\pgfpointorigin}
                \pgfpathlineto{\pgfpoint{0pt}{1.5\pgf@circ@res@step}}
            \else
            % do nothing, don't draw
            \fi
        \else
            \pgfpathmoveto{\pgfpointorigin}
            \pgfpathlineto{\pgfpoint{0pt}{1.5\pgf@circ@res@step}}
        \fi
        \pgfusepath{draw}
    \endpgfscope
    }
}

\makeatother
\begin{document}
\begin{circuitikz}

    \draw (0,0) node[tlground]{} to [R] ++(0,2) node[vcc]{};
    \draw (2,0) node[tlground]{} to [R] ++(0,2) node[vcc, no leads, anchor=tailless]{};

\end{circuitikz}
\end{document}
```

Result: 

[image]

Rmano, 2021-01-06:
Closing this issue due to inactivity; please feel free to reopen if needed.

Rmano, 2021-01-06:
A

closed 2021-01-06 by Rmano