circuitikz/circuitikz #426: Component both node symbol and bipole

dl1chb, 2020-07-20:
Hi,

I am about to create some block-symbols for optical parts. One of them is a grating coupler. A grating coupler is uses as an "optical pad" to couple in/out light from a light source. So in this case I have two scenarios:

1st: I want to draw a block diagram having the view on the chip only. In this case it acts as a one pole and a node like symbol would suite the needs
2nd: in addition I want to have a laser source coupling light into it, so in this case a bipole would suite the needs

Is it possible (or maybe not wanted) to create a symbol as both kinds simultaniously?

BR

Beckus

Rmano, 2020-07-20:
Yes, it is possible. An example is the `not port`, which has the node version and the inline one. For the inline node to work correctly, you have to 
- design the node so that its normal orientation is the horizontal one, and
- add the `left` and `right` anchors
Once you have the main it's simple to generate the inline one (just use a different name). 

Although the `ieee` ports are a better example, you can see as a very simple one the traditional "not" port starting here: 

https://github.com/circuitikz/circuitikz/blob/5687d3b7f967ecd1c0cf328ddac45ce93d4a97ec/tex/pgfcirctripoles.tex#L775-L777

and then the "bipolar transformation" is here:

https://github.com/circuitikz/circuitikz/blob/5687d3b7f967ecd1c0cf328ddac45ce93d4a97ec/tex/pgfcircpath.tex#L947-L956

Notice that this is a bit complicated by the fact that the `node` style has lead wires by default. I will gladly help if you have any doubts or problems!

Rmano, 2020-07-21:
@dl1chb  Given that you are working around the same zone of code ;-), please consider that I would like to merge https://github.com/circuitikz/circuitikz/pull/428  --- I noticed the problem reviewing your code (thanks!)

dl1chb, 2020-07-21:
@Rmano thanks for telling me. I decided to try it here first, because it seems the node only component is sufficient. I will then wait for #428 until it is finished, update my fork and create it to be merge into that.

Rmano, 2020-07-21:
Ok, nice. Now that I think about it --- to work correctly as a bipole, the node shape must have border anchors, even if approximate. The bipoles normally use the `width` x `height` rectangle, but it's overridden in some case.

dl1chb, 2020-07-21:
Okay, so for the grating coupler I won't probably need this kind of co-existence between those two options. But you gave a quiet good hint and explanation about how to do it. So what should we do with this issue? It is closed for now for me, but maybe it comes up later (and maybe not only for me).

Rmano, 2020-07-22:
Let it open for now. Maybe it could end in a kind of a small addendum at the end of te manual explaining how to do it...

Rmano, 2020-12-15:
I'll close this, for now, feel free to reopen if needed!

closed 2020-12-15 by Rmano