circuitikz/circuitikz #429: Colors not applying to components

LoganBarnett, 2020-07-26:
When I create a LaTeX preview in `org-mode` (Emacs), I run into an issue where not all of the color information is honored or passed to Circuitikz.

Here's using the first example from the Circuitkz docs:

``` latex
\begin{circuitikz}[]
\draw (0,0) to[isource] (0,3) -- (2,3)
to[R] (2,0) -- (0,0);
\end{circuitikz}
```
And then the output:
[image]

The lines appear to be honored, but these two components do not. If I add `\ctikzset{color/.initial=gray}` to the start of the document it nicely works around my issue:

[image]

I apologize that I'm not running this external of Emacs. I will say I do have simple documents like `\dfrac{1}{2}` rendering just fine:
[image]

This is my color config that I'm sending:
``` emacs-lisp
(setq org-format-latex-options
      '(:foreground default
        :background "#242730"
        :scale 1.0
        :html-foreground "gray"
        :html-background "Transparent"
        :html-scale 1.0
        :matchers ("begin" "$1" "$" "$$" "\\(" "\\[")
        )
      )
```

Here's the version I'm on according to `tlmgr version circuitikz`:
```
package:     circuitikz
category:    Package
shortdesc:   Draw electrical networks with TikZ
longdesc:    The package provides a set of macros for naturally typesetting electrical and (somewhat less naturally, perhaps) electronic networks. It is designed as a tool that is easy to use, with a lean syntax, native to LaTeX, and directly supporting PDF output format. It has therefore been based on the very impressive PGF/TikZ package.
installed:   Yes
revision:    55843
sizes:       doc: 1433k, run: 9245k
relocatable: No
cat-version: 1.2.2
cat-license: lppl gpl
cat-topics:  graphics diagram-circ pgf-tikz
cat-contact-repository: https://github.com/circuitikz/circuitikz
cat-contact-bugs: https://github.com/circuitikz/circuitikz/issues
collection:  collection-pictures
```

As I mentioned above, I have a workaround in hand and it's fine to use. I'm very new to LaTeX, TikZ, and Circuitikz so I think with more time I could probably put together a report sans-Emacs, but I wanted to start with something.

Thanks for all the work on this project! I'm building diagrams right now for a home made rotary tumbler with a cooling fan :)

Rmano, 2020-07-26:
I do not think that `emacs` has anything to do with it ;-)

If you look at the manual of `circuitikz`, section **"Color"**, around page 157, you'll see:

[image]

So, if you want a circuit say, blue, the best thing is using `\begin{circuitikz}[color=blue]` and that should work. Notice however that you sill have problem with transparency with poles (read it in the manual) where you will need to explicitly set the background color if different from white.

Rmano, 2020-07-26:
For example, if you use this simple circuit:
```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\pagecolor{cyan}
\begin{document}
\begin{circuitikz}[
    color=red]
    \draw[black] (1.5,1.2) -- ++(0,-2);
    \draw (0,1) to [R, o-o] ++(3,0);
    \draw (0,0) to [generic, o-o] ++(3,0);
\end{circuitikz}
\end{document}
```

you will be surprised to find this: 

[image]

So you need to change the definition of the poles:

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\pagecolor{cyan}
\begin{document}
\tikzset{ccirc/.style={shape=ocirc, fill=cyan}}
\ctikzset{o-o/.style = {\circuitikzbasekey/bipole/nodes/left=ccirc, \circuitikzbasekey/bipole/nodes/right=ccirc }}
\begin{circuitikz}[
    color=red]
    \draw[black] (1.5,1.2) -- ++(0,-2);
    \draw (0,1) to [R, o-o] ++(3,0);
    \draw (0,0) to [generic, o-o] ++(3,0);
\end{circuitikz}
\end{document}
```

to have what you probably want:

[image]

... and I know, it's probably a bug, but the solution (giving how `circuitikz` is built) is not easy at all.

Rmano, 2020-07-26:
PR #430 will simplify the usage of poles with non-white backgrounds.

Rmano, 2020-08-02:
I think I can close this issue. If you think it's not solved, please feel free to reopen.

closed 2020-08-02 by Rmano

LoganBarnett, 2020-08-02:
Sorry for my late reply. Work has been frantic!

This is great! Thanks for the explanation and the PR to smooth it over for future use. This is fine to close :)