circuitikz/circuitikz #431: Problems with connection wires

jesseopdenbrouw, 2020-07-30:
I have a small document containing a simple circuit, When I run this trough the latest circuitikz, the connection between point (1) and (A) (the 3.3k resistor) is not properly rendered. In older versions of circuitikz I had no problems. Can anyone confirm this behaviour?

```
\documentclass[12pt]{article}
\usepackage[betterproportions,european]{circuitikz}
\usepackage{siunitx}

\tikzset{bookcircuit/.style={line width=1pt,scale=1.25}}

\begin{document}
\begin{center}
\begin{tikzpicture}[bookcircuit]
\draw (0,0) to [V, V=$\SI{12}{\volt}$] ++(0,2)
            to [R, R=$\SI{1.2}{\kilo\ohm}$, -*] ++(2,0) node (1) {} node [above] {$U_x$}
			to [R, R=$\SI{2.7}{\kilo\ohm}$, -*] ++(0,-2)
        (1) to [R, R=$\SI{3.3}{\kilo\ohm}$, -o] ++(2,0) node (a) [right] {A}
            to [open] ++(0,-2) node (b) [right] {B}
            to [short, o-.] (0,0)
;
\end{tikzpicture}
\end{center}

\end{document}
```

Rmano, 2020-07-30:
This is, I think, because the nodes 1 and a have non zero size. Use `coordinate (a)` instead. It is commented in the list of incompatible changes at the start of the manual, I think....

Rmano, 2020-07-30:
See issue #417

jesseopdenbrouw, 2020-07-30:
This seems to fix the issue, but now I have to work out my complete document, which is a bit annoying.

Rmano, 2020-07-30:
You can use `\usepackage{circuitikz-1.1.2} ` if you have several circuits using the old (buggy) behavior, as explained in the manual. Or add `, coordinate` to your node options. 

The old behavior (forcing the use of `.center` anchors for nodes) was simply not acceptable in my opinion: it's a clear Ti*k*Z feature that connections between nodes joins the borders, and not the centers, and overriding it creates much more problems. See the last example I made in
https://github.com/circuitikz/circuitikz/issues/76#issuecomment-652980687 - using the buggy code makes that kind of drawing impossible.

PR #432 (2020-08-02): Add the "nodes make gaps!" FAQ
Reorganize the FAQ section while at it...

Fixes #431
closed 2020-08-02 by Rmano