circuitikz/circuitikz #439: Request: Transmission gate

SJulianS, 2020-09-23:
As far as I can see, there is no support for the [transmission gate](https://en.wikipedia.org/wiki/Transmission_gate) symbol, yet. It would be great if you could add that one in a future release.

While looking for the symbol, I stumbled upon this discussion with a quick (though not very pretty) fix: https://tex.stackexchange.com/questions/526795/circuitikz-transmission-gate-bowtie-symbol

Rmano, 2020-09-23:
Yes, I know --- the symbol has not been added yet. I will try to look at it, I have very little time on my hand now. 

It would be useful to have some hint/reference to standards/etc about the exact form and size, compared to a NOT gate for example.

Rmano, 2020-09-27:
[image]

Rmano, 2020-09-27:
The only doubt I have is if I should let the default `tgate scale` at 1 --- it really looks better with 0.7.

Any opinion?

Rmano, 2020-09-27:
@jason-s, @asinghvi17, @jesseopdenbrouw   any opinion about the default size? I think  I prefer 0.7...

jesseopdenbrouw, 2020-09-28:
Seems a reasonable value to me.

Rmano, 2020-09-28:
> Seems a reasonable value to me.

Yes, the problem is that with the scale equal to one the *area* (visual impact) of the double transmission gate is **4** times the area of the not-gate. With 0.7 (more or less sqrt(2)) it's just twice, and I find it more balanced. Given that I wasn't able to find a standard reference, I think I'll stick with it. Thanks for the comment!

closed 2020-09-30 by Rmano