circuitikz/circuitikz #440: Some additional Block Symbols

myzinsky, 2020-09-25:
It would be nice to have also these symbols:

[image]

[image]

Rmano, 2020-09-26:
Yes, this is a good idea. The `allpass` is missing, yes, and should be added. 

For the other one --- probably it would be better to have a block with two generic labels --- I will look into it as soon as I have a bit of time.

Rmano, 2020-09-27:
In version 1.2.4, you'll be able to do this:

[image]

closed 2020-09-27 by Rmano

myzinsky, 2020-09-27:
Thats so awesome! Thank you very much!