circuitikz/circuitikz #442: Circle Around Transistor

myzinsky, 2020-09-27:
I did a workaround like presented on Stack-Overflow:

```latex
\documentclass[convert = false, border=10pt,]{standalone}
\usepackage[siunitx, straightvoltages]{circuitikz}

\def\CalcC#1{%
\coordinate (base) at (#1.B);
\coordinate (collector) at (#1.C);
\coordinate (emitter) at (#1.E);
\draw (barycentric cs:base=0.5,collector=0.5,emitter=0.5) circle [radius=14pt];
}

\begin{document}
\begin{circuitikz}[european]
    %\tikzstyle{help lines}=[blue!50];
    %\draw[style=help lines] (0,0) grid (5,5);
	\draw (4,0) node[ground](gnd1){};
	\draw (5,0) node[ocirc](Agnd){};
	\draw (0,0) node[ocirc](Egnd){};
	\draw (5,5) node[ocirc, label={$V_{DD}=+12\,\mbox{V}$}](vdd1){};
	\draw (5,3) node[ocirc, label=A](A){};
	\draw (0,2) node[ocirc, label=E](E){};
	\draw (4,2) node[npn](npn1){}; 
	\CalcC{npn1};
	% Lines:
	\draw(Egnd) -- (gnd1);
	\draw(Agnd) -- (gnd1);
	\draw(1,5)  -- (vdd1);
	\draw (npn1.E) to [short,-*] (gnd1.north);
	\draw (npn1.C) -- ( 4,3) -- (A.west);
	\draw (npn1.C) -- ( 4,3) to [R,*-*] (4,5); 
	\draw (npn1.B) -- ( 2,2) to [R,*-*] (2,5); 
	\draw (npn1.B) -- ( 2,2) to [R,*-*] (2,0); 
	\draw (npn1.B) -- ( 2,2) to [C,*- ] (E.east); 
	% Volatages:
	\draw (A.south) to [open, v^=$U_a$] (Agnd.north); 
	\draw (E.south) to [open, v=$U_e$] (Egnd.north); 
\end{circuitikz}
\end{document}
```

[image]

But of course it would be great to have this also natively.

Rmano, 2020-09-27:
This is a lot of work and, for what I know, circles around transistors is a deprecated style since at least the '90.
I would consider a PR, if it's optional and well integrated, but personally I don't like them and I don't think it's worth the effort.

myzinsky, 2020-09-28:
Yes fair enough. I would do a a PR but I dont understand how tikz works under the hood. I'm just a LaTeX user :-)

Rmano, 2020-09-28:
I will let this open, let's see if someone steps in. Anyway, the macro that was suggested in the link above is quite nice to use... 
My other problem is that --- what do you do with the other transisors? FET, MOS, etc... ? Should all of them have optional circles?

dl1chb, 2020-12-04:
This is from IEEE-91-1984 [1].

As far as I know, this circle represents a transistor in a housing, not one in an integrated circuit. Maybe this is obsolete, maybe still valid. Then for a MOSFET the same rule should be applied.

What is the consecutively following document of IEEE-91-1984? What does ist say about this?

[1]: https://ieeexplore.ieee.org/document/27895

Rmano, 2020-12-14:
I am working on it...

Rmano, 2020-12-15:
Please check and comment the PR #470

PR #470 (2020-12-15): Transistor circles
This adds transistor's circles. Fixes #442 

[image]

[image]
closed 2020-12-15 by Rmano