circuitikz/circuitikz #443: Single Port Speaker

myzinsky, 2020-09-27:
[image]

A Single-Port Speaker would be also great!

Rmano, 2020-09-27:
This seems not a standard symbol, or block, and it's easy to do with a `pic` in plain TikZ... Unless you can give me any reference, I am not prone to add random symbols to `circuitikz` , sorry...

Rmano, 2020-09-28:
For example: 

```
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{circuitikz}
    \draw node[vcoshape](V){} -- ++(2,0) node[loudspeakershape, anchor=south, rotate=-90](L){};
    \draw [-Triangle] (V.east) -- (L.south);
\end{circuitikz}
\end{document}
```

[image]

myzinsky, 2020-09-28:
Ah great, I was not aware that I can use the loudspeaker shape! Thanks

closed 2020-09-28 by myzinsky