circuitikz/circuitikz #444: VFO Symbol

myzinsky, 2020-09-27:
[image]

Also this would be nice.

Rmano, 2020-09-27:
I am not sure what symbol is this... The resolution is really low ans I do not know the meaning of VFO.

Can you please give more info?

myzinsky, 2020-09-27:
Oh Sorry for that. 

VFO is Variable Frequency Oscillator. The shape is like the all pass but it has an arrow in the middle pointing into the upper right corner.

Rmano, 2020-09-27:
We have the standard VCO (the round one). Usage of round shapes is normal with generators and mixers... Or am I wrong?

Rmano, 2020-10-01:
Anyway, you can have it easily now:

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{circuitikz}[
    ]
    \node [allpassshape, name=AP]{};
    \draw [-Triangle, shorten >=4pt, shorten <=4pt] ([xshift=4pt]AP.sw) -- ([xshift=-4pt]AP.ne); 
\end{circuitikz}
\end{document}
```

[image]

Rmano, 2020-10-01:
I think that can be closed, what do you think?

closed 2020-10-01 by myzinsky