circuitikz/circuitikz #447: wrong label information in voltage source

judober, 2020-09-30:
The voltage source seems to use the label of other elements if no label is specified.
```
\documentclass[class=report, ngerman,11pt,a4paper,tikz,]{standalone}

\usepackage[european, straightvoltages, americaninductors, betterproportions, nooldvoltagedirection]{circuitikz}

\begin{document}
\begin{tikzpicture}

\draw (90:3) [L, o-, l={\(L_{ab}\)}] to (90:1.5) [V, -*, v={\(v_{12}\)}] to (90:0);

\end{tikzpicture}%
\end{document}

```
[Voltagesource.pdf](https://github.com/circuitikz/circuitikz/files/5306453/Voltagesource.pdf)

Rmano, 2020-10-01:
Confirmed:

[image]

Rmano, 2020-10-01:
As a workaround, you can use for now

```
\draw (90:3) [L, o-, l={\(L_{ab}\)}] to (90:1.5) [V, -*, l=, v={\(v_{12}\)}] to (90:0);
```

At least is not a regression; tested backward to `0.8.3` and it's still there. Awesome that it didn't show up before.

Rmano, 2020-10-01:
Ach! ;-) you used the wrong notation... the element are `to[element, option] (coordinate)`, not `[option] to (coordinate)`; the last one use the plain Ti*k*Z `to` that does not reset the component variables.

```
\documentclass[class=report, 11pt,a4paper,tikz,]{standalone}

\usepackage[european, straightvoltages, americaninductors, betterproportions, nooldvoltagedirection]{circuitikz-0.8.3}

\begin{document}
\begin{tikzpicture}

\draw (90:3) to[L, o-, l={\(L_{ab}\)}] (90:1.5) to[V, -*, v={\(v_{12}\)}] (90:0);

\end{tikzpicture}%
\end{document}
```

[image]

closed 2020-10-01 by Rmano

judober, 2020-10-01:
Allright thanks!