circuitikz/circuitikz #448: blue voltage arrow from manual label wrong rotation

judober, 2020-09-30:
I have two questions regartind the blue voltage from manual page 152:
first: I had to replace the node position to above when using this on a voltage source. Is there a general approach for a version that works with voltage sources as well as other bipoles?
second: The retation seems not to use the "smart" behavior of the element labels (see example). Can this behaviour be used?

```
\documentclass[class=report, ngerman,11pt,a4paper,tikz,]{standalone}
\usepackage[european, straightvoltages, americaninductors, betterproportions, nooldvoltagedirection]{circuitikz}
\begin{document}
\begin{tikzpicture}

\def\myvv#1#2{%
\draw [thin, blue, ->,]
(#1-Vfrom) .. controls (#1-Vcont1) and (#1-Vcont2).. (#1-Vto)
node [pos=0.5, above,	% Note compared to manual: below -> above
rotate=\ctikzgetdirection{#1}] at (#1-Vlab) {#2}; }

\draw (-30:3) [L, o-, l={\(L_{ab}\)}] to (-30:1.5) [V, -*, name=V1, l={\(\)}] to (-30:0);
\myvv{V1}{\(v_{12}\)}

\end{tikzpicture}%
\end{document}
```

[Voltagesource2.pdf](https://github.com/circuitikz/circuitikz/files/5306479/Voltagesource2.pdf)

Rmano, 2020-10-01:
Inline image:

[image]

Rmano, 2020-10-01:
This is quite complex, really. One of the problem is that the "smart" label positioning works only for labels, not for voltages --- I have prepared a simple example here (attached below). Notice that the "normal" voltages labels, called $v_n$, are always straight, even if I use the `bipole/is voltage=false` trick. Smart voltage label would be nice, but it's a lot of work. 

To get the anchor used by the label, you can use `\ctikzgetanchor{}{}` as in the file below; but now, even in this case, the information reported by `\ctikzgetdirection{}` seems wrong (by 180 degrees) in this case. mmm... Will look at it.

```latex
\documentclass[class=report, 11pt,a4paper,tikz,border=5pt]{standalone}
\usepackage[european, straightvoltages, americaninductors, betterproportions, nooldvoltagedirection]{circuitikz}
\begin{document}
\begin{tikzpicture}

\def\myvv#1#2{%
\draw [thin, blue, ->,]
(#1-Vfrom) .. controls (#1-Vcont1) and (#1-Vcont2).. (#1-Vto)
node [pos=0.5, anchor=\ctikzgetanchor{#1}{Vlab},
rotate=\ctikzgetdirection{#1}] at (#1-Vlab) {#2}; }

% Normal voltages
\begin{scope}[yshift=2cm]
\draw (-30:3) to[L, o-, l={\(L_{ab}\)}] (-30:1.5) to [V, -*, name=V1, v=$v_n$,]  (-30:0);
\end{scope}

% extended voltage
\draw (-30:3) to[L, o-, l={\(L_{ab}\)}, v=$v_{ab}$] (-30:1.5) to[V, -*, name=V1, ] (-30:0);
\myvv{V1}{\(v_{12}\)}

% extended voltage, no special source treatment
\begin{scope}[yshift=-2cm]
\draw (-30:3) to[L, o-, l={\(L_{ab}\)}] (-30:1.5) to[V, -*, name=V1, v, bipole/is voltage=false]  (-30:0);
\myvv{V1}{\(v_{12}\)}
\end{scope}

% normal voltage, no special source treatment
\begin{scope}[yshift=-4cm]
\draw (-30:3) to[L, o-, l={\(L_{ab}\)}] (-30:1.5) to[V, -*, name=V1, v=$v_n$, bipole/is voltage=false]  (-30:0);
\end{scope}
\end{tikzpicture}%
\end{document}
```

[image]

Rmano, 2020-10-01:
Anyway, in this case, you have an easy solution: you can use the label-positioning options of Ti*k*Z for placing your voltage label. Maybe you need to leave a manual knop for above / below, but what about this?

[image]

```latex
\documentclass[class=report, 11pt,a4paper,tikz, border=4pt]{standalone}
\usepackage[european, straightvoltages, americaninductors, betterproportions, nooldvoltagedirection]{circuitikz}
\newcommand{\myvv}[3][below]{%
    \draw [thin, blue, ->,]
    (#2-Vfrom) .. controls (#2-Vcont1) and (#2-Vcont2).. (#2-Vto)
    node [pos=0.5, #1, sloped] {#3}; 
}
\begin{document}
\begin{tikzpicture}

\draw (-30:3) to[L, o-, l=$L_{ab}$, v, name=L1] (-30:1.5) to[V, -*, name=V1,] (-30:0);
\myvv{V1}{$v_{12}$}
\myvv[above]{L1}{$v_{ab}$}

\end{tikzpicture}%
\end{document}
```

judober, 2020-10-01:
Yes, this is quite good. Still not perfect in some cases:
```
\documentclass[class=report, 11pt,a4paper,tikz, border=4pt]{standalone}
\usepackage[european, straightvoltages, americaninductors, betterproportions, nooldvoltagedirection]{circuitikz}
\newcommand{\myvv}[3][below]{%
	\draw [thin, blue, -latex,] (#2-Vfrom) -- (#2-Vto) node [pos=0.5, #1, sloped] {#3}; 
}
\begin{document}
	\begin{tikzpicture}
		
		\draw (90:3) to[L, o-, l=$L_{ab}$, v, name=L1] (90:1.5) to[V, -*, name=V1,] (-30:0);
		\myvv[above]{V1}{$v_{12}$}
		\myvv{L1}{$v_{ab}$}
		
	\end{tikzpicture}%
\end{document}
```
[Voltagesource2.pdf](https://github.com/circuitikz/circuitikz/files/5311863/Voltagesource2.pdf)

Additinally:
Is it possible to use the default circuitikz arrow tips?
And can I move the voltage arrow to the other side of the element? E.g. is there something like xx-Vfrom_back?

Rmano, 2020-10-01:
> Yes, this is quite good. Still not perfect in some cases:

Yes; technically it is doing what you ask (voltage label `sloped`) but in this case is not what you want. You need some manual adjustment here, probably with `\rotatebox`

> Is it possible to use the default circuitikz arrow tips?

Unfortunately, they are not really arrowhead. But you can simulate them quite well with a `-Triangle[]`, look here: https://tex.stackexchange.com/a/549354/38080 

> And can I move the voltage arrow to the other side of the element? E.g. is there something like xx-Vfrom_back?

That should work with adding `v_` or `v^` to the options.

```
\documentclass[class=report, 11pt,a4paper,tikz, border=4pt]{standalone}
\usepackage[european, straightvoltages, americaninductors, betterproportions, nooldvoltagedirection]{circuitikz}
\newcommand{\myvv}[3][below]{%
	\draw [thin, blue, -latex,] (#2-Vfrom) -- (#2-Vto) node [pos=0.5, #1, sloped] {#3};
}
\begin{document}
	\begin{tikzpicture}

		\draw (90:3) to[L, o-, l=$L_{ab}$, v, name=L1] (90:1.5) to[V, -*, v_, name=V1,] (-30:0);
                \myvv{V1}{\rotatebox{90}{$v_{12}$}}
                \myvv{L1}{\rotatebox{90}{$v_{ab}$}}

	\end{tikzpicture}%
\end{document}
```

[image]

(PS: you can do a screenshot and paste the image directly in the answer to have it inline...)

judober, 2020-10-02:
Ok thanks a lot. This works for me.

Rmano, 2020-10-02:
Thanks! I will try to tackle the main problem also --- there is something pesky in the code (this is why is marked experimental in the manual!). So I'll leave this open for now.

Rmano, 2020-10-04:
So... I finally think this is not an error. Voltage labels are always drawn straight, and the correct way to mimic (but in blue) the normal positioning is not using `rotate=\ctikzgetdirection{#1}` at all. 

```latex
\def\myvv#1#2{\draw [thin, blue, ->,]
     (#1-Vfrom) .. controls (#1-Vcont1) and (#1-Vcont2).. (#1-Vto)
     node [anchor=\ctikzgetanchor{#1}{Vlab}] at (#1-Vlab) {{#2}}; }
```

`ctikzgetdirection` is the direction of the bipole, which is correctly reported as 150 (degree); the probelm is that it is not using in the voltage text positioning but for determining the anchor.

So it is not really a bug, maybe a lack of feature ;-)

judober, 2020-10-05:
I advanced this a little to get something similar to the smart label behaviour. However, one still has to chose between two versions depending on if v oder v_ is chosen (the voltage source has confusingly the inverse behaviour).
I didn't know the limits for horizontal text in the original implementation so the text is only horizontal for vertical lines.

```
\documentclass[class=report, 11pt,a4paper,tikz, border=4pt]{standalone}
\usepackage[european, straightvoltages, americaninductors, betterproportions, nooldvoltagedirection]{circuitikz}

\newcounter{ang}

\newcommand{\vcross}[3]{
	\draw [thin, blue, -Triangle,] (#2-Vfrom) .. controls (#2-Vcont1) and (#2-Vcont2).. (#2-Vto) node [pos=0.5, #1, sloped] {#3};
}

\newcommand{\vhorz}[2]{
	\draw [thin, blue, -Triangle,] (#1-Vfrom) .. controls (#1-Vcont1) and (#1-Vcont2).. (#1-Vto)
	node [anchor=\ctikzgetanchor{#1}{Vlab}] at (#1-Vlab) {{#2}};
}

\newcommand{\myvv}[2]{
	\setcounter{ang}{\ctikzgetdirection{#1}}
	\ifnum \value{ang} < 90 {
		\vcross{below}{#1}{#2}
	}
	\else {\ifnum \value{ang} = 90 {
		\vhorz{#1}{#2}
	}
	\else {\ifnum \value{ang} < 270 {
		\vcross{above}{#1}{#2}
	}
	\else {\ifnum \value{ang} = 270 {
		\vhorz{#1}{#2}
	}
	\else{
		\vcross{below}{#1}{#2}
	}\fi}\fi}\fi}\fi
}

\newcommand{\myvvf}[2]{	% fliped version
	\setcounter{ang}{\ctikzgetdirection{#1}}
	\ifnum \value{ang} < 90 {
		\vcross{above}{#1}{#2}
	}
	\else {\ifnum \value{ang} = 90 {
		\vhorz{#1}{#2}
	}
	\else {\ifnum \value{ang} < 270 {
		\vcross{below}{#1}{#2}
	}
	\else {\ifnum \value{ang} = 270 {
		\vhorz{#1}{#2}
	}
	\else{
		\vcross{above}{#1}{#2}
	}\fi}\fi}\fi}\fi
}

\begin{document}
	\begin{tikzpicture}
		
		\draw (30:4) to[L, o-, l=$L_{ab}$, v, name=L1] (30:2) to[V, -*, v, name=V1,] (0:0);
		\myvvf{V1}{$v_{12}$}
		\myvv{L1}{$v_{longName}$}
		
	\end{tikzpicture}%
\end{document}
```

Rmano, 2020-10-06:
Hmmm... this can be transformed to an enhancement request to export also label direction (the label position is already there, giving that the label node is accessible).

Rmano, 2020-10-07:
Added in PR #452 :

[image]

judober, 2020-10-15:
So, one question: what is the difference between annotation and voltage label? does it make sense to use an annotation instead of the voltage?

judober, 2020-10-29:
I came around playing with this a bit. However, \ctikzgetrotation doesn't seem to work? The example below errors with `Unknown function V1label` and the same for L1label. I use circuitikz 1.2.5.

```
\documentclass[class=report, 11pt,a4paper,tikz, border=4pt]{standalone}
\usepackage[european, straightvoltages, americaninductors, betterproportions, nooldvoltagedirection]{circuitikz}

\newcommand{\vsmart}[3]{
	\draw [thin, blue, -Triangle,] (#2-Vfrom) .. controls (#2-Vcont1) and (#2-Vcont2).. (#2-Vto) node [pos=0.5, anchor=\ctikzgetanchor{#2}{Vlab}, rotate=\ctikzgetrotation{#2label}] {{#3}};
}

\begin{document}
	\begin{tikzpicture}
		
		\draw (30:4) to[L, o-, l=$L_{ab}$, v, name=L1] (30:2) to[V, -*, v={\(v_{12}\)}, name=V1,] (0:0);
		\vsmart{below}{V1}{$v_{12}$}
		\vsmart{below}{L1}{$v_{longName}$}

	\end{tikzpicture}%
\end{document}
```

Rmano, 2020-10-29:
Will look into it

Rmano, 2020-10-29:
Well, the easy thing is that it's called `\ctikzgetdirection` ;-). But still, the `rotate` key does not like it, I have to investigate.

judober, 2020-10-29:
Well, I tried `\ctikzgetdirection` too but it didn't work. In the manual on page 153 however (in the screenshot obove too) the text says 

> since1.2.5 you can access the label rotation using with the command \ctikzgetrotation{nodename}

and so I thought this is supposed to work...

Rmano, 2020-10-29:
Yes, that is a typo, look at the code. For the more fundamental problem (the erroring out) I have an idea of what's happening, but I have to check and I have no time right now... I hope to solve it asap. It seems an expansion problem in the TikZ parameter list.

Rmano, 2020-10-29:
Ok, found the problem. There is no label defined for `V1`, so no direction macro.

```
documentclass[class=report, 11pt,a4paper,tikz, border=4pt]{standalone}
\usepackage[european, straightvoltages, americaninductors, betterproportions, nooldvoltagedirection]{circuitikz}

\newcommand{\vsmart}[3]{
	\draw [thin, blue, -Triangle,] (#2-Vfrom) .. controls (#2-Vcont1) and (#2-Vcont2).. (#2-Vto) node [pos=0.5, anchor=\ctikzgetanchor{#2}{Vlab}, rotate=\ctikzgetdirection{#2label}] {{#3}};
}

\begin{document}
	\begin{tikzpicture}
		
		\draw (30:4) to[L, o-, l=$L_{ab}$, v, name=L1] (30:2) to[V, -*, v={\(v_{12}\)}, name=V1, l=~] (0:0);
		\vsmart{below}{V1}{$v_{12}$}
		\vsmart{below}{L1}{$v_{longName}$}

	\end{tikzpicture}%
\end{document}
```
[image]

I think it's commented in the manual somewhere...

closed 2020-10-29 by Rmano

judober, 2020-10-30:
Ok, this is as far as I have come with this:
```
\documentclass[class=report, 11pt,a4paper,tikz, border=4pt]{standalone}
\usepackage[european, straightvoltages, americaninductors, betterproportions, nooldvoltagedirection]{circuitikz}[
bipoles/label=~,
]

\usepackage{xifthen}
\usetikzlibrary{math}
	
\newcommand{\vhorz}[2]{
	\draw [thin, blue, -Triangle,] (#1-Vfrom) .. controls (#1-Vcont1) and (#1-Vcont2).. (#1-Vto)	node [anchor=\ctikzgetanchor{#1}{Vlab}] at (#1-Vlab) {{#2}};
}

\newcommand{\dpCcoord}{}	% Center-Coord
\newcommand{\dpVcoord}{}	% Label-Coord

\newcommand{\vcrossauto}[2]{
	\renewcommand{\dpCcoord}{(#1.center)}
	\renewcommand{\dpVcoord}{(#1voltage.center)}
	\tikzmath{
		coordinate \dpC, \dpV;
		\dpC = \dpCcoord;
		\dpV = \dpVcoord;
		if \dpCy < \dpVy then{
			{\draw [thin, blue, -Triangle,] (#1-Vfrom) .. controls (#1-Vcont1) and (#1-Vcont2).. (#1-Vto) node [pos=0.5, above, sloped] {#2};};
		} else {
			{\draw [thin, blue, -Triangle,] (#1-Vfrom) .. controls (#1-Vcont1) and (#1-Vcont2).. (#1-Vto) node [pos=0.5, below, sloped] {#2};};
		};
	}
}

\newcommand{\dprot}{}	% Label-Rotation

\newcommand{\varr}[2]{
	\renewcommand{\dprot}{\ctikzgetdirection{#1label}}
	\ifthenelse{\dprot = 0}{
		\vhorz{#1}{#2}
	}{
		\vcrossauto{#1}{#2}
	}
}

\newcommand{\ihorz}[2]{
	\draw [red, thin, -{Triangle[]}]
	([shift={(\ctikzgetdirection{#1}:-0.06)}] #1-Ipos) --
	([shift={(\ctikzgetdirection{#1}:0.06)}] #1-Ipos)
	node [anchor=\ctikzgetanchor{#1}{Ilab}] at (#1-Ipos) {{#2}};
}

\newcommand{\icross}[3]{
	\draw [red, thin, -{Triangle[]}]
	([shift={(\ctikzgetdirection{#2}:-0.06)}] #2-Ipos) --
	([shift={(\ctikzgetdirection{#2}:0.06)}] #2-Ipos)
	node [pos=0.5, #1, sloped] {#3};
}

\newcommand{\icrossauto}[2]{
	\renewcommand{\dpCcoord}{(#1-Ipos)}
	\renewcommand{\dpVcoord}{(#1current.center)}
	\tikzmath{
		coordinate \dpC, \dpI;
		\dpC = \dpCcoord;
		\dpI = \dpVcoord;
		if \dpCy < \dpIy then{
			{\icross{above}{#1}{#2};};
		} else {
			{\icross{below}{#1}{#2};};
		};
	}
}

\newcommand{\iarr}[2]{
	\renewcommand{\dprot}{\ctikzgetdirection{#1label}}
	\ifthenelse{\dprot = 0}{
		\ihorz{#1}{#2}
	}{
		\icrossauto{#1}{#2}
	}
}

\begin{document}
	\begin{tikzpicture}
		
		\draw (150:4) to[L, o-, l=$L_{ab}$, v=v, name=L1, i=~] (150:2) to[V, -*, v_={\(v_{12}\), l=~, i=~}, name=V1] (0:0);
		\varr{V1}{$v_{12}$}
		\varr{L1}{$v_{longName}$}
		\iarr{V1}{$i$}
		\iarr{L1}{$i$}
		
	\end{tikzpicture}%
\end{document}
```
This already works quite well, I added a red current too.
Two question: can i globally define default values for l,v,i so that I do not have to define them for every bipole (tried with `bipoles/label=~` which had no effect). And: the current direction comes from the dipoles direction which can be wrong. Can I get the direction of the current arrow somehow? (Ifrom and Ito seem to be less useful here as I only want the arrow tip)

Rmano, 2020-11-01:
About your second question, there is no a "default label" for components --- and the anchors can be generated only if we have to typeset a label (this is different for `i`, `v`, `f` where you can just define them and have the anchors for the symbol; but notice that to have the anchors for the voltage *labels*, for example, you need to have the label with a least `v=~` or `v=\null` (`\null` is `\hbox{}`, a void box). 

You can always define a style for the components you use: 

```
\tikzset{
    L+/.style={L, v=\null, i=~, f, l=~, a=~}, % use \null or ~ whatever you prefer
    V+/.style={V, v=~, i=~, f, l=~, a=~},
}
```

and then use them like (in your example):

```
\draw (150:4) to[L+, o-, l=$L_{ab}$, name=L1,] (150:2) to[V+, -*, v_={\(v_{12}\)}, name=V1] (0:0);
```

so that you can always override the direction or the label if you want (like `to[L+, v<, ...] ...`).

For the current direction, no, there is not a pre-defined function, but you can easily (well, after a bit of Ti*k*Z manual digging):
```
\newcommand{\darr}[1]{
        \pgfmathanglebetweenpoints{\pgfpointanchor{#1-Ifrom}{center}}{\pgfpointanchor{#1-Ito}{center}}
        \edef\myangletmp{\pgfmathresult}
        \typeout{\myangletmp}
        \node [flowarrow, rotate=\myangletmp, color=green] at (#1-Ipos){};
}
```

and you can use them like 

```
	\draw (150:4) to[L+, o-, l=$L_{ab}$, name=L1,i>_] (150:2) to[V+, -*, v_={\(v_{12}\)}, name=V1] (0:0);
		\varr{V1}{$v_{12}$}
		\varr{L1}{$v_{longName}$}
		\iarr{V1}{$i$}\darr{V1}
                \iarr{L1}{$i$}\darr{L1}
```

[image]

Rmano, 2020-11-01:
BTW, adding something as `every bipole add=` style could be interesting... will think about it.

Rmano, 2020-11-01:
@judober  if you want to match the arrow size (Not necessary I think, but...) you can find info here: https://tex.stackexchange.com/questions/549347/circuitikz-arrowhead

judober, 2021-07-08:
I updated my code in case some one can use it. Thanks for your input.

```Latex
\documentclass[class=report, 11pt,a4paper,tikz, border=4pt]{standalone}
\usepackage[european, straightvoltages, americaninductors, betterproportions, nooldvoltagedirection]{circuitikz}[
]

\usepackage{xifthen}
\usetikzlibrary{math}

\tikzset{
    L+/.style={L, v=~, i=~, f, l=~, a=~},
    V+/.style={V, v=~, i=~, f, l=~, a=~},
}

\newcommand{\vhorz}[3][blue]{
	\draw [thin, #1, -Triangle,] (#2-Vfrom) .. controls (#2-Vcont1) and (#2-Vcont2).. (#2-Vto)	node [anchor=\ctikzgetanchor{#2}{Vlab}] at (#2-Vlab) {{#3}};
}

\newcommand{\vcrossauto}[3][blue]{
	\tikzmath{
		coordinate \dpC, \dpV;
		\dpC = (#2.center);
		\dpV = (#2voltage.center);
		if \dpCy < \dpVy then{
			{\draw [thin, #1, -Triangle,] (#2-Vfrom) .. controls (#2-Vcont1) and (#2-Vcont2).. (#2-Vto) node [pos=0.5, above, sloped] {#3};};
		} else {
			{\draw [thin, #1, -Triangle,] (#2-Vfrom) .. controls (#2-Vcont1) and (#2-Vcont2).. (#2-Vto) node [pos=0.5, below, sloped] {#3};};
		};
	}
}

\newcommand{\varr}[3][blue]{
	\tikzmath{
		\dprot = \ctikzgetdirection{#2label};
		if \dprot == 0 then{
			{\vhorz[#1]{#2}{#3}};
		}else{
			{\vcrossauto[#1]{#2}{#3}};
		};
	}
}

\newcommand{\iarr}[3][red]{	%1. optional color, 2. Referenz to element, 3. label
        \pgfmathanglebetweenpoints{\pgfpointanchor{#2-Ifrom}{center}}{\pgfpointanchor{#2-Ito}{center}}
        \edef\myangletmp{\pgfmathresult}
        % \typeout{\myangletmp}
        \node [currarrow, rotate=\myangletmp, color=#1] at (#2-Ipos){};

        % Name not rotated -> label not rotated
        \pgfmathparse{notequal(\ctikzgetdirection{#2label},0)}
        \edef\namerottmp{\pgfmathresult}
        
        % rotate 180 if angle is between 90 and 270
        \pgfmathparse{\myangletmp + ifthenelse(and(\myangletmp>90,\myangletmp<270),180,0)}
        \edef\mylabangletmp{\pgfmathresult}
        % \typeout{\mylabangletmp}
        
        % placement above or below
        \tikzmath{
	        coordinate \dpC, \dpI;
	        \dpC = (#2-Ipos);
	        \dpI = (#2current.center);
	        if {\namerottmp} then {	
		        if \dpCy < \dpIy then{ %above
			        {\node [above, rotate=\mylabangletmp, #1] at (#2-Ipos) {#3};};
		        } else { %below
			        {\node [below, rotate=\mylabangletmp, #1] at (#2-Ipos) {#3};};
		        };
	        } else {	%not rotated -> use default position
		        {\node [anchor=\ctikzgetanchor{#2}{Ilab}, #1] at (#2-Ipos) {#3};};
	        };
        }
}

\begin{document}
	\begin{tikzpicture}
		
		\draw (130:4) to[L+, o-, l=$L_{ab}$, v=v, name=L1, i=i] (130:2) to[V+, -*, v_={\(v_{12}\)}, name=V1, i>_=~] (0:0);
		\varr[purple]{V1}{$v_{12}$}
		\varr{L1}{$v_{longName}$}
		\iarr{V1}{$i$}
		\iarr[green]{L1}{$i_{longName}$}
		
	\end{tikzpicture}%
\end{document}
```

judober, 2021-07-08:
Actually, I just noticed that this does not work as I intended:
Using the definition L+ (or similar) always generates (normal) voltage and current arrows. I can change the defintion to
```Latex
\tikzset{
    L+/.style={L, v, i, f, l, a},
    V+/.style={V, v, i, f, l, a},
}
```
fixes this. But I depend on `#2current.center` and thus I need the invisible label.
For the voltage it might be possible to circumvent this by using the bipole-position and compare it with the arrow-position. But for the current I need `#2current.center` as the arrow does not change when using `i_`.
Is it possible to define an (invisible) currentlabel withouth drawing the current arrow (and the same for the voltage)?

Rmano, 2021-07-08:
Not sure if I understand (could you make a small example?) Can't you use `\ctikzgetanchor{#2}{Ilab}` ?

Rmano, 2021-07-08:
Anyway, for now no, there is no way to have the void label for `v` and `i` and `f` without the corresponding arrow. I can look into it, if I find an easy way to do it...

judober, 2021-07-09:
Let's say I want only a current label for the first source and a voltage for the second. Using the definitions I get
```Latex
\begin{document}
	\begin{tikzpicture}
		
		\draw (40:4) to[L+, o-, l=$L_{ab}$, name=L1] (40:2) to[V+, -*, name=V1] (0:0);
		\draw (130:4) to[L+, o-, l=$L_{ab}$, name=L2] (130:2) to[V+, -*, name=V2] (0:0);
		\draw (220:4) to[L+, o-, l=$L_{ab}$, name=L3] (220:2) to[V+, -*, name=V3] (0:0);

		\iarr{V1}{$i$}	% current for 1. voltage source
		\varr{V2}{$v$}	% voltage vor 2. voltage source

	\end{tikzpicture}%
\end{document}

```

[image]

So every voltage and current arrow (black) is present due to using L+, V+.

Using L and V instead errors:
```Latex
\begin{document}
	\begin{tikzpicture}
		
		\draw (40:4) to[L, o-, l=$L_{ab}$, name=L1] (40:2) to[V, -*, name=V1] (0:0);
		\draw (130:4) to[L, o-, l=$L_{ab}$, name=L2] (130:2) to[V, -*, name=V2] (0:0);
		\draw (220:4) to[L, o-, l=$L_{ab}$, name=L3] (220:2) to[V, -*, name=V3] (0:0);

		\iarr{V1}{$i$}	% current for 1. voltage source
		\varr{V2}{$v$}	% voltage vor 2. voltage source

	\end{tikzpicture}%
\end{document}
```
I have to define `v=~` etc for the dipoles I want currents and voltages for.
```Latex
\begin{document}
	\begin{tikzpicture}
		
		\draw (40:4) to[L, o-, l=$L_{ab}$, name=L1] (40:2) to[V, -*, name=V1, i=~, l=~] (0:0);
		\draw (130:4) to[L, o-, l=$L_{ab}$, name=L2] (130:2) to[V, -*, name=V2, v=~, l=~] (0:0);
		\draw (220:4) to[L, o-, l=$L_{ab}$, name=L3] (220:2) to[V, -*, name=V3] (0:0);

		\iarr{V1}{$i$}	% current for 1. voltage source
		\varr{V2}{$v$}	% voltage vor 2. voltage source

	\end{tikzpicture}%
\end{document}
```
[image]

I still have the problem that the blue voltage arrow does not fully cancel out the black one. It would be nice if just defining `v` (and `v_` ) would not only define the voltage arrow anchors (described in 5.8.1) but also an anchor for the voltage label position.

Rmano, 2021-07-09:
Ok, I think I understand the problem. I will see if I can add the "label references" even when there is no label to print; it was not there because the idea is that when using a simple `v` then the user would take responsibility for everything (even the placement of the voltage label) and when using `v=~` the user wanted the arrows/sign automatically set and just take responsibility for the voltage label. 

I still thing you are trying to over-automatize it; for example, for the current I would do the following --- using the NewDocumentCommand to have two different optional argument --- and probably something similar for the voltages. Then I'll add the `v` wtc only when I need it, not to all components...

```latex
\documentclass[tikz]{standalone}
\usepackage{circuitikz}

\NewDocumentCommand{\iarr}{O{red} D(){below} m m}{% color, where, name, label
    \pgfmathanglebetweenpoints{\pgfpointanchor{#3-Ifrom}{center}}{\pgfpointanchor{#3-Ito}{center}}
        \edef\myangletmp{\pgfmathresult}
        %\typeout{\myangletmp}
        \node [currarrow, rotate=\myangletmp, color=#1 ] at (#3-Ipos){};
        \node [color=#1, rotate=\myangletmp, #2] at (#3-Ipos) {#4};
    }

\NewDocumentCommand{\varr}{O{blue} D(){below} m m}{% color, where, name, label
    \draw [thin, #1, -Triangle,] (#3-Vfrom) .. controls (#3-Vcont1) and (#3-Vcont2).. (#3-Vto)	node [midway, sloped,  #2] {#4};
    }

\begin{document}
	\begin{tikzpicture}
		\draw (40:4) to[L, o-, l=$L_{ab}$, name=L1] (40:2) to[V, -*, name=V1, i] (0:0);
		\draw (130:4) to[L, o-, l=$L_{ab}$, name=L2] (130:2) to[V, -*, name=V2, v] (0:0);
		\draw (220:4) to[L, o-, l=$L_{ab}$, name=L3] (220:2) to[V, -*, name=V3] (0:0);

                \iarr[blue]{V1}{$i$}	% current for 1. voltage source
                \iarr(above){V1}{$i$}	% current for 1. voltage source
                \varr(above){V2}{$v$}	% voltage vor 2. voltage source

	\end{tikzpicture}%
\end{document}
```

[image]

Anyway, I will open a enhancement ticket to track this.

judober, 2021-07-12:
First of all thanks for showing me `\NewDocumentCommand`.
Regarding automatisation you are probably right. However, a problem I have with your proposal: When inverting the current direction, the current letter is turned upside down:
```Latex
\begin{document}
	\begin{tikzpicture}
		
		\draw (40:4) to[L, o-, l=$L_{ab}$, name=L1] (40:2) to[V, -*, name=V1, i>] (0:0);
		\draw (130:4) to[L, o-, l=$L_{ab}$, name=L2] (130:2) to[V, -*, name=V2, v_] (0:0);
		\draw (220:4) to[L, o-, l=$L_{ab}$, name=L3] (220:2) to[V, -*, name=V3] (0:0);

		\iarr(above){V1}{$i$}	% current for 1. voltage source
		\varr(below){V2}{$v$}	% voltage vor 2. voltage source

	\end{tikzpicture}%
\end{document}
```

[image]

which can be fixed defining

```Latex
\NewDocumentCommand{\iarr}{O{red} D(){below} m m}{% color, where, name, label
    \pgfmathanglebetweenpoints{\pgfpointanchor{#3-Ifrom}{center}}{\pgfpointanchor{#3-Ito}{center}}
        \edef\myangletmp{\pgfmathresult}
        %\typeout{\myangletmp}
        \node [currarrow, rotate=\myangletmp, color=#1 ] at (#3-Ipos){};

        % rotate 180 if angle is between 90 and 270
	\pgfmathparse{\myangletmp + ifthenelse(and(\myangletmp>90,\myangletmp<270),180,0)}
	\edef\mylabangletmp{\pgfmathresult}

        \node [color=#1, rotate=\mylabangletmp, #2] at (#3-Ipos) {#4};
    }
```

[image]

An what I wanted to achieve originally was to have the same rotation for the voltage/current label as for the bipole-label. See e.g. this:

```Latex
\begin{document}
	\begin{tikzpicture}
		
		\draw (40:4) to[L+, o-, l=$L_{ab}$, v=v, name=L1, i=i] (40:2) to[V+, -*, v_={\(v_{12}\)}, name=V1, i=\(i\)] (0:0);
		\varr[purple]{V1}{$v_{12}$}
		\varr{L1}{$v_{longName}$}
		\iarr{V1}{$i$}
		\iarr[green]{L1}{$i_{longName}$}
		
	\end{tikzpicture}%
\end{document}
```
[image]

Thus, I need the to decect the label rotation. I tried

```Latex
\NewDocumentCommand{\iarr}{O{red} D(){below} m m}{% color, where, name, label
    \pgfmathanglebetweenpoints{\pgfpointanchor{#3-Ifrom}{center}}{\pgfpointanchor{#3-Ito}{center}}
        \edef\myangletmp{\pgfmathresult}
        %\typeout{\myangletmp}
        \node [currarrow, rotate=\myangletmp, color=#1 ] at (#3-Ipos){};

        % Name not rotated -> current label not rotated
        \pgfmathparse{notequal(\ctikzgetdirection{#3label},0)}
        \edef\namerottmp{\pgfmathresult}

        % rotate 180 if angle is between 90 and 270
        \pgfmathparse{(\myangletmp + ifthenelse(and(\myangletmp>90,\myangletmp<270),180,0))}
        \edef\mylabangletmp{\pgfmathresult}

        \node [color=#1, rotate=\mylabangletmp, #2] at (#3-Ipos) {#4};
    }
    \NewDocumentCommand{\varr}{O{blue} D(){below} m m}{% color, where, name, label
        \draw [thin, #1, -Triangle,] (#3-Vfrom) .. controls (#3-Vcont1) and (#3-Vcont2).. (#3-Vto) node [midway, sloped, #2] {#4};
        }
    
    \begin{document}
	    \begin{tikzpicture}
		    
		    \draw (90:4) to[L, o-, l=$L_{ab}$, name=L1] (90:2) to[V, -*, name=V1, i>, l=~] (0:0);
		    \draw (180:4) to[L, o-, l=$L_{ab}$, name=L2] (180:2) to[V, -*, name=V2, v_] (0:0);
    
		    \iarr(above){V1}{$i$}	% current for 1. voltage source
		    \varr(below){V2}{$v$}	% voltage vor 2. voltage source
    
	    \end{tikzpicture}%
    \end{document}
```

[image]

But this places the current label in the curcuit and thats why I used the position of the original label.

Rmano, 2021-07-12:
I am quite convinced that I'll try to generate the label info always (even if there is just an `i`). And the direction info for current arrows too. Do you think that this could fix it? 
Finding a syntax to put the labels but not the symbols seems quite awkward now that I think about it...

judober, 2021-07-13:
Yes I think having the label info would fix this problem.

Rmano, 2021-07-14:
Ok, let's move to #567

PR #452 (2020-10-08): Label and annotation information retrieval
Fixes #448 (or at least more or less helps it ;-), alas) 

[image]
PR #456 (2020-10-29): Fix typo in the manual
Thanks to @judober --- Fixes #448 

[image]