circuitikz/circuitikz #458: Colors of compoments not converted in dark background themes in Beamer

hurak, 2020-11-02:
It seems that the colors of components are somehow hardwired because if circuitikz is used within some dark background theme in Beamer, the components stay black, hence invisible in the dark background theme. But the connecting "wires" are correctly converted from black to white. Below is a minimum working example
```latex
\documentclass{beamer}

\mode<presentation>
{
 \usetheme{Pittsburgh}
 \usecolortheme{owl}                                      % if commented out, white background
 \setbeamercovered{transparent}
}

\usepackage[utf8]{inputenc}
\usepackage{babel}
\usepackage[T1]{fontenc}

\usepackage{tikz}
\usepackage{circuitikz}

\begin{document}
\begin{tikzpicture}
\begin{circuitikz}
    \draw (0,3)   
    to [short,o-,i>=i] (2,3)  
    to [R=$R$] (2,0)
    to [short,-o](0,0);   
\end{circuitikz}
\end{tikzpicture}
\end{document}```
```

Rmano, 2020-11-02:
Try using `\begin{circuitikz}[color=white}`. But yes, color is tricky, there is a whole chapter about it in the manual...

hurak, 2020-11-02:
Excellent! This saves my slides (from the need to be boring white) :-)

closed 2020-11-02 by hurak

Rmano, 2020-11-02:
Notes a couple of things:
1.  if shouldn't nest `tikzpictures` and `circuitikz`, they're basically synonymous
2. if your background is black, you should also issue `\ctikzset{open poles fill=black}`, otherwise, they'll be filled white.

```
\documentclass{beamer}

\mode<presentation>
{
 \usetheme{Pittsburgh}
 \usecolortheme{owl}% if commented out, white background
 \setbeamercovered{transparent}
}

\usepackage[utf8]{inputenc}
\usepackage{babel}
\usepackage[T1]{fontenc}

\usepackage{tikz}
\usepackage[RPvoltages]{circuitikz}
\ctikzset{open poles fill=black}

\begin{document}
\begin{circuitikz}[color=white]
    \draw (0,3)   
    to [short,o-,i>=i] (2,3)  
    to [R=$R$] (2,0)
    to [short,-o](0,0);   
\end{circuitikz}
\end{document}
```

[image]