circuitikz/circuitikz #459: Flow arrow empty shape

patrikradvanyi, 2020-11-18:
Hi!

Can you create an option for a flow arrow which has an empty head? In Hungary this is the standard notation of the current arrow and it will be easier to use in education. I'm attached a picture about it below.

[image]

Thank you! 😄

Rmano, 2020-11-18:
This is possible already if you use the "Advanced currents, voltage and flows" (around page 154). You just define your own macro for the arrow and then apply it to your component, using the provided anchors.

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\newcommand{\myflow}[3][above]{%arguments: [options (default above)] {component name} {label to add}
    \draw [-{Triangle[open]}] (#2-Ffrom) -- (#2-Fto) node[pos=0.5, #1]{#3};
}
\begin{document}
\begin{circuitikz}
    \draw (0,0) to[short, f, name=S1] ++(3,0);
    \myflow{S1}{$I$}
\end{circuitikz}
\end{document}
```

[image]

patrikradvanyi, 2020-11-18:
Oh' okay! Now it's clear! 😄 This level is too commplicated for me yet.  I just wanted advise to enhance the package something like this:

(...) to[short,f=$I$,**openarrow**] (...)

Thak you again!

closed 2020-11-18 by patrikradvanyi

Rmano, 2020-11-18:
> (...) to[short,f=$I$,**openarrow**] (...)

yes, that could be done, but you know --- there is an infinity of possible options, for this, I added the "advanced" thing...

patrikradvanyi, 2020-11-18:
Of course you right. I'm just too dummy for this 😅