circuitikz/circuitikz #462: Text for transmission line stub

safak-oezesenlik, 2020-12-07:
Hi,

is this a bug?

[image]

I can't name the TL stubs correctly.

`\begin{figure}[!htbp]
    \centering
    \begin{tikzpicture}
    \draw (0,0) node[](in){} to[TL,o-,l=\mbox{\(\mathit{TL}_1\)}] ++(3,0) coordinate(there) to[TL,l=\mbox{\(\mathit{TL}_1\)}] ++(3,0) coordinate(here) to[TL,-o,l=\mbox{\(\mathit{TL}_1\)}] ++(3,0);
    \draw (here) to[short,*-] ++(0,-0.5) node[tlinestub,rotate=270]{\rotatebox{90}{\(\mathit{TL}_1\)}};
    \draw (there) to[short,*-] ++(0,-0.5) node[tlinestub,rotate=270]{TL1};
    \end{tikzpicture}
    \caption{Input matching for Cree \lstinline{CG2H80030D}.}
    \label{fig:lumped_matching}
\end{figure}`

Best regards

Rmano, 2020-12-07:
This is a very old component --- the label is supposed to be like the second vertical stub. Not optimal, I know... but I am not sure it is possible to fix it without disruption, although I think about it --- maybe adding a couple of anchors. 
Where do you think the label should go?

Rmano, 2020-12-07:
BTW, it is not a bug because it is not supposed to work, although I agree it should. If you look at the manual, there is no hint that the argument of the node is used:

[image]

safak-oezesenlik, 2020-12-07:
Hi, 

can it be solved to work like for a microstrip stub?

[image]
[image]

Rmano, 2020-12-07:
@safak-oezesenlik  yes --- basically the "cylinder-style" thingies are to be considered legacy. In my opinion, they are not really symbols...

safak-oezesenlik, 2020-12-07:
The "official" symbols for transmission lines are cylinder-like, tho.

What is a quick fix for this? I need to submit my thesis in two weeks... I just need the next to be beside the stub.

Rmano, 2020-12-07:
You can use this version of the stub --- maybe I will push this change to the library, it's better than nothing --- but looking at the code I found another bug ;-)

Simply copy the `\makeatletter`...`\makeatother` thing in your preamble *after* loading `circuitikz`.

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}

\makeatletter
\pgfdeclareshape{tlinestub}{
    \savedmacro{\ctikzclass}{\edef\ctikzclass{RF}}
    \saveddimen{\scaledRlen}{\pgfmathsetlength{\pgf@x}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}}
    \savedanchor{\northeast}{%
        \pgfmathsetlength{\pgf@circ@scaled@Rlen}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}
        \pgf@circ@res@step = \ctikzvalof{bipoles/tline/width} \pgf@circ@scaled@Rlen
        \pgf@x=1.2\pgf@circ@res@step
        \pgf@circ@res@step = \ctikzvalof{bipoles/tline/width} \pgf@circ@scaled@Rlen
        \pgf@y=.2\pgf@circ@res@step%
    }
    % the center is on the left side of the shape for facility of usage
    \anchor{north}{\northeast\pgf@x=0.5\pgf@x\relax}
    \anchor{east}{\northeast\pgf@y=0cm\relax}
    \anchor{south}{\northeast\pgf@y=-\pgf@y \pgf@x=0.5\pgf@x\relax}
    \anchor{west}{\pgfpointorigin}
    \anchor{north east}{\northeast}
    \anchor{north west}{\northeast\pgf@x=0cm\relax}
    \anchor{south east}{\northeast\pgf@y=-\pgf@y}
    \anchor{south west}{\northeast\pgf@x=0cm\pgf@y=-\pgf@y}
    \anchor{center}{\pgfpointorigin}
    \anchor{text}{\northeast\pgf@xa=\pgf@x\pgf@ya=\pgf@y
        \pgfpoint{\dimexpr-.5\wd\pgfnodeparttextbox+.6\pgf@xa}
        {\dimexpr-.5\dp\pgfnodeparttextbox+.5\ht\pgfnodeparttextbox+\pgf@ya}}
    \behindforegroundpath{
        \pgfstartlinewidth=\pgflinewidth

        \pgf@circ@scaled@Rlen=\scaledRlen
        \pgf@circ@res@step=\ctikzvalof{bipoles/tline/width} \pgf@circ@scaled@Rlen
        % this is clearly a bug... it nullifies the line above
        \pgf@circ@res@step=0.6\pgf@circ@scaled@Rlen

        \pgfscope\begin{pgftransparencygroup}
            \pgfpathellipse{\pgfpoint{0.5\pgf@circ@res@step}{0\pgf@circ@res@step}}{\pgfpoint{0.125\pgf@circ@res@step}{0\pgf@circ@res@step}}{\pgfpoint{0\pgf@circ@res@step}{0.25\pgf@circ@res@step}}
            \pgf@circ@maybefill
            \pgfpathmoveto{\pgfpoint{0.5\pgf@circ@res@step}{0.25\pgf@circ@res@step}}
            \pgfpathlineto{\pgfpoint{1.5\pgf@circ@res@step}{0.25\pgf@circ@res@step}}
            \pgfpatharc{90}{-90}{0.125\pgf@circ@res@step and 0.25\pgf@circ@res@step}
            \pgfpathlineto{\pgfpoint{0.5\pgf@circ@res@step}{-0.25\pgf@circ@res@step}}
            \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
            \pgf@circ@draworfill
            \pgfpathellipse{\pgfpoint{0.5\pgf@circ@res@step}{0\pgf@circ@res@step}}{\pgfpoint{0.125\pgf@circ@res@step}{0\pgf@circ@res@step}}{\pgfpoint{0\pgf@circ@res@step}{0.25\pgf@circ@res@step}}
            \pgfusepath{draw}
        \end{pgftransparencygroup} \endpgfscope
        \pgfpathmoveto{\pgfpointorigin}
        \pgfpathlineto{\pgfpoint{0.5\pgf@circ@res@step}{0pt}}
        \pgfusepath{draw}
    }
}
\makeatother

\begin{document}
    \begin{tikzpicture}
    \draw (0,0) node[](in){} to[TL,o-,l=\mbox{\(\mathit{TL}_1\)}] ++(3,0) coordinate(there) to[TL,l=\mbox{\(\mathit{TL}_1\)}] ++(3,0) coordinate(here) to[TL,-o,l=\mbox{\(\mathit{TL}_1\)}] ++(3,0);
    \draw (here) to[short,*-] ++(0,-0.5) node[tlinestub,rotate=270]{\rotatebox{90}{TL1}} ;
    \draw (there) to[short,*-] ++(0,-0.5) node[tlinestub,rotate=270]{TL1};
    \end{tikzpicture}
\end{document}
```
[image]

safak-oezesenlik, 2020-12-07:
This is great, thank you so much!

Is it possible to get the text closer to the circuit element after rotating?

I hope what I have requested is possible.

Rmano, 2020-12-07:
Do you mean the non-rotating one? Yes, it's a bit far away... but if I change this I will break the other one. I'll do

```latex
 \draw (here) to[short,*-] ++(0,-0.5) node[tlinestub,rotate=270]{\rotatebox{90}{\kern -3pt\relax TL1}} ;
```

PR #464 (2020-12-08): Add a text anchor for legacy tlinestub.
Also, the shape ignored the keys --- changed this and the key so
that the default shape will not change.

Fixes   #462

This will change past behavior but given that it was completely broken...
closed 2020-12-08 by Rmano