circuitikz/circuitikz #467: European XNOR Gate

Schlepptop, 2020-12-14:
According to the IEC 60617-12, an XNOR Gate should not be represented as a negated XOR (XOR with a small circle) but rather like this:
[image]

I understand that this is probably less intuitive, but would it be possible to at least make this an option, if not the default?

Rmano, 2020-12-14:
Hmmm --- I am not an expert at all of European symbols (never used in the last 40 years spent in Italy, US, and Spain). 
For sure that change should be added with an option, not in an absolute way. Moreover, IEC say this:

[image]

So I am not sure that the symbol is still current.

Rmano, 2020-12-14:
Momentarily, you can use this trick to use the port you like:

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\pgfcircdeclareeurologicport{xnor}{$=$}{2}{}
\begin{document}
\begin{circuitikz}[
    european]
    \node [xnor port]{};
\end{circuitikz}
\end{document}
```

[image]

Schlepptop, 2020-12-14:
> Hmmm --- I am not an expert at all of European symbols (never used in the last 40 years spent in Italy, US, and Spain). 
> For sure that change should be added with an option, not in an absolute way. Moreover, IEC say this:
> 
> [image]
> 
> So I am not sure that the symbol is still current.

The 1997 version of the standard was withdrawn in favor of one published in 2012. Afaik the stance on this specific symbol didn't change, though.

Rmano, 2020-12-15:
What about this: 

```latex
\begin{document}
\begin{circuitikz}[
    european]
    \node [xnor port]{};
\end{circuitikz}

\ctikzset{european xnor style=iec}

\begin{circuitikz}[
    european]
    \node [xnor port]{};
\end{circuitikz}
\end{document}

Rmano, 2020-12-15:
Hmmm... ok, although I found also this https://www.robotshop.com/community/tutorials/show/electronics-done-quick-7-logic-gates 

I find the European symbols confusing. The fact that the shapes are the same and you have to read the labels makes the thing much more difficult.

Schlepptop, 2020-12-15:
> What about this:
> 
> ```
> \begin{document}
> \begin{circuitikz}[
>     european]
>     \node [xnor port]{};
> \end{circuitikz}
> 
> \ctikzset{european xnor style=direct}
> 
> \begin{circuitikz}[
>     european]
>     \node [xnor port]{};
> \end{circuitikz}
> \end{document}
> ```
> 
> [image]

That looks good to me.

PR #471 (2020-12-15): Add a different shape for european xnor port
Fixes #467 

[image]
closed 2020-12-15 by Rmano