circuitikz/circuitikz #469: Extend vacuum tubes to PMTs

ferdymercury, 2020-12-15:
Thank you for this very nice Package.

I wanted to suggest a future feature expanding on the already existing diode, triode and pentodes.

[image]

It would be nice to be also to draw photomultiplier tubes, which consist of several Dynodes in addition to anode and cathode.

Here some inspiring screenshots from different drawing styles I found in the literature:
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

See https://www.hamamatsu.com/resources/pdf/etd/PMT_handbook_v4E.pdf
http://lmu.web.psi.ch/docu/manuals/bulk_manuals/PMTs/Photonis_PMT_basics.pdf

Rmano, 2020-12-15:
Uy, these are very very specific components --- I suspect that a one-off Ti*k*Z `pic` would be more interesting. 
Let's ping the author of the tubes... @jesseopdenbrouw

ferdymercury, 2020-12-15:
Thanks! An alternative would be to just define the IEC dynode symbol that looks like a curved capacitor with a line across, and then one builds itself its tube as needed in the tex file.

Rmano, 2020-12-15:
Yes, that could be an idea. Which size should that component be, for example, with respect to the tubes?

ferdymercury, 2020-12-15:
Well, that depends on how many dynodes a tube has.
For the symbol, it would be helpful that the size of the dynode symbol is smaller than that of a resistance, so that a resistor fits well inbetween two dynodes, as in the pic below.

[image]

Rmano, 2020-12-16:
What about this?

[image]

Questions: should be the arc half a circle or less? Configurable? 

Now you can configure width, height and the position of the arc.

ferdymercury, 2020-12-16:
Thanks so much. I think it is a little bit less than half a circle. See https://www.noao.edu/ets/Mechanical/Policies/ANSI%20Y32.2-1975.pdf point 7.1.4.1 on page 51
Looks like it stops at about +/- 60 or 70 degrees, it's hard to see on the IEC picture. I don't think the angle needs to be configurable.

Rmano, 2020-12-16:
Now is completely configurable:

[image]

ferdymercury, 2020-12-16:
Gracias Romano, genial!!

ferdymercury, 2020-12-16:
I guess that, to fully build the PMT tube, one would also need a photocathode monopole, and an anode monopole.

[image]

[image]

Rmano, 2020-12-16:
Ah! Play with the parameters of dynode and... 

[image]

Rmano, 2020-12-16:
I think that the grid can be easily made with simple lines...

ferdymercury, 2020-12-16:
simply amazing... thanks!

Rmano, 2020-12-16:
Now, should I put it with the antennas (RF) or with the tubes? Opinion?

Rmano, 2020-12-16:
I think tubes is best, after magnetron

Rmano, 2020-12-16:
The merge will close this isuue, feel free to repoen if needed (also to @jesseopdenbrouw )

ferdymercury, 2020-12-16:
Thanks so much.
One minor nomenclature thing. If this element is thought to be used for anode, dynode and cathode, then it should be maybe renamed to the generic term electrode.
The 'electrode' customization creates dynodes, anodes or cathodes, rather than the 'dynode' customization.

On the other hand, if the default values of the parameters create a dynode, then I guess it makes sense to call it "dynode" heheh.

Rmano, 2020-12-16:
> On the other hand, if the default values of the parameters create a dynode, then I guess it makes sense to call it "dynode" heheh.

Yes, that one was my reasoning ;-)

PR #473 (2020-12-16): Add dynode component
Thanks to ferdymercury https://github.com/circuitikz/circuitikz/issues/469

Fixes #469 

[image]
closed 2020-12-16 by Rmano