circuitikz/circuitikz #475: node labels and color

ferdymercury, 2020-12-18:
I found here https://tex.stackexchange.com/questions/226534/label-node-in-circuitikz a way to label the node.

I was wondering if there is a way to change the color of both the node name and the dot (circ). (I could not find how).

If yes, maybe all these informations could be added to the manual?

Thanks in advance.

Rmano, 2020-12-18:
I am not sure if I understand... is this that you are looking for?

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{circuitikz}[
    ]
    \draw (0,0) to[R] ++(3,0) coordinate(a) to[R] ++(3,0);
    \path[color=red] (a) node[circ]{} node[above]{$A$};
\end{circuitikz}
\end{document}
```

[image]

This is really a Ti*k*Z thing; the only problem (and it is noticed in the manual, in the "color" chapter) is that for `circuitikz` components to follow the path color, you need to use `[color=red]` (not the short form `[red]` normally used in Ti*k*Z).

ferdymercury, 2020-12-18:
Yes, you understood correctly my question.

The workaround I had used was:
`\fill[red] (coordinate) node[anchor=south] {$A$} circle (2pt);`

Maybe your example can be added on the manual 'How to add labels on nodes' even if it is not using specifically circutikz?
Thanks!

ferdymercury, 2020-12-18:
Also, would it be possible to achieve the same in one line instead of two, using the label keyword, as in https://tex.stackexchange.com/questions/226534/label-node-in-circuitikz ?

Rmano, 2020-12-18:
Like this? 

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{circuitikz}
    \draw (0,0) to[R] ++(3,0) node[color=red, circ, label={[red]above:$A$}]{} to[R] ++(3,0);
\end{circuitikz}
\end{document}
```

[image]

About the manual: this is more a Ti*k*Z thing; the `label` keyword applies to any node --- look at 17.10, "The label and pin options", around page 250.

ferdymercury, 2020-12-18:
Thanks! Solved

closed 2020-12-18 by ferdymercury