circuitikz/circuitikz #485: Infinite loop

mohe2015, 2021-01-11:
Hi,

I don't know if this is a bug in circuitikz or in tikz itself but the following example creates an intinite loop:
```latex
\documentclass[]{article}
\usepackage{circuitikz}
\begin{document}

\begin{circuitikz}
	\draw (0,0) -- +()
	
	\draw ;
\end{circuitikz}}

\end{document}
```
Obviously this is invalid code (the second draw doesn't really matter it's just important that something comes after the first invalid one). This bit me in a project for university so I would love a fix if this is easily fixable.

Rmano, 2021-01-11:
Hmmm... with either `circuitikz` and plain `tikz`:

```latex
\documentclass[]{article}
\usepackage{tikz}
\begin{document}

\begin{tikzpicture}
    \draw 
	\draw ;
\end{tikzpicture}

\end{document}
```

you get an error before the loop:

```
! Package pgf Error: No shape named `' is known.

See the pgf package documentation for explanation.
Type  H <return>  for immediate help.
 ...                                              
                                                  
l.6 	\draw (0,0) -- +()
``` 

anything can happen after an error... 

Anyway, notice that even the simpler 

```
\begin{circuitikz}
    \draw \node{};
\end{circuitikz}
```

(typing `\node` instead of `node`) will have the same behavior. Basically, it is not a bug of `circuitikz`, it's the standard behavior of Ti*k*Z. 

It would be better not to loop? Maybe yes. But it's definitely a more basic issue (note that it's impossible to prevent looping in LaTeX or TeX --- `\def\blub{\blub}\blub`  will do the same...

closed 2021-01-11 by Rmano

mohe2015, 2021-01-11:
May have something to do with my IDE, i need to check that tomorrow. Thanks a lot for a fast response though.

Rmano, 2021-01-12:
Probably your IDE (like most of them) is compiling in `nonstopmode`, so it's skipping the error and falls directly in the infinite loop...
 https://tex.stackexchange.com/questions/258814/what-is-the-difference-between-interaction-nonstopmode-and-halt-on-error