circuitikz/circuitikz #492: Shift position of voltage node in open path

matthuszagh, 2021-02-10:
I'm attempting to draw the following circuit:

```tex
\usepackage[american]{circuitikz}
\newcommand*{\equal}{=}
\begin{circuitikz}
  \draw (0,0) node[left]{B} to[short,o-,i=$I_{\mathrm{B}}$] ++(1,0)
  to[vsource,l_=$V_{\mathrm{BE}}\equal\SI{0.7}{V}$] ++(0,-2)
  to[short,-*] ++(1,0) coordinate(e)
  to[short] ++(1,0) coordinate(a)
  to[isource,invert,l=$\beta I_{\mathrm{B}}$] ++(0,2)
  to[short,-o] ++(1,0) coordinate(c) node[right]{C}
  (e) to[short,-o] ++(0,-1) node[right]{E}
  (c) to[open,v_=$V_{\mathrm{CE}}>\SI{0.3}{V}$] ($(a)+(1,0)$);
\end{circuitikz}
```

This produces

[image]

I'd like to shift the open path voltage node to the right so that it doesn't interfere with the current source. Is there a way to do this simply without using a workaround (for instance, a way to set the node anchor)?

If I use the following workaround, I get the desired result, but I was hoping there might be a more built-in way to achieve this.

```tex
\begin{circuitikz}
  \draw (0,0) node[left]{B} to[short,o-,i=$I_{\mathrm{B}}$] ++(1,0)
  to[vsource,l_=$V_{\mathrm{BE}}\equal\SI{0.7}{V}$] ++(0,-2)
  to[short,-*] ++(1,0) coordinate(e)
  to[short] ++(1,0) coordinate(a)
  to[isource,invert,l=$\beta I_{\mathrm{B}}$] ++(0,2)
  to[short,-o] ++(1,0) coordinate(c) node[right]{C}
  (e) to[short,-o] ++(0,-1) node[right]{E}
  (c) to[open,v_=$$] ($(a)+(1,0)$)
  ($(c)!0.5!($(a)+(1,0)$)$) node[anchor=west]{$V_{\mathrm{CE}}>\SI{0.3}{V}$};
\end{circuitikz}
```

I couldn't find the answer to this question in the manual, but please do let me know if I've missed something. Thanks!

Rmano, 2021-02-10:
Well, I think your "workaround" is the Right Thing To Do, but another possibility is to hide the width of the voltage label, like:

     to[open,v=\rlap{$V_{\mathrm{CE}}>\SI{0.3}{V}$}]

...and then adjust the bounding box. 

BTW, if you are on a new `circuitikz` (≥ 1.3.0) the  ugly `\def\equal` workaround is not needed anymore:

```latex
\documentclass[border=10pt]{standalone}
\usepackage[american, siunitx]{circuitikz}
\usepackage{amsmath}
\begin{document}
\begin{circuitikz}
  \draw (0,0) node[left]{B} to[short,o-,i=$I_{\mathrm{B}}$] ++(1,0)
  to[vsource,l_={$V_{\mathrm{BE}}=\SI{0.7}{V}$}] ++(0,-2)
  to[short,-*] ++(1,0) coordinate(e)
  to[short] ++(1,0) coordinate(a)
  to[isource,invert,l=$\beta I_{\mathrm{B}}$] ++(0,2)
  to[short,-o] ++(1,0) coordinate(c) node[right]{C}
  (e) to[short,-o] ++(0,-1) node[right]{E}
  (c) to[open,v=\rlap{$V_{\mathrm{CE}}>\SI{0.3}{V}$}] ($(a)+(1,0)$);
  % path to enlarge the bounding box because we hide the size
  \path (c) ++(2,0);
\end{circuitikz}
\end{document}
```
[image]

Rmano, 2021-02-10:
Another option (this does no need to mess with the bounding box):

    to[open,v={\hspace*{1cm}$V_{\mathrm{CE}}>\SI{0.3}{V}$}]

Rmano, 2021-02-10:
Finally, you can use the "advanced voltage" trick --- 

```latex
  (c) to[open,v=\null, name=vce] ($(a)+(1,0)$);
  \node[right] at (vce-Vlab) {$V_{\mathrm{CE}}>\SI{0.3}{V}$};
```
Although this seems a bit too to the right for my tastes.

matthuszagh, 2021-02-10:
Thanks for all the great suggestions! For what it's worth, I personally like the last suggestion best (it feels like the most "faithful" use of tikz to me). I used `\node[right=-10pt]` because I agree that the way you posted places the label too far to the right. I'm closing this issue since this answers my question.

closed 2021-02-10 by matthuszagh