circuitikz/circuitikz #496: Adding a symbol for Graphene Fet.

emcees2, 2021-03-09:
Is it possible to add a symbol for the Graphene Fet?
[image]

And how should i do it?

Kind regards,
Cees

Rmano, 2021-03-09:
Probably the best way to proceed is starting from a form of `igbt` and tweak the shape a bit. As a stop-gap (the hexagon is not a hexagon, I know) the idea could be this:

```latex
\begin{tikzpicture}[]
    \tikzset{gfet/.style={
        nigbt,
        circuitikz/transistors/arrow pos=end,
        circuitikz/tripoles/nigbt/outer base height=0.3,
        circuitikz/tripoles/nigbt/outer base thickness=2,
        circuitikz/tripoles/nigbt/width=0.5,
        circuitikz/tripoles/nigbt/height=1.3,
    }}
    \draw node [gfet](G){} (G.inner up) -- ++(0.4,-0.27) --++(0,-0.3) -- (G.inner down);
\end{tikzpicture}
```

[image]

I will see if I can give  it a shot as soon as I have a bit of time. Basically, the idea is to look into `tex/pgfcirctripoles.tex`, duplicate the `igbt` component, and modify it to create the new one.

Rmano, 2021-03-12:
Do you like these?:

[image]

Rmano, 2021-03-12:
@emcees2 I forgot the ping for the above message. I think I have them --- I'll try to integrate them soon.
How do you like to be cited/thanked in the manual?

PR #499 (2021-03-13): Graphene fets
[image]

Closes #496
closed 2021-03-13 by Rmano