circuitikz/circuitikz #501: Flip-flop with nu=1 also draws pin at "down"

BillNace, 2021-03-14:
If I use the following to make an active-low preset on a flip-flop, I also suddenly get a pin on the reset position (down).

`\documentclass[tikz]{standalone}
\usepackage{tikz}
\usepackage[RPvoltages]{circuitikz}
\tikzset{flipflop DQ/.style={flipflop, scale=.7,
         flipflop def={t1=D, t6=Q, c3=1,
         clock wedge size=.3, font=\normalsize}
}}
\tikzset{flipflop DQP/.style={flipflop, scale=.7, % Has D, Q, and Preset
         flipflop def={t1=D, t6=Q, c3=1, nu=1,
         clock wedge size=.3, font=\normalsize}
}}
\begin{document}

\begin{circuitikz}
  \node[flipflop DQP] (ff0) at (0,0) {};
  \node[flipflop DQ]  (ff1) at (3,0) {};
\end{circuitikz}

\end{document}`

Rendered thusly:
[image]

Rmano, 2021-03-14:
Yes, thanks. This is fixed in #502

Rmano, 2021-03-14:
See also a longish answer in https://tex.stackexchange.com/questions/587213/how-do-i-use-the-reset-connection-on-a-flip-flop

closed 2021-03-14 by Rmano

Rmano, 2021-03-14:
This is a small, but quite nasty bug. I will release version 1.3.2 today-ish