circuitikz/circuitikz #506: Weird Labelling & Re-Placement of Paths and Coordinates since Version 1.2.7

dedoenitz, 2021-03-22:
This is not minimal, since in "larger" diagrams the interdependencies between components make it difficult to pinpoint the exact problem (and exactly this seems to be the problem here), but consider this example:

```LaTeX
\documentclass[]{article}

\usepackage[T1]{fontenc}
\usepackage[utf8]{inputenc}
\usepackage[english]{babel}
\usepackage[a4paper, left=0mm, right=0mm]{geometry}
\usepackage{tikz}
\usepackage[european]{circuitikz} % circuitikz-1.2.7

\begin{document}

	\begin{figure}[h!]
	\begin{center}
	\begin{circuitikz}[x=1cm, y=1cm, scale=1.5]
		\coordinate (Xb) at (0,0); % X, Bottom
		\draw (Xb) [diode, *-o] to ++(-1,1) coordinate[name=Xl] [diode, o-*] to ++(1,1); % ~
		\draw (Xb) [diode, *-o] to ++(1,1) coordinate[name=Xr] [diode, o-*] to ++(-1,1) coordinate[name=Xt]; % ~
		\node[below=1cm] at (Xt) {$X$}; % ~
		\draw (Xl) [open,v=$U_E$] to (Xr); % ~
		\draw (Xr) [open,v^={\ }] to (Xl); % ~
		\draw (Xt) [short] to ++(0,2) coordinate[name=Uin];
		\draw (Xb) [short] to ++(0,-2) coordinate[name=Ugnd];
		\draw (Uin) [short] to ++(1.5,0) coordinate[name=C1] [C,l=$C_1$, -*] to (Ugnd -| C1); % C1
		\draw (C1) [short, *-] to ++(0.8,0) coordinate[name=D1] [zD, invert, a=$D_1$] to ++(0,-2) node[pnp, scale=1.5, anchor=B](T1){} [R, a={}, l=$R_1$, *-*] to (Ugnd -| D1); % D1, T1, R1
		\node[right] at (T1) {$T_1$}; % T1 Name
		\draw (T1.E) [R, a=$R_2$, -*] to (Uin -| T1.E) coordinate[name=R2] [short, -*, a={}] to (D1); % R2
		\draw (R2) [short, *-] to ++(0.8,0) coordinate[name=R3] [open] to (R3 |- Ugnd); % (R3, D2)
			\node[above] at (R3) {1}; %%% Test R3 (correct placement)
			\node[right=0.8cm] at (R3) {2}; %%% Test R3++(0.8, 0)
		\draw (R3) [short, *-] to ++(0.8,0) coordinate[name=D3l] [D, invert, *-, a=$D_3$] to ++(2.35,0) coordinate[name=D3r]; % D3
			\node[above] at (D3r) {Test}; %%% Test D3r
		\draw (D3l) -- ++(0,-1.6) node[npn, scale=1.5, rotate=90, anchor=C](T2){}; % T2
		\node[below right=0.5cm] at (T2) {$T_2$}; % T2 Name
		\draw (T2.E) node[npn, scale=1.5, rotate=90, anchor=B](T3){}; % T3
		\node[below right=0.5cm] at (T3) {$T_3$}; % T3 Name
		\draw (T3.C) [short, -*] to (T3.C -| T2.C);
		\draw (T2.B) [short] to (T1.C -| T2.B) coordinate[name=Ictrl]; %
		\draw (T3.E) [short] to (T3.E -| D3r) -- (D3r);
		\draw (T1.C) [short, -*] to (Ictrl);
		\draw (Ictrl) node[npn, scale=1.5, xscale=-1, anchor=C](T4){}; % T4
		\node[below right=0.5cm] at (T4) {$T_4$}; % T4 Name
		\draw (T4.E) [open, v^=$U_0$] to (T4.E |- Ugnd); % Uref
			\node[above] at (R3) {3}; %%% Test R3 (correct placement)
		\draw (R3) [R, l=$R_3$] to (T1.C -| R3) -- (T4.E -| R3) coordinate[name=D2] [zD, invert, -*, l=$D_2$] to (R3 |- Ugnd); % R3, D2 %[short, -, l={}] to
			\node[above] at (R3) {4}; %%% Test R3 (incorrect placement)
		\draw (T4.E) [short, -*] to (D2);
		\draw (D3r) [short, *-] to ++(0.75,0) coordinate[name=R4] [R, l=$R_4$] to (T3.B -| R4) [short, -*, l={}] to (T3.B); % R4
		\draw (T4.B) -- (T4.B -| R4) -- ++(0.4,0) node[genericpotentiometershape, rotate=90, anchor=wiper](R6){}; % R6
		\node[right=0.15cm] at (R6) {$R_6$}; % R6 Name
		\draw (R4) [short, *-] to (R4 -| R6.right) coordinate[name=VD] [R, bipole nodes={none}{none}, l=$R_5$] to (R6.right); % R5
		\draw (R6.left) [R, -*, l=$R_7$] to (R6.left |- Ugnd); % R7
		\draw (VD) [short, *-] to ++(0.8,0) coordinate[name=C2] [C, -*, l=$C_2$] to (C2 |- Ugnd); % C2
		\draw (C2) [short, *-o] to ++(0.6,0) coordinate[name=Uo] [open, o-o, v^=$U_A$] to (Uo |- Ugnd);
		\draw[] (Ugnd) [short, -o] to (Uo |- Ugnd);
	\end{circuitikz}%
	\end{center}%
	\end{figure}

\end{document}
```
Using the **fallback point for version 1.2.7 correctly** yields
[image]
Note how the "test marker numbers" 1, 3-4 (which are set in the indented and triple-%-commented lines in the example code above) correctly overlap, since they all are placed at the same coordinate `(R3)`, even though being created at different times.

Now consider the output **using the latest version(s)**:
[image]
Suddenly some paths and components are misaligned. As the "test marker" 4 is now shifted shows that the coordinate `(R3)` was incorrectly altered in line 44.
Also note the neatly arranged component labels at the bottom of the bridge rectifier _X_, that shouldn't be there.

Sorry again for not providing a MWE. However, again, I really have no idea what's going on.

Rmano, 2021-03-22:
I have to check more (bed time now) but the syntax 

    [diode, *-o] to ++(-1,1)

is wrong, or at least works only by chance. You should use

    to[diode, *-o] ++(-1,1)

dedoenitz, 2021-03-22:
Ah ok, thanks, this was the issue.
I remember realizing this in an earlier version, but, seeing that switching to the correct syntax shown in the documentation didn't change anything, I just assumed that's an alternative and never bothered changing all my code. Given this particular problem, I would've never thought about this.
Interestingly enough, only the affected paths needed to be changed in this example.

Also thanks for your hint for naming coordinates and your general support for this awesome project.

closed 2021-03-22 by dedoenitz

Rmano, 2021-03-23:
Thanks for the nice words!

Yes, the syntax you used seems to work sometimes. Let me explain what's happening (I hope you don't mind --- this problem has happened a couple of times already and this will be a reference answer). 

When Ti*k*Z finds a `to` in a path, it executes a special macro, which is supposed to be defined into the `to` options --- that macro is called the `to path` one. If none is defined, `to` simply draws a line like it was a `--`. The styles like `short`, `resistor` etc will set the `to path` macro to the appropriate one to draw the component, do the connection, and set up the labels and so on.

When you say `\draw ..... [style] to (coord)`, in the `[...]` you are setting  `style`  *globally for the rest of the path* --- so the next `to` *did* get the correct `to path` macro, but a lot of styles will be lingering even *after* that, while using `to[style]` keeps the changes local to that element. This is why you ended up needing to do a strange thing like

    [R, bipole nodes={none}{none}, l=$R_5$] 

which should not be needed at all: the previous `*-` pole specification is **global** to the path, so it's lingering to the following definitions.  Look:

```latex
\documentclass[border=2pt]{standalone}
\usepackage{circuitikz}
\begin{document}
\begin{circuitikz}
    \draw (0,1)  [short, *-] to ++(2,0) [R, bipole nodes={none}{none}, l=$R_5$] to ++(3,0);
    \draw (0,0) [short, *-] to ++(2,0) [R, l=$R_5$] to ++(3,0);
    \draw (0,-1) [diode, *-] to ++(2,0) to ++(3,0);
\end{circuitikz}
\end{document}
```

[image]

See? If you do that correctly, you have

```latex
\documentclass[border=2pt]{standalone}
\usepackage{circuitikz}
\begin{document}
\begin{circuitikz}
    \draw (0,0) to[short, *-] ++(2,0) to[R, l=$R_5$] ++(3,0);
    \draw (0,-1) to[diode, *-] ++(2,0) to ++(3,0);
\end{circuitikz}
\end{document}
```

[image]

That can be (ab)used sometimes, to have all the element in the path with the same poles, like `\draw [*-*] to[R] ... to[C] ... ` but it's a bit of forcing the machine. 

Finally, if the global styles you set do not move the final point of the path, you got lucky and the connection will work (although, probably, with bad line joins in case of 90 degrees connection). Otherwise, you see the gaps you were seeing. Before 1.3.0, the `to path` macros was doing a lot of unnecessary movements and was breaking the line joins, so the behavior did change --- but the fact that it worked in 1.2.7 and not in 1.3.0 was just sheer luck ;-).

If the globally set things do not change the path position, you got lucky and all works