circuitikz/circuitikz #515: Symbol request: Ferroelectric devices

MayeulC, 2021-04-12:
Hello, and thank you for your work on this awesome library, I've used it for a few years, trough my master's degree and now my PhD.

I'm working on ferroelectric capacitor and transistors, that have a hysteretic mechanism. Ideally, I ould like to use circuitikz for my schematics, but I lack some ferroelectric devices symbols. I'm not sure what your policy on including non-mainstream symbols is, but there seems to be a handful of these already. I am not against sponsoring this work, donating, or putting a bounty on it out of my pocket, for about 15€, but that would need to be agreed before :)

Unfortunately, there doesn't seem to be a universally adopted standard for drawing ferroelectric devices:

1. I often see a thicker transistor gate [with a light color][lightblue]
2. Some instead draw part of a hysteretic curve [on top of the existing symbol][hyst], this is the design I currently favour due to being monochrome and easily distinguishable without contextual information.
3. Lastly, some [combine both][greenhyst].

I'd argue for the second one, since it is unambiguous. The extra loop can be used on top of transistors or capacitors alike.

Ultimately, I would like to see about 5 new symbols:
* fecap
* fefet (N-type, with bulk, without, maybe P-type as well)
  * nfemos
  * nfefet
  * pfemos
  * pfefet

If necessary, I can help a bit with the implementation, though I am a bit short on time, and it might take me a few months for such a "trivial-looking" thing.

Thanks in advance for your comments!

----
# Ferroelectric Capacitor (`fecap`)

The symbol for this one is pretty clear and unambiguous. I am currently using a workaroung inspired by [that document](http://www.elfsoft2000.com/projects/speaker.pdf). I wish there was a simple and documented way to extend the symbol library. I tried modifying the `pgf` code, to no avail. In any case, here is what I use now:

```latex
\newcommand{\ferrocap}[1] % #1 = name from to[lamp,color=white,n=#1]
{ % Based on http://www.elfsoft2000.com/projects/speaker.pdf
	\draw[thick] (#1.se)++(0,-0.1) --++(0.25,0) -- ++(0.2,0.425) -- ++(0.25,0); % How do I do () -- (#1.n)++(dx,dy) ?
}
\begin{circuitikz}
		[scale=1.2]\draw
		(1,0) node[nfet,rotate=-90] (t) {}
		(t.S) to[short, -o] (0,0) node[left] {$BL$}
		(t.G) to[short, -o] +(0,0.3) node[left] {$WL$}
		(t.D) -| (2,0) node[right] {$n$} to[C, *-o, n=XX1] (2,-1) node[left] {$PL$}ç
		(t.bulk) -- +(0,0) node[ground,scale=0.7] {};
		\ferrocap{XX1}
\end{circuitikz}
```
That produces:
[image]

I'm not sure it's the prettiest, it feels a bit cramped too.

# Ferroelectric transistor (`FeFET`).

Both p-FeFET and n-FeFET should be doable, but I've only seen the latter so far. It would be nice to have both variants with and without a bulk connection.

[image]
[image]

---
## Possible variants

I toyed a bit with possible alternatives. You probably have a greater experience with designing symbols than I do.

Here is a possible variant with a smoother curve (I have a svg for it):

[image]

[lightblue]: https://www.researchgate.net/figure/FeFET-crossbar-XNOR-cell-and-interface-circuit-designs_fig1_324712275
[greenhyst]: https://www.researchgate.net/figure/Section-of-the-a-2TNOR-memory-array-with-selector-n-FET-Sel-nFET-and-n-type-FeFET_fig3_340573238
[hyst]: https://www.researchgate.net/figure/Color-online-a-3D-cross-sectional-schematic-view-of-a-MoS2-FeFET-and-b-its-circuit_fig3_284200177

Rmano, 2021-04-12:
Hmmm... the ferrocap would be quite easy to implement (even the "smooth" curve version --- I found it strange that it doesn't show any hysteresis, though) as a separate component. But I am thinking about the possibility of just having the curve as a separate symbol that can be superimposed... maybe some `/.style` trick can ease the usage. 

Let me think a bit about this.

Rmano, 2021-04-12:
BTW:

> % How do I do () -- (#1.n)++(dx,dy) ?

`() -- ([shift={(dx,dy)}] #1.n)`

MayeulC, 2021-04-12:
Thanks for your answer, as well as mentioning the `shift` keyword.

> it would not be so crazy to start having something like external modules before or after...

Hmm, I can imagine an extension interface for users, but I'd be wary of "dependency hell" if circuits are split into multiple packages, I think it would be much easier to maintain this monorepo/monopackage for now, even with a lot more symbols.

> I found it strange that it doesn't show any hysteresis, though

If you say it's inverted, yeah, that's true, I should have mirrored it. If you say that because there is only a step instead of a full loop, that's also true, but I find that adding a full hysteresis cycle makes it a bit less readable:

[image]
[image]

> Let me think a bit about this.

Feel free to! I'd be interested in hearing your opinions on the topic.

> But I am thinking about the possibility of just having the curve as a separate symbol that can be superimposed

That's definitely doable, but it would probably be more complex than `\draw (0,0) to[Cfe] (2,-1) {};` for ferrocaps. Ferroelectric transistors are often regularly mixed with regular transistors too, so that style should be applied on a component-per-compononent basis.

> I would suggest simply go with donations to the TeX/LaTeX community (CTAN, TUG) in general

I can do that :) I was just pointing out that I was not against the idea, you can also use that as a point of data in case you reconsider it in the future.

Rmano, 2021-04-12:
The capacitor with straight lines is quite easy — probably it would be better to have some configurable parameters, but as a starting point:

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\makeatletter
\pgfcircdeclarebipolescaled{capacitors}
{
    \pgfcirc@border@extend@full{1.8}{1}{1.8}{1}
}
{\ctikzvalof{bipoles/capacitor/height}}
{ferrocap}
{\ctikzvalof{bipoles/capacitor/height}}
{\ctikzvalof{bipoles/capacitor/width}}
{
    \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@left}{\pgf@circ@res@up}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@left}{\pgf@circ@res@down}}

    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@right}{\pgf@circ@res@up}}
    \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}{\pgf@circ@res@down}}
    \pgfusepath{draw}
    \pgfscope
        \pgfpathmoveto{\pgfpoint{-1.8\pgf@circ@res@right}{\pgf@circ@res@up}}
        \pgfpathlineto{\pgfpoint{-1.8\pgf@circ@res@right}{0.7\pgf@circ@res@up}}
        \pgfpathlineto{\pgfpoint{1.8\pgf@circ@res@right}{0.7\pgf@circ@res@down}}
        \pgfpathlineto{\pgfpoint{1.8\pgf@circ@res@right}{\pgf@circ@res@down}}
        \pgfusepath{draw}
    \endpgfscope
}

\pgfcirc@activate@bipole@simple{l}{ferrocap}
\pgfcirc@style@to@style{ferrocap}{feC}
\makeatother
\begin{document}
\begin{tikzpicture}[]
    \tikzset{}
    \draw (0,0) to[feC, l=$C_1$, a=a1] ++(3,0) to[feC, l=$C_2$, a=a2] ++(0,-2);
\end{tikzpicture}
\end{document}
```

[image]

Rmano, 2021-04-12:
It will be a bit more complex if the magnetic line is supposed to be rotation-invariant, though. And probably will not be so fancy for vertical capacitors.

MayeulC, 2021-04-12:
Thank you a whole lot for the snippet, that will come in really handy for defining custom symbols, and including new ones while using previous `circuitikz` releases. I'd suggest adding something like this to the manual :)

I am not sure configurable parameters are really necessary, though? If it's *that* easy to add a new variant based on the previous one, people could probably do it if documented?
Conversely, if a radically better symbol comes up, the new version could be upstreamed?

Thanks to your snippet, it should be a bit easier for me to play around with the symbol; I'll ask fellow researchers what they think would be the best shape.

Thanks a lot for your reactivity and thoughtful answers!

Rmano, 2021-04-12:
> I'd suggest adding something like this to the manual :)

Ejem.... https://circuitikz.github.io/circuitikz/circuitikzmanualgit.pdf#d8  😝 

Maybe it's a bit terse, but something is there!

Anyway, for the transistor probably the trick would be to add a flag as I did for Schottky base, https://github.com/circuitikz/circuitikz/commit/a8b1e8033d15158ef6abdc16b24c9c7032f17ffd

MayeulC, 2021-04-12:
Hmm, the snippet you provided doesn't seem to compile for me with lualatex, xelatex, or pdflatex, reportedly due to these lines:
```latex

\pgfcirc@activate@bipole@simple{l}{ferrocap}
\pgfcirc@style@to@style{ferrocap}{feC}
```

Here is part of my log:

```
! Undefined control sequence.
l.29 \pgfcirc@activate@bipole@simple
                                  {l}{ferrocap}
The control sequence at the end of the top line
of your error message was never \def'ed. If you have
misspelled it (e.g., `\hobx'), type `I' and the correct
spelling (e.g., `I\hbox'). Otherwise just continue,
and I'll forget about whatever was undefined.

! LaTeX Error: Missing \begin{document}.

See the LaTeX manual or LaTeX Companion for explanation.
Type  H <return>  for immediate help.
 ...                                              
                                                  
l.29 \pgfcirc@activate@bipole@simple{l
                                    }{ferrocap}
You're in trouble here.  Try typing  <return>  to proceed.
If that doesn't work, type  X <return>  to quit.

! Undefined control sequence.
l.30 \pgfcirc@style@to@style
                          {ferrocap}{feC}
The control sequence at the end of the top line
of your error message was never \def'ed. If you have
misspelled it (e.g., `\hobx'), type `I' and the correct
spelling (e.g., `I\hbox'). Otherwise just continue,
and I'll forget about whatever was undefined.
```

MayeulC, 2021-04-12:
> > I'd suggest adding something like this to the manual :)
> 
> Ejem.... https://circuitikz.github.io/circuitikz/circuitikzmanualgit.pdf#d8 stuck_out_tongue_closed_eyes
> 
> Maybe it's a bit terse, but something is there!

My, my, sorry about that, I guess I haven't read the manual completely since 68e0ed09de8855fc4317bd268ffa04a229d42bb3, so I should do it. I was also looking for "custom" as a keyword. I'll read that part, and try to figure out why I get the above error.

> Anyway, for the transistor probably the trick would be to add a flag as I did for Schottky base, [a8b1e80](https://github.com/circuitikz/circuitikz/commit/a8b1e8033d15158ef6abdc16b24c9c7032f17ffd)

I'll look into that, that really sounds like a neat solution!

Rmano, 2021-04-12:
For that error --- you probably have an old `circuitikz` package. That functions are from 1.2.7 or around there, I do not remember... you can safely upgrade locally, `circuitikz` works with quite ancient Ti*k*Z also (at least for now).

Ah --- found a nice thing. If you add 

```latex
\pgfsetcornersarced{\pgfpoint{0.2\pgf@circ@res@right}{0.2\pgf@circ@res@up}}
```
*after* the `\pgfscope` you will have rounded first magnetization curves (yes, I worked a bit on that, it's on IEEE...):

[image]

MayeulC, 2021-04-12:
Ah, indeed. I'd have expected that Archlinux + texlive would have a rather up-to-date version, but that works with the latest `sty`.

That last one looks really nice, although I think I'd go to about `0.5` instead of `0.7`, since it makes it easier to distinguish if printed small... but that's subjective I guess, I am not that experienced with defining symbols. _(As an aside... it would be very simple to define a symbol for negative capacitance with the characteristic S-shape if it was made `-0.2`, if someone is looking for that...)_

> first magnetization curves 

Well, it's very similar, but that's _electric_ polarization, not magnetic :slightly_smiling_face:
I am not that well versed in the physical aspect of that, mind you. But yeah, they tend to be more rounded in practice, if only because of the various capacitances, and multiple distinct crystalline domains!

Rmano, 2021-04-12:
Yes, with `0.5` is definitely better.

So, do you think that the curve should be rotation-invariant or not? I think that the shape is nice as-is, but I am not sure what the usage is... if I use an invariant polarization curve for the vertical (horizontal path) capacitor the symbol will become very strange... and I see that you did not rotate the curve for transistors.

Rmano, 2021-04-12:
> it would be very simple to define a symbol for negative capacitance with the characteristic S-shape if it was made `-0.2`, if someone is looking for that...)

Well, that could be done with a simple parameter (it's simple because we do not define anchors on the line, otherwise it would be a bit more complex). Both parameters (the 1.8 and the 0.5) can be made adjustable, although to take into account the size changes on the 1.8 would be difficult, if needed (but probably using the average value will do).

MayeulC, 2021-04-12:
I wouldn't worry about negative capacitance yet. It's quite unproven technology, and currently hard to manufacture due to variability. As a result, designing whole circuits with those isn't really a thing yet, and I don't think there is a conventional symbol yet... I was musing the fact that this could be one.

MayeulC, 2021-04-12:
I don't think rotational invariance is needed. Here the goal is just to identify which devices are ferroelectric. Moreover, the hysteresis loop exists in both directions. If you want to be fancy, maybe add a tag that creates a mirrored loop?

Here are a few examples grabbed from ddg images:

> and I see that you did not rotate the curve for transistors.

If you are talking about my `\ferrocap` command, I simply didn't consider the case. If talking of the transistors, I have a bit of trouble following what you mean by rotation invariance and rotating the curve, then, since on the transistors I drew, the X axis became vertical, didn't it?

<details><summary>Here are a few random usage examples grabbed from duckduckgo</summary>

[image] ([source](https://www.aist.go.jp/aist_e/list/latest_research/2008/20080624/20080624.html))

[image]([source](http://wulixb.iphy.ac.cn/article/doi/10.7498/aps.69.20200123))

[image]
([source](https://www.frontiersin.org/articles/10.3389/fnins.2019.00855/full))

[image]
([source](https://www.researchgate.net/figure/Section-of-the-a-2TNOR-memory-array-with-selector-n-FET-Sel-nFET-and-n-type-FeFET_fig3_340573238))
[image]
([source](https://www.researchgate.net/figure/Program-and-erase-operation-of-FeFET-cells-in-AND-architecture-unit-cell-representation_fig5_319405513))

I also found a MEMS circuit that could be replicated with FeFETs:

[image]
([source](https://www.semanticscholar.org/paper/Dynamic-Ternary-Content-Addressable-Memory-Is-and-Zhong-Cao/ecb37340afb585826610e715656e2a9ea4cd2219))

</details>

The last one is interesting. Though it uses a different technology, it points something out: it is quite often interesting to indicate the value stored in a ferroelectric device. That doesn't have to be part of the symbol, but makes it close to the "capacitive sensor" symbol in practice.

Rmano, 2021-04-12:
The idea is that now we have (bear with the horrible hand drawing):
[image]
the red ones (when you rotate the component, the sigmoid rotates rigidly) is what I did and is easy. Keeping the sigmoid fixed while rotating the component is much more difficult.

About the mirror: didn't one of `...to[feC, mirror]...` or `...to[feC, invert]...` work? I can't check now...

MayeulC, 2021-04-12:
> About the mirror: didn't one of ...to[feC, mirror]... or ...to[feC, invert]... work? I can't check now...

Yes, that works fine for the capacitor. Most interesting FETs are perfectly symmetrical, so they're also fine.

As you draw it, red is perfectly fine. I'm slightly confused by your bottom-left quadrant though, here is what I thought you would obtain (C and T lines):

[image]

**Red seems perfectly fine for me**. Although G{2,4,5} look better than the red C counterparts, it is probably a bit complex to determine heuristically (basically, always keep X greater than Y), and could also be left as a variant to pick from.

**I am okay with red for every transistor**.

Rmano, 2021-04-13:
This is what I have now: 

[image]

I will try to open a PR as soon as I can. Tomorrow I will look into transistors.

What do you think about the names (component and anchors?)

MayeulC, 2021-04-14:
That looks quite good to me, thank you a lot for your support!

About the names, they look fine to me. I'm more used to `Cfe` than `feC`, but the latter is probably more correct anyway.

The anchors look quite nice, but I don't really see a use for having so many of these. Generally, I'd indicate the middle of the capacitor, since the dielectric itself stores polarization information. However, the "curve" anchors are probably appropriate for that as well.

> I will try to open a PR as soon as I can. Tomorrow I will look into transistors.

Please take your time if you have other tasks, I am not that much in a hurry :)

Rmano, 2021-04-14:
[image]

Adding extra anchors here will be at least a bit complex (and they will be global, in the sense that if you change the shape parameters mid-circuits they will be wrong), but I can try. Do you think it's worth the effort adding the anchors?

Again, I am open to change the option name if it sounds... itanglish.

Rmano, 2021-04-14:
Ah, btw, tell me if the form of citing you in the source and manual is ok for you.

Rmano, 2021-04-15:
@MayeulC  with the last commit I think that this thing is finished... waiting for your comments/suggestions!

Rmano, 2021-04-16:
@MayeulC  --- did you look at the PR #518 ? What do you think?

Rmano, 2021-04-19:
I will probably merge this today, to make a release to solve #519 as soon as possible. @MayeulC  can you comment, especially about https://github.com/circuitikz/circuitikz/issues/515#issuecomment-819819399

MayeulC, 2021-04-19:
Hello @Rmano, sorry for not commenting earlier, I got quite busy at the end of the week. I apologize for delaying the release.

I am okay with being cited. Thank you for the offer, as you did all of the implementation work!

This looks awesome at a glance, but I would also like to ask my colleagues what they think of the symbols. I'll come back to you ASAP.

Thank you a lot for making them fillable and adding a central anchor. These look really good!

Rmano, 2021-04-19:
Ok, thanks! I am waiting for your feedback.

Rmano, 2021-04-20:
Hi! There was a bug in 1.3.3 and I'd like to release a 1.3.4 early, possibly today. I'd like to add these symbols to it...

MayeulC, 2021-04-20:
Please do release it! I've received rather positive feedback about the symbols, and if there are further cosmetic nits, they could be addressed in a future release, couldn't they?
I would have suggested marking these as "beta" in the manual, however it doesn't look like there will be that many nitpicks, so releasing as-is is probably all right!

Thanks again!

Rmano, 2021-04-20:
Going to merge. Yes, cosmetic changes are acceptable (my policy is to be strictly backward compatible is the changes could "invalidate" a circuit, but not when they are just appearance).  Please reopen or open another issue if there are problems.

You're welcome!

closed 2021-04-20 by Rmano