circuitikz/circuitikz #519: Problemas de compatibilidade com label distance, depois da versão 1.3.3 do circuitikz

JetherReis, 2021-04-18:
I've been using circuitikz since version 0.3.0, and for at least two years, I'm using also the tikz library "circuits.plc.ladder".

Today (april, 4, 2021) I updated circuitikz to version 1.3.3 form 1.3.0, and as soon as I used it, the distance of the labels used in ladder diagrams using the circuits.plc.ladder library, became too big and cannot be controled anymore.

See the files attached in the zip file:

[CircuitikzIssues.zip](https://github.com/circuitikz/circuitikz/files/6332284/CircuitikzIssues.zip)

Jether Reis

Rmano, 2021-04-19:
Hmmm, probably a clash with the `label distance` new switch. Will check. Could you please translate the title into English? Thanks.

Rmano, 2021-04-19:
Probably the solution is just to remove the lines 35 and 36 here: https://github.com/circuitikz/circuitikz/commit/1f7b5989910b0a857052cfa55639814f004955ed#r49692875

JetherReis, 2021-04-19:
> 
> 
> Hmmm, probably a clash with the `label distance` new switch. Will check. Could you please translate the title into English? Thanks.

Sorry, I had started to write the text in Portuguese when I was going to send it, I realized that I should write it in English, but I forgot to translate the title, thanks for the warning.

Jether Reis

Rmano, 2021-04-19:
Hi, no problem! Could you confirm that the fix is working?

closed 2021-04-19 by Rmano

JetherReis, 2021-04-19:
> 
> 
> Hi, no problem! Could you confirm that the fix is working? Please re-open in case of problems.
> Thanks!

Ok it's working, thanks.

Jether Reis

PR #520 (2021-04-19): Fix clash of "label distance" with circuits.*
Although compatibility is not guaranteed, that was an abuse
of the name space. Restrict the labels to the circuitikz/
subdomain only.

Fixes #519