circuitikz/circuitikz #522: Power Electronic symbols

JetherReis, 2021-04-19:
About two years ago I started teaching power electronics, and I missed some symbols related to this area, such as UJT, PUT, GTO, SCS, CGS, LASCR, Opto-TRIAC, Schockley diode etc.

To be able to use the TUJ symbol, I tried to use the nogate option on the "hemt" transistor but the option is not working as the nobase option on the BJT transistors, the same occurring with the "pjfet" and "njfet" transistors.

So, I would like to request the inclusion of these symbols, and while this does not occur, I request the correction in the behavior of the nogate option in the "hemt", "pjfet" and "njfet" transistors.

Here is a file demonstrating these behaviors and the device that I used to circumvent the behavior of the nogate option, in the making of the UJT symbol.

Jether Reis
[SymbolsRequest.zip](https://github.com/circuitikz/circuitikz/files/6336483/SymbolsRequest.zip)

Rmano, 2021-04-19:
I am adding the content of the zip file here, to help reading the issue. Could you please state exactly what are the changes that you propose? It's not so clear to me... 

[image]

```latex
\documentclass[tikz, border=10pt]{standalone}

\usepackage[american,cuteinductors,smartlabels,nooldvoltagedirection,EFvoltages,siunitx]{circuitikz}

%\newcommand{\rfst}[1]{\ensuremath{\scriptstyle\mathrm{#1}}}
\newcommand{\coordxy}[4]{\pgfextractx{#1}{\pgfpointanchor{#3}{#4}} \pgfextracty{#2}{\pgfpointanchor{#3}{#4}}}

\begin{document}

  \begin{circuitikz}
    \newdimen\EGax  \newdimen\EGay
    \newdimen\BAax  \newdimen\BAay
    \newdimen\BBax  \newdimen\BBay
    \newdimen\NBax  \newdimen\NBay
    \newdimen\IUax  \newdimen\IUay
                    \newdimen\Supy
                    \newdimen\Infy

    \newdimen\EGbx  \newdimen\EGby
    \newdimen\BAbx  \newdimen\BAby
    \newdimen\BBbx  \newdimen\BBby

    \newdimen\PWipX \newdimen\PWipY

    \newdimen\TmpAX
    \newdimen\TmpBX
    \newdimen\TmpCX

		\draw

    ( 1.0000 , 0.5000 ) node[hemt,nobase] (T1) {}
    ( 3.0000 , 0.5000 ) node[hemt,nogate] (T2) {}
    ( 6.0000 , 0.5000 ) node[hemt,nobase] (T3) {}
    ( 9.0000 , 0.5000 ) node[pjfet,nogate] (T4) {}
    (12.0000 , 0.5000 ) node[njfet,nogate] (T4) {}
    (15.0000 , 0.5000 ) node[npn,nobase] (T5) {}
    ;
		\coordxy{\EGax}{\EGay}{T1}{G}
		\coordxy{\BAax}{\BAay}{T1}{S}
		\coordxy{\BBax}{\BBay}{T1}{D}
		\coordxy{\NBax}{\NBay}{T1}{nobase}
		\coordxy{\IUax}{\IUay}{T1}{inner up}

		\pgfmathsetlength{\EGay}{\IUay - 0.0750cm}

		\pgfmathsetlength{\Supy}{\EGay + 1.7500cm}
		\pgfmathsetlength{\Infy}{\EGay - 1.7500cm}

		\pgfmathsetlength{\TmpAX}{\EGax - 0.0100cm}
		\pgfmathsetlength{\TmpBX}{\NBax - 0.0150cm}
		\pgfmathsetlength{\TmpCX}{\EGax + 0.2000cm}

    \draw[white,ultra thick] ( \TmpAX , \NBay ) -- ( \TmpBX , \NBay  );
    \draw                    ( \EGax  , \EGay ) -- ( \TmpCX , \EGay  );
    \draw[-latex]            ( \TmpCX , \EGay ) -- ( \NBax  , \NBay  );

    \draw 

    ( \BAax  , \BAay  ) node[below] {$B_1$}
    ( \BBax  , \BBay  ) node[above] {$B_2$}
    ( \EGax  , \EGay  ) node[left ] {$E$}

    ;
		\coordxy{\EGbx}{\EGby}{T2}{G}
		\coordxy{\BAbx}{\BAby}{T2}{S}
		\coordxy{\BBbx}{\BBby}{T2}{D}

    \draw 

    ( \BAbx  , \BAby  ) node[below] {$B_1$}
    ( \BBbx  , \BBby  ) node[above] {$B_2$}
    ( \EGbx  , \EGby  ) node[left ] {$E$}
    ;
  \end{circuitikz}

\end{document}
```

Rmano, 2021-04-19:
PD use only one option of voltage direction --- in your case, only `EFvoltages` is used.

Rmano, 2021-04-19:
Let's see --- which is the symbol that you miss? Just the first one? Sorry, I do not understand your code. If you could add comments it would be nice.

JetherReis, 2021-04-19:
> 
> 
> I am adding the content of the zip file here, to help reading the issue. Could you please state exactly what are the changes that you propose? It's not so clear to me...
> 
> [image]
> 
> ```
> \documentclass[tikz, border=10pt]{standalone}
> 
> \usepackage[american,cuteinductors,smartlabels,nooldvoltagedirection,EFvoltages,siunitx]{circuitikz}
> 
> %\newcommand{\rfst}[1]{\ensuremath{\scriptstyle\mathrm{#1}}}
> \newcommand{\coordxy}[4]{\pgfextractx{#1}{\pgfpointanchor{#3}{#4}} \pgfextracty{#2}{\pgfpointanchor{#3}{#4}}}
> 
> \begin{document}
> 
>   \begin{circuitikz}
>     \newdimen\EGax  \newdimen\EGay
>     \newdimen\BAax  \newdimen\BAay
>     \newdimen\BBax  \newdimen\BBay
>     \newdimen\NBax  \newdimen\NBay
>     \newdimen\IUax  \newdimen\IUay
>                     \newdimen\Supy
>                     \newdimen\Infy
> 
>     \newdimen\EGbx  \newdimen\EGby
>     \newdimen\BAbx  \newdimen\BAby
>     \newdimen\BBbx  \newdimen\BBby
> 
>     \newdimen\PWipX \newdimen\PWipY
> 
>     \newdimen\TmpAX
>     \newdimen\TmpBX
>     \newdimen\TmpCX
> 
> 		\draw
> 
>     ( 1.0000 , 0.5000 ) node[hemt,nobase] (T1) {}
>     ( 3.0000 , 0.5000 ) node[hemt,nogate] (T2) {}
>     ( 6.0000 , 0.5000 ) node[hemt,nobase] (T3) {}
>     ( 9.0000 , 0.5000 ) node[pjfet,nogate] (T4) {}
>     (12.0000 , 0.5000 ) node[njfet,nogate] (T4) {}
>     (15.0000 , 0.5000 ) node[npn,nobase] (T5) {}
>     ;
> 		\coordxy{\EGax}{\EGay}{T1}{G}
> 		\coordxy{\BAax}{\BAay}{T1}{S}
> 		\coordxy{\BBax}{\BBay}{T1}{D}
> 		\coordxy{\NBax}{\NBay}{T1}{nobase}
> 		\coordxy{\IUax}{\IUay}{T1}{inner up}
> 
> 
> 		\pgfmathsetlength{\EGay}{\IUay - 0.0750cm}
> 
> 		\pgfmathsetlength{\Supy}{\EGay + 1.7500cm}
> 		\pgfmathsetlength{\Infy}{\EGay - 1.7500cm}
> 
> 		\pgfmathsetlength{\TmpAX}{\EGax - 0.0100cm}
> 		\pgfmathsetlength{\TmpBX}{\NBax - 0.0150cm}
> 		\pgfmathsetlength{\TmpCX}{\EGax + 0.2000cm}
> 
>     \draw[white,ultra thick] ( \TmpAX , \NBay ) -- ( \TmpBX , \NBay  );
>     \draw                    ( \EGax  , \EGay ) -- ( \TmpCX , \EGay  );
>     \draw[-latex]            ( \TmpCX , \EGay ) -- ( \NBax  , \NBay  );
> 
>     \draw 
> 
>     ( \BAax  , \BAay  ) node[below] {$B_1$}
>     ( \BBax  , \BBay  ) node[above] {$B_2$}
>     ( \EGax  , \EGay  ) node[left ] {$E$}
> 
>     ;
> 		\coordxy{\EGbx}{\EGby}{T2}{G}
> 		\coordxy{\BAbx}{\BAby}{T2}{S}
> 		\coordxy{\BBbx}{\BBby}{T2}{D}
> 
>     \draw 
> 
>     ( \BAbx  , \BAby  ) node[below] {$B_1$}
>     ( \BBbx  , \BBby  ) node[above] {$B_2$}
>     ( \EGbx  , \EGby  ) node[left ] {$E$}
>     ;
>   \end{circuitikz}
> 
> \end{document}
> ```

These drawings refer to the artifice I made to specifically address the lack of the UJT symbol.
When trying to use the "nogate" option on the "hemt" transistor to redesign the emitter terminal, I realized that this option is not working as it is on the BJT transistor.
The first drawing is the symbol I drew by erasing (covering the gate connection with a white line), and redrawing the symbol of the emitter terminal, the others are attempts to eliminate the gate terminal with the nogate option, for comparison, the last it is the successful result of the nobase option on a BJT transistor.

Jether Reis

Rmano, 2021-04-19:
Ah, ok. I will fix the `nobase/nogate` on `hemt` which is clearly useful (once you have this, removing the base terminals on `n/pjfet` is not useful, they will be the same circuit). Anyway, maybe you can consider using this to cancel out/add the UJT terminal:

```latex
\begin{circuitikz}
        \node [hemt, circuitikz/transistors/thickness=2](H) at (0,0) {};
        % nobase does bno work with hemt --- will fix
        % white out base connection
        \draw[white, line cap=none] ([xshift=-1\pgflinewidth]H.nobase) -- ([xshift=-0.5\pgflinewidth]H.G);
        \draw[Triangle-] (H.nobase) -- ++(-0.25,0.25) -- ++(-0.25,0) coordinate(H-E);
        \node[above] at (H.D) {$B_2$};
        \node[below] at (H.S) {$B_1$};
        \node[left] at (H-E) {$E$};
\end{circuitikz}
```

[image]

Rmano, 2021-04-19:
Anyway, I see the usefulness of a UJT symbol, will try to add it. Also GTO seems quite easy modifying `thyristor`.

JetherReis, 2021-04-19:
About the other symbols, here are some that I got on the internet:

PUT:
[image]

GTO:

[image]
[image]

MCT:

[image]

Others:

[image]

Rmano, 2021-04-19:
Well, they are a lot! 

I will be happily accept PRs... 😉. For now I will try to fix the nogate thing and try to add the ujt. Couple questions:

1. Am I right that the UJT comes with only that arrow direction? 
2.  Which symbol for the GTO is the standard one? I see 3 or 4 different ones...

JetherReis, 2021-04-20:
> 
> 
> Well, they are a lot!
> 
> I will be happily accept PRs... 😉. For now I will try to fix the nogate thing and try to add the ujt. Couple questions:
> 
>     1. Am I right that the UJT comes with only that arrow direction? No: see https://en.m.wikipedia.org/wiki/Unijunction_transistor although symbols are different and uglier
> 
>     2. Which symbol for the GTO is the standard one? I see 3 or 4 different ones...

Rashid in Power Electric Handbook 2001, p. 417, uses the UJT symbol I showed, see circuit below:
[image]

The GTO symbol, I heve found in some datasheet one of these:

[image]
e.g. 5SGA 06D4502 and H0500KC20#/H0500KC25#

Rmano, 2021-04-22:
Ok, UJTs are in. I will add GTO, probably over the week-end. Please have a look at #528

Rmano, 2021-04-22:
Modifying the standard Thyristor I have this thing:

[image]

which I find quite horrible. I am thinking about it --- maybe I should make a different type of Thyristor, with better proportion, call it SCR and derive the 3 formats of GTO I'd like to support from it. 

Thoughts?

Rmano, 2021-04-23:
Much better now. I have added a style option (more or less, I have to do a bit of cleaning up)

[image]

The arrows are configurable in the plain GTO symbol (normally there are no arrows)

Rmano, 2021-04-23:
In
[image]
the horizontal bar must have the wire thickness or the device border thickness?

Rmano, 2021-04-23:
Ok, I think PR #528 is there. I will wait a bit and then I'll merge it, which will close this issue.
Please feel free to open other issues if you think there is a need for more components, one at a time or at least one family at a time, with some reference about official shape if it exists.

PR #528 (2021-04-25): Add UJT and GTO symbols
Fixes #522 (when finished)

For now UJT are in.

[image]
closed 2021-04-25 by Rmano