circuitikz/circuitikz #524: PV Module

andre-a-alves, 2021-04-19:
Firstly, thank you for building and maintaining this repository. I use it all the time for school and work, and I have not found anything that comes close when it comes to being able to make exactly what I am looking for.

While there is a component for pvsource already included in the package, the common PV Module symbol depicted below is not included. Since it is so simple, I have been lazy and have just been drawing it with Tikz whenever I need it.
[image]

Later this week or more likely next week, I would like to add this symbol as a contribution. It seems like a rather straightforward symbol to add, and having dabbled a little in TeX, I think making it would also be a good experience for me. I propose adding it as a path-style component and including it in 3.8.6 Special Sources in the manual.

Rmano, 2021-04-20:
Ok, fine with me. Notice that this is a very strange symbol for a source --- the ideal thing would be to find a formal reference for it (as in a standard, datasheets, etc.). The other important thing is that, if it is in the `sources` class, it is important to get the relative size (say, with a `battery` or `dcV`) right. 
I am assigning this issue to you.

andre-a-alves, 2021-04-20:
Thanks, @Rmano 

I'll see what I can do about finding the formal reference in a standard somewhere. In the meantime, here are some quick links to a few academic papers that employ the symbol (I completely recognize and agree it's a strange symbol, and I didn't trust it myself until I kept stumbling across it):
* [Efficiency calculation of inverter for PV applications Using MATLAB and SPICE](https://doi.org/10.1109/OPTIM.2017.7975033)
* [Numerical simulation of current-voltage characteristics of photovoltaic systems with shaded solar cells](https://doi.org/10.1016/0038-092X(96)00006-0)
* [Photovoltaic Ground Fault Simulations Using SPICE](https://scholar.google.com/scholar_url?url=https://www.osti.gov/servlets/purl/1116071&hl=en&sa=T&oi=gsb-ggp&ct=res&cd=0&d=7065118478337734951&ei=EoN-YK7SLILCsQKigaGgDA&scisig=AAGBfm1E2sA22vtgwK-h0O8BPGRABoazqg)

It has also appeared in some of my lectures at university, and is used by SMA in [this](https://manuals.sma.de/SBx1SPUS40/en-US/391466379.html) section of one of their [manuals](https://manuals.sma.de/SBx1SPUS40/en-US/index.html).

Regarding the sizing, it is usually a little longer than a standard source symbol, so I'll base the proportionality to other sources' widths. No matter what, I'll test it out to ensure functionality and provide an example of its use.

Rmano, 2021-04-20:
You can probably start from either the `generic` shape or from a `block`. For the internal number, if you do not want it to rotate with the element, I suggest you look at the `meter` element.

closed 2021-05-01 by Rmano

Rmano, 2025-12-14:
Hi, how is it used? Can you give a "all over the place" example? (But please do not use a closed issue, open a new one). `circuitikz` has two symbols, one is the André Alves one you see above, and; 

[image]

The block you showed can easily be built, anyway...

PR #537 (2021-05-01): Solar modules
This PR adds a photovoltaic module as a special source and Closes #524.

I could not find a formal reference for the component other than inclusion in academic papers as mentioned in the issue and inclusion in a few training manuals.