circuitikz/circuitikz #529: Add DC/DC Converter

Pratched, 2021-04-26:
Hey folks, cheers for the good work. Great stuff to work with. Is it possible to simply add a DC/DC-converter shape? I assume its basically copy&paste as interchanging the sine with another equal sign as you did with `sdcac` and `sacdc`?

Rmano, 2021-04-27:
Hi! Wouldn't ` \draw (0,0) to [twoportsplit, t1={$=$}, t2={$=$}] ++(3,0);` fits the bill?

[image]

Pratched, 2021-04-27:
Well yes it does. It is a suitable easy solution, that I havnt thought of. Cheers!
However, I still would suggest to add a DC/DC converter in the future since it is as often used as an DC/AC and would be easy to find for newbies. 
If necessary I would get rid of an DC/AC or AC/DC since they are basically the same when using `rotate`. But this is just one electrical engineer view rather tikz developer.

Rmano, 2021-04-27:
It is true that the symbol is slightly different though. I'll see if I can add it easily. 
The problem with `rotate` is that then you have to remember that the anchors rotate, too ;-)...
In the meantime you can define a style:

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\tikzset{sdcdc/.style={twoportsplit, t1={$=$}, t2={$=$}}}
\begin{tikzpicture}[]
    \draw (0,0) to [sdcdc] ++(3,0);
\end{tikzpicture}
\end{document}
```

Pratched, 2021-04-27:
Fair enough. Havnt thought about that either :)
Maybe a little note around sdcac reading "see twoportsplit for further designs" could also do the trick. Whereas I realise once you know/read it, it is rather obvious...

However, cheers for the quick response and help :)

Rmano, 2021-04-27:
It was easy, and clearly, the block is fitting better with the other ones. Thanks!

Rmano, 2021-04-27:
Please feel free to re-open if needed!

closed 2021-04-27 by Rmano