circuitikz/circuitikz #534: CFD: simple (very!) subcircuits...

Rmano, 2021-04-29:
I have decided to bite the bullet and try to create a simple subcircuit facility for `circuitikz`. The idea is here:

[image]

You can read more here: 
[subcircuitikz.pdf](https://github.com/circuitikz/circuitikz/files/6400572/subcircuitikz.pdf)

What do you think? (I will post the implementation in answer to this). @mredaelli @der-stefan @sistlind ?

Rmano, 2021-04-29:
The implementation is quite easy, once I wrapped my head around `##` and `\expandafter`s  ...  the idea is taken by my answer here: https://tex.stackexchange.com/questions/583825/define-custom-component-from-circuit 

```latex
\makeatletter
\newbox\ctikz@scratchbox
\long\def\ctikzsubcircuitdef#1#2#3{%
    \expandafter\gdef\csname #1@Anchor\endcsname{}
    \expandafter\gdef\csname #1@setanchors\endcsname{
        \setbox\ctikz@scratchbox=\hbox{%
        \begin{circuitikz}
        \draw (0,0) \csname#1\endcsname{T-#1}{};
        \foreach [count=\i] \anchor in {#2}
        % reference anchor is -center
        \draw (0,{2-\i/2}) let \p1 = ($(T-#1-reference)-(T-#1-\anchor)$) in
            node[right]{\anchor: \x1,\y1 \expandafter\xdef\csname #1@Anchor\anchor\endcsname{++(\x1,\y1)}};
        \end{circuitikz}
        }
    }
    \expandafter\gdef\csname#1\endcsname##1##2{\csname#1aux\endcsname{##1}{\csname #1@Anchor##2\endcsname}}
    \expandafter\gdef\csname#1aux\endcsname##1##2{%
    % move to the anchor
    ##2
    % reference anchor should be -reference
    coordinate (##1-reference)
    #3
    }
}
\long\def\ctikzsubcircuitactivate#1{\csname #1@setanchors\endcsname}
\makeatother

```

Rmano, 2021-05-01:
Ok, I merged in #536 ... let's see what happens!

Skillmon, 2021-07-03:
A few comments on the code shown above:

- you don't need the `#1@aux` function. Instead you could just use the code directly: `\expandafter\gdef\csname#1\endcsname##1##2{\csname #1@Anchor##2\endcsname coordinate(##1-reference)#3}`
- `\ctikzsubcircuitdef` should be defined `\protected`
- `\ctikzsubcircuitdef` has a bunch of spurious spaces which end up in output (not those inside the stuff that's only executed inside of `tikzpicture`/`circuitikz`). That does no harm when used strictly inside the preamble/vertical mode, but when someone uses it in horizontal mode those will show up.

A bit more customisability could be achieved with some sort of key=value interface for a subcircuit. I'll prototype something.

Rmano, 2021-07-03:
Yes, about the third point: I've been lazy... I will try to fix it asap.

For the other options, it would be nice to add parameters (like names of labels and so on), but I do not think that this could go too far --- the problem is that the way I manage the "pseudo anchors" will not work with parameters that change the size of the subcircuit. And the possibility to use the pseudo-anchor to place the subcircuit is the only feature that makes the subcircuit worth against a simple "`\def`" ;-)

Rmano, 2022-02-21:
Closing, it seems that the simple subcircuits are sufficient for now.

closed 2022-02-21 by Rmano