circuitikz/circuitikz #566: Capacitors geometry parameters are a mess

Rmano, 2021-07-07:
In the sense that:
 
* `C`, `ferroC`, `sC` share the same plate distance and is called `width`; `vC` width includes the arrow and the plate distance is different.
* height is 0.7 normally, but 0.6 for `eC` and 0.8 for `piezo`
* etc etc

I noticed this looking at the comment of @muzimuzhi on https://tex.stackexchange.com/questions/603986/tighter-capacitor-plates-in-circuitikz

I can:

1.  let everything as is and document it (I like not...)
2.  change things so that all the capacitors use the same parameters and normalize them; leave the old parameters in and make all of them (minus the simple `bipoles/capacitor/*` ones)  a nop (to avoid breaking old circuits messing with them, although they are internal and not documented)
3. like 2, but trying to define with `.code` all the old parameters to make them work (for some will be easy, for other not-so-much or plainly too complex).

I tend to think that a mix between 2 and 3 would be the best option. And make all heights equal to 0.7. (one cal always use local definition or style to have different-looking capacitors).

What do people think? 

@mredaelli @der-stefan @sistlind @Skillmon ?

Skillmon, 2021-07-07:
Imho, a way to set all at once would be nice, *but* I'd keep them as separate keys to also allow greater flexibility (distances which might look good for `C` might look too cramped for a `vC` or `sC`). So this would boil down to:

- keep the keys as they currently are and document them
- additionally, define a general key that sets all capacitor widths at once

This doesn't require any rewrite, gives greater flexibility and adds the convenience key. It would require the most documentation work of all options though.

Skillmon, 2021-07-07:
And if it doesn't take too much effort, I'd also decouple the capacitor width from the arrow angle in `vC` (but that would just be a nice to have extra).

Rmano, 2021-07-07:
> * keep the keys as they currently are and document them
> 
> * additionally, define a general key that sets all capacitor widths at once

Yes, this is what I did for the resistors (if only I'd know how to link code snippets here...). The only problem is that the *defaults* are different for the individual keys --- but that could be secondary; just leave the defaults differents and accept the fact that if you use the global key all of them will be set equal.

(About `vC`: they already are. I have just to do some simple sum when applying the global value to the individual keys, I hope that \numexpr can work in keys assignment...

Rmano, 2021-07-11:
I did a mix of the two approaches. There was really no sense in maintaining (undocumented) parameters and paper over them, so I changed them *and* added an official way to change the styles, even on a component-per-component base via styes. I think it's nice now... and symmetric with the existing options for resistors and inductors.

closed 2021-07-11 by Rmano