circuitikz/circuitikz #567: Label and anchor info for advanced voltages

Rmano, 2021-07-09:
As noticed by @judober (see issue #448), when using advanced voltages we can say:

1.  `v`, `i`, `f` : no arrows/symbols are added, and now voltage/current/flow label is generated
2. `v=~`: the arrows/symbols are generated, and a series of anchors and info (rotation, etc) are generated for the blank label.

There is no provision for 

3. not having arrow/symbols *but* having the information for the label placement generated (that could be useful if, for example, you want just to change the symbol but you are happy with the labels as they are). 

I think that it would be nice to have an option to _not_ draw the symbols (but generate the anchors) and placing the labels (that could be a blank one).

I am open to suggestions for a syntax; I do not want to create all the `v_^` combinations, so I think that an extra key like `suppress v` or `!v` to add to suppress drawing could be ok.

Rmano, 2021-07-14:
@judober  you can play with it if you want with: 

[circuitikzgit.sty.zip](https://github.com/circuitikz/circuitikz/files/6814824/circuitikzgit.sty.zip)

But I'll do a couple of tests and then I hope to merge (and release) soon

Rmano, 2021-07-14:
> Maybe something like `suppress arrows` as the v-label does not have to be blank. Then one could just use `v=~` as desired.

And yes, I did exactly that ;-)

Rmano, 2021-07-14:
```latex
\newcommand{\iarronly}[1]{% name
    \node [currarrow, color=red, anchor=center,
    rotate=\ctikzgetdirection{#1-Iarrow}] at (#1-Ipos) {};
}
\newcommand{\varronly}[1]{% name
    \draw [color=blue] (#1-Vfrom) .. controls (#1-Vcont1)
        and (#1-Vcont2)..  (#1-Vto) node [currarrow,
        sloped, anchor=tip, allow upside down,pos=1]{};
}
\begin{circuitikz}[]
    \ctikzset{!vi/.style={no v symbols, no i symbols}}
    \ctikzset{bipole voltage style/.style={color=blue},
        bipole current style/.style={color=red}}
    \draw (120:6) to[R, *-,  name=R, v^=$v_R$, !vi]
        (120:3) to[short, i=$i_R$, name=SR, !vi] (0,0);
    \draw (180:4) to[L, o-, l=$L$, name=L2, v=$v_L$, !vi]
        (180:2) to[V, -*, name=V2, v_=$v_2$, !vi] (0:0);
    \iarronly{SR}\varronly{R}\varronly{L2}\varronly{V2}
\end{circuitikz}
```

[image]

Rmano, 2021-07-14:
I am not sure if adding the `!vif` shorthand could be useful for everybody... hmmm...

Rmano, 2021-07-14:
I merged it because I want to release soon. Please @judober  feel free to comment (others too ;-)!), I can always change things (before releasing)

judober, 2021-07-14:
Wow, works like a charm. 
It also removes the need for custom colored label printing when I accept different rotation for label and vlabel (example below).

```Latex
\documentclass[class=report, 11pt,a4paper,tikz, border=4pt]{standalone}
% Version from https://github.com/circuitikz/circuitikz/issues/567
\usepackage[european, straightvoltages, americaninductors, betterproportions, nooldvoltagedirection]{circuitikzgit}[
]

\ctikzset{!vi/.style={no v symbols, no i symbols}}
\ctikzset{bipole voltage style/.style={color=blue},
	bipole current style/.style={color=red}}

\newcommand{\iarronly}[1]{% name
    \node [currarrow, color=red, anchor=center,
    rotate=\ctikzgetdirection{#1-Iarrow}] at (#1-Ipos) {};
}
\newcommand{\varronly}[1]{% name
    \draw [color=blue] (#1-Vfrom) .. controls (#1-Vcont1)
        and (#1-Vcont2)..  (#1-Vto) node [currarrow,
        sloped, anchor=tip, allow upside down,pos=1]{};
}

\begin{document}
	\begin{tikzpicture}
		\draw (120:6) to[R, *-,  name=R, v_<=$v_R$, !vi, l=R] (120:3) to[short, i=$i_R$, name=SR, !vi] (0,0);
		\draw (180:4) to[L, o-, l=$L$, name=L2, v=$v_L$, !vi] (180:2) to[V, -*, name=V2, v_=$v_2$, !vi] (0:0);
		\iarronly{SR}
		\varronly{R}
		\varronly{L2}
		\varronly{V2}
	\end{tikzpicture}
\end{document}
```

[image]

closed 2021-07-14 by Rmano