circuitikz/circuitikz #569: Bug with +() coordinates

Rmano, 2021-07-09:
See https://tex.stackexchange.com/questions/604366/how-do-i-fix-incorrect-label-placement

```latex
\documentclass[border=0.5cm]{standalone}
% \usepackage[oldvoltagedirection]{circuitikz}[=v0.8.3]
\usepackage[oldvoltagedirection]{circuitikz}

\begin{document}
\pgfcircversion
\begin{circuitikz}%[circuitikz/label/align=rotate]
\draw (0,0) to [L,l=$L_{nn}$] +(5,0);
\end{circuitikz}
\end{document}
```

gives:

[image]

Rmano, 2021-07-09:
This change to `pgfcirclabel.tex`: 
[pgfcirclabel.zip](https://github.com/circuitikz/circuitikz/files/6793262/pgfcirclabel.zip)

seems to fix this:

[image]

ctmalan, 2021-07-09:
I am busy installing TexLive2021 on a new machine - will report tomorrow.

ctmalan, 2021-07-10:
Can confirm the error on:  This is pdfTeX, Version 3.141592653-2.6-1.40.23 (TeX Live 2021/W32TeX) (preloaded format=pdflatex 2021.7.10)  10 JUL 2021 10:39

Can also confirm that the upgated pgfcirclabel.tex fixed the issue.

Rmano, 2021-07-10:
Can you also please give the Pgf/Tikz version for failing and not failing? You can see them with `\pgfversion`...

Thanks for the test, I'll merge the fix as soon as I can (and do a release)

Rmano, 2021-07-10:
No need, I checked it with overleaf. This is a nasty combination of a bug fix on TikZ and an old change in circuitikz that worked thanks to the bug in TikZ... will merge soon.

PR #570 (2021-07-11): Fix the problem with `+()` style relative coordinates
This has been created (I think) by a combination of the change
in 6c1ca8c6345d4000ad42ca82c0cc05662d86e71c (yes!) and the change
https://github.com/pgf-tikz/pgf/commit/95d8a97e200a83c91ca64977f6856897f7cea6d6
(probably).

See
https://tex.stackexchange.com/questions/604366/how-do-i-fix-incorrect-label-placement

Thanks to https://tex.stackexchange.com/users/27312/skvery for noticing

Fixes #569
closed 2021-07-11 by Rmano