circuitikz/circuitikz #571: twoportsplitshape doesn't support t1 and t2 parameters

bodokaiser, 2021-07-10:
The following

```latex
\begin{circuitikz}
    \draw (0,0) node[twoportsplitshape, t1=foo, t2=bar](a){};
\end{circuitikz}
```

outputs

```txt
Package pgfkeys Error: I do not know the key '/tikz/t2', to which you passed '90', and I am going to ignore it. Perhaps you misspelled it.
```

while

```latex
\begin{circuitikz}
    \draw (0,0) to[twoportsplitshape, t1=foo, t2=bar] (4,0);
    \draw (0,2) node[twoportshape, t=foobar](a){};
\end{circuitikz}
```

works.

The `twoportsplit` is mentioned on [p. 81][1]

[1]: https://ftp.tu-chemnitz.de/pub/tex/graphics/pgf/contrib/circuitikz/doc/circuitikzmanual.pdf

Rmano, 2021-07-10:
Yes, it's supposed to fail. When you're using

     \draw (0,1) to[twoportsplit, t1=foo, t2=bar] (4,1);  % NOT  twoportsplitshape, it works by sheer chance!

the default namespace is `/tikz/circuitikz` (set by the `twoportsplit` style which is a `circuitikz` one). When using a `node`,  the namespace is the base `/tikz`). So you should use:

    \draw (0,0) node[twoportsplitshape, circuitikz/t1=foo, circuitikz/t2=bar](a){};

which works. 

The fact that the `t` keys also exist in the base `/tikz` shape is a very dangerous thing, it can clash with more or less everything; I would like to remove it but, you know, backward compatibility...

In your file, if you want, you can add 

    \tikzset{t1/.code={\ctikzset{t1=#1}},t2/.code={\ctikzset{t2=#1}}}

in your preamble and then 

    \draw (0,0) node[twoportsplitshape, t1=foo, t2=bar](a){};

will also work.

bodokaiser, 2021-07-10:
Thank you very much!

closed 2021-07-10 by bodokaiser