circuitikz/circuitikz #580: node[x]{}; produces nothing with some components

ctmalan, 2021-07-27:
`\draw (-1,1) node[op amp](R){};` draws a op amp
`\draw (-1,1) node[ooosource](Q){};` draws nothing

Rmano, 2021-07-27:
User error:

[image]

`ooosource` is a path-style name, so you have to use:

`\draw (-1,1) node[ooosourceshape](Q){}` ;

ctmalan, 2021-07-29:
Dear Romano,

I find it difficult to reproduce the following diagram with your suggestion.

[image: image.png]

Here is my current hack:

{\small\centering
\hfill\begin{circuitikz}[scale=0.95]
%\matrix[column sep=1cm,ampersand replacement=\&]{\begin{scope}
 \draw[thick,] (-2,3) to[short,o-o]
(-2,-1)node[anchor=north]{load\textsubscript{A}};
   \draw[thick] (-1,3) to[short,o-o]
(-1,-1)node[anchor=north]{load\textsubscript{B}};
\draw[white] (0,1)to[very thin,ooosource,n=Q] (-0.5,1);

\draw (Q.left)+(0.07,0)node [anchor=east] {\scriptsize$\sum$};
\draw (-1,0)circle [radius=0.25] ++(0.25,0) -| (Q.sec3);
\draw (-2,2)circle [radius=0.25] ++(0.25,0) -| (Q.tert3);
\draw(Q.left) -- (1,1);
{% square watthourmeter a
    \draw (1,1) to [twoport,  name=e, %i_=$\I$,
    t=\shortstack{{$E_\text{sum}$}\\~\\\si{VAh}} ] ++(1,0);
    \draw[very thick] (e.west) -- (e.east);
    \comment{%link
    \draw ($(e.north)+(0,0.125)$) -- ($(e.center)+(0,+0.75)$) -|
    ($(e.center)+(-1.5,0.0)$) to[short, -o] ++(0,-2);
      }
    \draw [thick] (e.north east) -- ++(0,0.125) -| (e.north west);
    \comment{\draw (e.south) -- ($(e.center)-(0,1)$)-- (2.5,2) to
[short,-*] (2.5,1.5);
% \draw ($(e.center)-(0,2)$) to [open, v^>=$\U$,] ($(e.center)+(-0.75,0)$);
    }}
   %summation
    \draw [dashed, semithick] (2,1) -- ++(0.6,0)
++(0.4,0)circle[,semithick,radius=0.4] node{$\sum$};

\end{circuitikz}\hfill}

Rmano, 2021-07-29:
Sorry, I can't see your post above correctly. I suspected you answered by mail and often images and formatting is destroyed...

Could you please post a minimal, *complete* example that I can compile?

Rmano, 2021-07-29:
For example, what is `\comment`?

Rmano, 2021-07-29:
This is what I have: 

[image]

what should

     \draw[white] (0,1)to[very thin,ooosource,n=Q] (-0.5,1);

do? you can't pass a line width in a `to` command (it is not and `edge`!); and if you draw it in white...

Rmano, 2021-07-29:
I suspect that you want to put your `ooosourceshape` like that:

```
\documentclass[12pt,a4paper]{article}
\usepackage[T1]{fontenc}
\usepackage{amsmath}
\usepackage[siunitx]{circuitikz}

\begin{document}
\begin{circuitikz}[scale=0.95]
    \draw[thick,] (-2,3) to[short,o-o]
    (-2,-1) node[anchor=north]{load\textsubscript{A}};
    \draw[thick] (-1,3) to[short,o-o]
    (-1,-1) node[anchor=north]{load\textsubscript{B}};

    \draw[very thin]  (-0.25, 1) node[ooosourceshape, xscale=-1](Q){};

    \draw (Q.left)+(0.07,0)node [anchor=east] {\scriptsize$\sum$};
    \draw (-1,0)circle [radius=0.25] ++(0.25,0) -| (Q.tert3);
    \draw (-2,2)circle [radius=0.25] ++(0.25,0) -| (Q.sec3);
    \draw(Q.left) -- (1,1);

\end{circuitikz}
\end{document}
```

[image]

Rmano, 2021-07-29:
Exploring the code for this problem I found a bug for the `ooosource` thing --- anchors were wrong when using `sources/scale`, see #581

ctmalan, 2021-07-29:
Dear Romano,

Thanks for solving this!

\newcommand{\comment}[1]{%
}%
\newcommand{\uncomment}[1]{#1%
}%

I use this for debugging, much better than \usepackage{comment}.

Kind regards,

c-:

closed 2021-07-29 by Rmano