circuitikz/circuitikz #586: Feature request: Introduce a new ground symbol

AndreaDiPietro92, 2021-10-16:
It would be interesting to introduce a new "ground" symbol defined as follows (code find on internet):

```
\def\Ground(#1)#2{%customized ground
			\begin{scope}[shift={(#1)}]
				%Component Shape
				\draw[line width = 2pt, line cap=round]
				(0,0) coordinate (#2 GND)++(-0.3,0)--++(0.6,0)
				(0,-0.15)++(-0.2,0)--++(0.4,0)
				(0,-0.3)++(-0.1,0)--++(0.2,0);  
			\end{scope}
		}
```
[image]

Rmano, 2021-10-16:
Hmmm... what is the difference with `tlground`? Just the line cap? 

[image]

Rmano, 2021-10-16:
You can do this:

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{circuitikz}
    \node [tlground, line cap=round]{};
\end{circuitikz}
\end{document}
```

[image]

and you can define your style:

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\tikzset{gnd/.style={tlground, line cap=round}}
\begin{document}
\begin{circuitikz}
    \node [gnd]{};
\end{circuitikz}
\end{document}
```

Rmano, 2021-10-16:
No problem! Happy to help.