circuitikz/circuitikz #590: ctikzset bipole/is voltage

judober, 2021-10-20:
I found the option `bipole/is voltage = false` in the manual (e.g. page 173) in order to unify the voltage arrows of sources. Is it possible to set this globally with something like `\ctikzset{bipole/is voltage =false}` for all voltage sources?

Rmano, 2021-10-21:
Hmm, really no. The best you can do is to define a style, like:

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\tikzset{%
    battery NS/.style={battery={#1}, bipole/is voltage=false},
    V NS/.style={V={#1}, bipole/is voltage=false},
    sV NS/.style={sV={#1}, bipole/is voltage=false},
}
\begin{tikzpicture}[]
    \draw (0,2) to [battery=vb, bipole/is voltage=false] ++(2,0) to[V=v, bipole/is voltage=false] ++(2,0) to[sV=sV, bipole/is voltage=false] ++(2,0) to[R, v=vR] ++(2,0);
    \draw (0,0) to [battery=vb] ++(2,0) to[V=v] ++(2,0) to[sV=sV] ++(2,0) to[R, v=vR] ++(2,0);
    \draw (0,-2) to [battery NS=vb] ++(2,0) to[V NS=v] ++(2,0) to[sV NS=sV] ++(2,0) to[R, v=vR] ++(2,0);
\end{tikzpicture}
\end{document}
```

[image]

...which made me find an inconsistency (if I call the last line with `...to[V NS, v=V]...` I have the voltage reversed in direction and position; which is probably a bug of `is voltage` but that I would probably never touch, giving that it is probably here since the start of times.

judober, 2021-11-03:
Thanks for implementing this!
I am not sure if I will end up using it, as it is quite disruptive as you stated.
However, as I see it, the main problem is the american voltage and current source which have an inherent direction that others don't. That is actually not a problem in my case.

For context, why I wanted that feature:
I had to draw a voltage and a current source side by side and the difference in the voltage arrows was quite annoying. Then I discovered `bipole/is voltage = false` which solved the problem. Then all other voltage source voltage arrows looked different and I thought a global switch would be helpful.

Rmano, 2021-11-03:
Well, maybe somebody will prefer to set it and manually adjust the cases where it breaks (what you mentioned and also batteries). At least, it has been an occasion to better describe the thing in the manual, and I think that *that* was needed anyway... 

I release 1.4.4 with the feature included, by the way, a couple of days ago. 

Thanks for reporting --- it is nice to know that somebody uses the package! ;-)