circuitikz/circuitikz #596: ConTeXt compatibility

TheTeXnician, 2021-12-03:
As circuitikz strives for format independence, ConTeXt compatibility should remain intact.

[On the ConTeXt mailing list](https://mailman.ntg.nl/pipermail/ntg-context/2021/104379.html) it was reported that `pgfcircutils.tex` violates format independence:

https://github.com/circuitikz/circuitikz/blob/0eb40ea2c9a54157e379db7789cad790e3569b64/tex/pgfcircutils.tex#L107-L113
https://github.com/circuitikz/circuitikz/blob/0eb40ea2c9a54157e379db7789cad790e3569b64/tex/pgfcircutils.tex#L387-L396

This prevents ConTeXt from even compiling circuitikz documents. Probably plain TeX TikZ syntax `\scope…\endscope` should be used.

Rmano, 2021-12-03:
I merged #597 --- please comment if it is not fixing the problem. Thanks!

TheTeXnician, 2021-12-03:
I think it's definitely right. I still get an error about `\pgfutil@protected` but maybe my test setup is messed up.

Aditya Mahajan commented [just a few minutes ago](https://mailman.ntg.nl/pipermail/ntg-context/2021/104388.html) that there might be issues with keyword arguments. I cannot test this but maybe there is more to do if anyone does a bit more tracing :wink:

Rmano, 2021-12-03:
Ok, please follow-up as needed. I will.need help with ConTeXt though...

Skillmon, 2021-12-04:
If `\pgfutil@protected` is throwing an undefined-error I suggest reporting the issue to the PGF/Ti*k*Z maintainers. At least the `parser` module (which admittedly is seldom used) will not load without it as well, and it is also used in the pgf module `calendar`, afaik. In the meantime one could live with the ugly code (similar to the fix for `\pgfutil@unexpanded` and `\pgfutil@ifx`):

```
\ifx\pgfutil@protected\pgf@circ@undefined
  \ifpgfutil@format@is@context
    \let\pgfutil@unexpanded\normalprotected
  \else
    \let\pgfutil@unexpanded\protected
  \fi
\fi
```

Rmano, 2021-12-04:
I would not mind adding it, if needed... @TheTeXnician ?

TheTeXnician, 2021-12-04:
Sorry, false alert. I just checked my setup. And I tested with the wrong ConTeXt version. I don't get the protected error on most recent LMTX, so everything seems to be fine there. Unfortunately, the multitude of ConTeXt versions on my system sometimes confuses me when testing.

Just to mention: Hans Hagen posted an example with working LMTX modules [on the mailing list](https://mailman.ntg.nl/pipermail/ntg-context/2021/104391.html). They might be of interest.

Rmano, 2021-12-05:
Ok, thanks. I have a couple of small enhancements for the ConTeXt side (using `\mirror` etc. to provide `\ctikzflip[x][y]{}`) --- when I manage to get them into shape I will do a release with the bug-fix and close this issue.

closed 2021-12-06 by Rmano