circuitikz/circuitikz #601: Can't  move straight voltage arrows away from european resistor

Filip62, 2021-12-09:
Hi, I would like to know if it's possible to put more distance between the resistor and the voltage arrow, in order to fit both the label and the arrow on one side.

Thank You.

[image]

```
\documentclass{article}

\usepackage[european, straightvoltages]{circuitikz}

\begin{document}
    Plotted using Ti\emph{k}Z version \pgfversion{} and CircuiTi\emph{k}Z version \pgfcircversion{}.
    
    \begin{circuitikz}
        \draw (0,0) to [R,l={$R_1$},i={$I_1$},v^={$U_1$}] (4,0);
    \end{circuitikz}
    
\end{document}
```

Rmano, 2021-12-09:
You can, using `voltage shift` (see https://texdoc.org/serve/circuitikz/0#subsubsection.5.5.5 )

```latex
\documentclass{article}

\usepackage[european, straightvoltages]{circuitikz}

\begin{document}
    Plotted using Ti\emph{k}Z version \pgfversion{} and CircuiTi\emph{k}Z version \pgfcircversion{}.

    \bigskip
    
    \begin{circuitikz}
        \draw (0,0) to [R,l={$R_1$},i={$I_1$},v^={$U_1$}, voltage shift=4] (4,0);
    \end{circuitikz}
    
\end{document}
```

[image]

*PD these kinds of questions are better asked at https://tex.stackexchange.com/questions/tagged/circuitikz --- you can have your answer in minutes ;-)*

Filip62, 2021-12-09:
I have no clue how was I able to miss that.

Thank You.

closed 2021-12-09 by Filip62

Rmano, 2021-12-09:
You're welcome! No worries... ;-)