circuitikz/circuitikz #602: Request for new functionality in the Electronic Tube component.

JetherReis, 2022-01-19:
As some electronic tubes can incorporate two functions in the same housing, such as the E82CC, I suggest that the contour line has the option of being sectioned into two halves, one in full line and the other in dashed with the option of not being drawn.
These two halves should also have the UP, DOWN, LEFT and RIGHT options, which would indicate the position of the solid line.
Attached are some examples.

[image]
[image]
[image]
.

Rmano, 2022-01-19:
It's interesting, but I don't think it's trivial. Tube code was contributed by @jesseopdenbrouw --- let's see what he thinks!

Rmano, 2022-01-20:
Hmmm... it's not trivial, as it is a bit tangled with the `fill` thing, and because the outer shape is drawn as a single path --- which can't have different line styles in it. Hmmm...

Rmano, 2022-05-02:
Hi @JetherReis  --- I finally managed to implement something... what do you think of PR #633 ?

JetherReis, 2022-05-11:
Thank you very much, it's better than I have expected.