circuitikz/circuitikz #605: Text color doesn't apply to components

schtandard, 2022-01-25:
The text color valid outside of the `circuitikz` environment applies to normal drawing operations but not to components.
```latex
\documentclass{article}

\usepackage{xcolor}
\usepackage{circuitikz}

\begin{document}

\color{red}

\begin{circuitikz}
  \draw (0,0) to [R] (2,0);
\end{circuitikz}

\end{document}
```
[image]

Interestingly, setting an invalid color makes the components inherit the correct one (and throws an error).
```latex
\documentclass{article}

\usepackage{xcolor}
\usepackage{circuitikz}

\begin{document}

\color{red}
\circuitikzset{color=asdf}

\begin{circuitikz}
  \draw (0,0) to [R] (2,0);
\end{circuitikz}

\end{document}
```
[image]

As the chapter on colors in the manual states that color support is quite limited, this may well be intended/accepted behavior. In that case I think it would be worthwhile to explicitly mention this case in the manual. (I could only find warnings against omitting the explicit `color=` option when setting the color inside the picture options.)

If this is easily fixed, that would be very nice though.

Rmano, 2022-01-25:
Yes, I know. It isn't easily corrected because the color management is a mess --- sometimes it explicitly sets black, sometimes it is inherited, and it depends on the specific shape.  It would be a lot of work to revise several hundred components (after understanding why it is doing what it's doing)...

My workaround is using an indirection like this:
```latex
\documentclass{article}

\usepackage{xcolor}
\usepackage{circuitikz}
\colorlet{mycolor}{red}

\begin{document}

\color{mycolor}
\circuitikzset{color=mycolor}

\begin{circuitikz}
  \draw (0,0) to [R] (2,0);
\end{circuitikz}

\end{document}
```

in my documents.

Rmano, 2022-01-25:
...anyway, let's keep this open and see if sometime I can find the time (or somebody volunteers!) to sort the thing...

Rmano, 2022-01-25:
Also, this seems to work (but if it does, it's by chance, so...)

```latex
\documentclass{article}

\usepackage{xcolor}
\usepackage{circuitikz}

\begin{document}

\color{red}
\circuitikzset{color=.}

\begin{circuitikz}
  \draw (0,0) to [R] (2,0);
\end{circuitikz}

\color{blue}

\begin{circuitikz}
  \draw (0,0) to [R] (2,0);
\end{circuitikz}
\end{document}
```
[image]

muzimuzhi, 2022-02-05:
Currently `/tikz/circuitikz/color` is initialized to `black` and since `\pgfsetcolor{\ctikzvalof{color}}` is used in circuitikz shapes, more specifically `\behindforegroundpath` and its friends, the shapes are drawn (and filled) in black.
https://github.com/circuitikz/circuitikz/blob/992851efcd18bbc22c55a0200d1e543e4983a6d8/tex/pgfcirc.defines.tex#L629
Depending on which is the expected behavior, this could either be documented or changed to `\ctikzset{color/.initial=.}`. Checking and changing every use of `\pgfset(|fill)color{\ctikzvalof{color}}` would be too much trouble.

muzimuzhi, 2022-02-06:
Maybe `/tikz/circuitikz/color` should be initialized to `none`, and only do `\pgfsetcolor{\ctikzvalof{color}}` when `color` is explicitly set.

An attempt:
```tex
\documentclass{article}
\usepackage{circuitikz}

\makeatletter
\newif\ifpgf@circ@setcolor

% similar to how `/tikz/draw` and `/tikz/fill` are defined
% 
% - `color=none`, \pgfusecolor is not executed at the beginning of node path(s)
% - `color` or `color=` (no or empty value), ignored
% - `color=<color name>`, \pgfusecolor{<color name>} will be executed
\ctikzset{
  color/.code={%
    \edef\pgf@circ@temp{#1}%
    \ifx\pgf@circ@temp\pgf@nonetext
      \pgf@circ@setcolorfalse
    \else
      \ifx\pgf@circ@temp\pgfutil@empty
      \else
        \pgf@circ@setcolortrue
        \pgfkeyssetvalue{\circuitikzbasekey/color}{#1}%
      \fi
    \fi
  },
  % init
  color=none
}

\def\pgf@circ@setcolor{%
  \ifpgf@circ@setcolor
    \pgfsetcolor{\ctikzvalof{color}}%
  \fi
}

% Update \backgroundpath for shape `resistorshape`, just to show the effect. 
% The real change to code base should happen in \pgfcircdeclarebipolescaled.
%
% This is conceptually equivalent to
%   \xpatchcmd\pgf@sh@bg@resistorshape % need xpatch pkg
%     {\pgfsetcolor{\ctikzvalof{color}}}
%     {\pgf@circ@setcolor}
%     {}{\PatchFailed}
% but \pgf@sh@bg@resistorshape is un-patchable because there're two consecutive 
% spaces in its replecement text, and they will be merged to one space char
% after re-scanning.
\edef\pgf@sh@bg@resistorshape{%
  \noexpand\pgf@circ@setcolor
  \unexpanded\expandafter{\expandafter\pgfutil@gobbletwo\pgf@sh@bg@resistorshape}%
}
\makeatother

\begin{document}
\color{red}
\tikz \draw (0,0) -- node[draw] {x} (1,0) -- node[draw=cyan] {y} (2,0);
\tikz \draw (0,0) to[R] (2,0) to[R,color=cyan] (4,0);

\color{blue}
\tikz \draw (0,0) -- node[draw] {x} (1,0) -- node[draw=cyan] {y} (2,0);
\tikz \draw (0,0) to[R] (2,0) to[R,color=cyan] (4,0);
\end{document}
```
[image]

Rmano, 2022-02-06:
Oh, wow, thanks @muzimuzhi  --- I was thinking something similar to this (basically, avoiding to set the color everywhere, given that I am trying to add hooks to every shape) but this seems a much better idea. 
Do you want to try a PR or are you happy if I try it (when I find time, I'm a bit stretched out now)? --- I will of course give credit 😊.
I want also to revise the path-building thing --- I do not like how it destroys arrows and similar things. I have to explore what `\pgfinterruptpath` & co does.

Rmano, 2022-02-06:
I have started a re-organization (and adding structure) for the shapes. I *think* that I can apply your method in the starting hook and remove most of the `\pgfsetcolor{\ctikzvalof{color}}` around. I should still revise the fill-ones (they are important because there are components that are expected to be filled by the stroke color, not the fill one).
Maybe I am overshooting, but... PR #610

Rmano, 2022-02-06:
@muzimuzhi  --- I'd like also the option of setting the initial value of `\ctikzset{color/.initial=.}` --- I am not sure to understand where it could fail.

muzimuzhi, 2022-02-07:
> @muzimuzhi --- I'd like also the option of setting the initial value of `\ctikzset{color/.initial=.}` --- I am not sure to understand where it could fail.

It won't fail, but solely executing `\pgfsetcolor{.}` seems redundant (I'm not 100% sure).

Rmano, 2022-02-22:
@muzimuzhi  I was trying to tackle the color thing, but I got stuck on this:

```latex
\pgfdeclareshape{sline}{
    \anchor{center}{
        \pgfpointorigin
    }
    \anchor{left}{\pgf@x=-0.2cm\pgf@y=0pt}
    \anchor{right}{\pgf@x=0.2cm\pgf@y=0pt}
    \backgroundpath{
        \pgfscope
        \pgfsetcolor{red}
        \pgfpathmoveto{\pgfpoint{-0.2cm}{0pt}}
        \pgfpathlineto{\pgfpoint{0.2cm}{0pt}}
        \pgfusepath{draw}
        \endpgfscope
    }
}
```

[image]

...why the text is red? ;-) this makes most of my text disappear... unless I scope-out all the filling part. It's a bit surprising...

Rmano, 2022-02-22:
@muzimuzhi  I was trying to implement your suggestion, solving the problems while to go, but I stumbled on this:

```latex
\documentclass[border=10pt]{standalone}
\usepackage{tikz}
\makeatother
\pgfdeclareshape{sline}{
    \anchor{center}{
        \pgfpointorigin
    }
    \anchor{left}{\pgf@x=-0.2cm\pgf@y=0pt}
    \anchor{right}{\pgf@x=0.2cm\pgf@y=0pt}
    \behindbackgroundpath{
        \pgfscope
        \pgfsetfillcolor{red}
        \pgfpathmoveto{\pgfpoint{-0.2cm}{0pt}}
        \pgfpathlineto{\pgfpoint{0.2cm}{0pt}}
        \pgfnode{circle}{center}{}{}{\pgfusepath{stroke}}
        \pgfusepath{draw}
        \pgftext[top,y=0.1cm]{zzz}
        \endpgfscope
    }
}
\makeatletter
\begin{document}
Text
\begin{tikzpicture}[]
    \node [sline]{};
\end{tikzpicture}
\end{document}
```

[image]

...and I am not sure why the color of the text is red... I imagine I missed it somewhere.

Rmano, 2022-02-22:
...and I start to be not so sure if the trivial (although not efficient) `\ctikzset{color/.initial=.}` is better...

muzimuzhi, 2022-02-22:
@Rmano Seems you have to reset the fill color for `\pgftext`:

```tex
\documentclass{article}
\usepackage{tikz}

\begin{document}
text \color{red} text
\begin{pgfpicture}
  \pgftext[right]{pgftext} % text color = outer "red"
  \pgfsetfillcolor{black}
  \pgftext[left]{pgftext} % text color = local "black"
\end{pgfpicture}
text
\end{document}
```
[image]

Or scope that `\pgfsetfillcolor`:
```tex
\pgfcsope
\pgfsetfillcolor{red}
<path>
\endpgfscope
\pgftext{text} % outer fill color is used
```

BTW the `\pgfdeclareshape` code is automatically wrapped in a `pgfscope`, so no need to add another one.

Rmano, 2022-02-22:
Thanks!

Hmmm... [I posted a question on TeX.SX also](https://tex.stackexchange.com/questions/634867/using-pgftext-in-behindbackground-shape-color-problems). The problem is that even if I scope the `\pgfsetfillcolor`, if there is an external `fill=green`, also the `\pgftext` will become green (and so now that I only conditionally set the color, it disappears). 

It's strange, I supposed that the `\pgftext` would follow stroke, not fill color.

muzimuzhi, 2022-02-22:
Not only `\pgftext`, but also normal text, follows fill color.

muzimuzhi, 2022-02-22:
btw, the example provided in https://github.com/circuitikz/circuitikz/issues/605#issuecomment-1047739354 has `\makeatletter` and `\makeatother` swapped :)

Rmano, 2022-02-22:
> btw, the example provided in [#605 (comment)](https://github.com/circuitikz/circuitikz/issues/605#issuecomment-1047739354) has `\makeatletter` and `\makeatother` swapped :)

Oops 😖

Rmano, 2022-02-23:
This is much more involved than I hoped (yes, because it was a mess). Now I am struggling with the components that must [*always* be filled with the stroke color](https://tex.stackexchange.com/questions/634987/pgf-basic-layer-struggling-again-with-colors) --- banging my head around.

*PD--- posting here the link so I can, in future, have everything related here*

Rmano, 2022-02-24:
Ok, now I found a strange thing. If I have

    \draw[draw=red, text=red] ... node[whatever]{}... ;

I can use *inside* the node definition `tikz@textcolor` or `tikz@strokecolor`. Unfortunately, if I have 

    \draw[red] ... node[whatever]{}... ; 

(or `color=red`) although the result is the same (I mean, the lines and text are red), only `\tikz@textcolor` is defined, while `\tikz@strokecolor` is left empty. @muzimuzhi, is that expected?

muzimuzhi, 2022-02-24:
Yes, `\draw[<color>]` will set both `\tikz@strokecolor` and `\tikz@fillcolor` empty, as both the doc (pgfmanual, sec. 15.2 or [this tikz.dev page](https://tikz.dev/tikz-actions#pgf./tikz/color)) and the [code](https://github.com/pgf-tikz/pgf/blob/b79c4c5c632110079c3827ce4963a27e6d5d1742/tex/generic/pgf/frontendlayer/tikz/tikz.code.tex#L256-L277) show.

I too don't know how the tikz color internals work, yet. Need more time.

closed 2022-04-15 by Rmano

brunolealdias, 2025-04-29:
My problem is this: I'm trying to have current arrows and current names in red, but the resistor symbol in the same wire turns red too. I would like to keep the resistor symbol black.

Heres a piece of my code:

\draw (5.33,4) to[R=$30\ \Omega$, color=black, f={\color{red}$i_3$}, color=red] (5.33,0) node[circ] {};

If I remove the 'color=red' then the current names stay in red, but the current arrows turn black. I can't get the current arrows and current names red and the resistor symbol black at the same time.

Rmano, 2025-04-29:
@brunolealdias  Hi, please do not use issues (especially closed ones) for usage questions next time. You can use a [discussion](https://github.com/circuitikz/circuitikz/discussions), or ask a question on [TeX StackExchange](https://tex.stackexchange.com/questions/tagged/circuitikz) or at [TopAnswers](https://topanswers.xyz/tex).

Anyway, I have an answer for you in https://github.com/circuitikz/circuitikz/discussions/863, and this question discovered a bug in the documentation that I'll fix... thanks!

PR #622 (2022-04-15): Tackle colors
Rewrite the color management of components.

Fixes #605