circuitikz/circuitikz #611: Adding socket and plug as possible terminals

n3odon, 2022-02-15:
It seems like there is a socket/plug or more generally a connector-system missing (like in the element 03-03-05 of the IEC-standard 60617). A female, male and connected version would be useful.

The standard is not free access, but several symbols can be found here (the requested symbol is on page 2):
https://studylib.net/doc/18913520/0113-graphic-symbols
Or in a slightly different version as a uploaded picture here:
[image]
Unfortunately, I don't have any detailed information about the size and relative position of the elements.

This is very useful for drawing control circuits with e.g. PLCs, where not all connections are necessarily made from the beginning. So a labeling option would be very useful (maybe with gray letters like in the tikz-relay package).

There is apparently a hack one could do with using the dynode symbol for the socket, but a more elegant way would be great.

Rmano, 2022-02-15:
The only thing to decide is the default size with respect, for example, a resistor or a switch. My gut feeling is that the plug+socket symbol should be a little smaller than a resistor --- what do you think?

n3odon, 2022-02-15:
Yes, I agree. I think 30% of the size of the resistor looks good for the terminal/plug. The socket/female size was nice at about 70% of the arc of the dynode. What do you think?

n3odon, 2022-02-15:
> I also found this, that I think shows reasonably shaped symbols:
>
>
>
> My plan is to add them (and maybe create a connectors class adding the bnc to them). Let's see when I can find a bit of time.
>
> Thanks!

Sounds great! A possible issue might be how to position the plug and socket. I would imaging placing them at the same position would be ‘plugged in’ state?!

n3odon, 2022-02-15:
> Ok. My plan is to have three components --- the plug and the socket (node type) plus one plug+socket that can be used as a path-type or node type. In all the drawings where they are near, I have seen the plug "touching" the center of the arc, so I think I'll do them that way; if somebody needs a different distance they can use the individual shapes.

Sounds good. If possible, it would be nice to be able to place ‘plugged in’ contacts next to empty contacts and still have them align properly - both for female and male side. I guess that would also make it possible to use the single nodes at the same place and make them look alike the combined element.

Rmano, 2022-02-16:
@n3odon I am not sure what you mean by a "plugged in" state... something like this:?

[image]

...that's because I never saw it, but it could be done (although the "clipping" is a bit tricky). I was thinking to make the shapes without leading wires (they are more flexible, for experience with antennas), but to make the second one easily stackable probably we need to have a little wire on the side of the plug.

n3odon, 2022-02-16:
The version with both female and male part (top version) is already the ‘plugged in’ state. Empty plug would be only the half circle. But it would be nice to have the empty female socket line up with the socket of the socket+plug version. I would think having the position at the center point of the circle and at the border it the rectangle should fix this nicely.

I don’t think you would need included wires in this case.

Rmano, 2022-02-16:
Ah, ok, so I misunderstood it.

The problem is that while the empty plugs is clearly a node-type component, the plugged one can be seen as a node or a path-style component. Clearly, the `left` anchor of the two object will be the same; then it's just a matter to use them correctly. I will show an example when I have something done...

Rmano, 2022-02-16:
Let's see what do you think... 

[image]

To play with them, you should add the following code after loading `circuitikz`, in the preamble:

```latex
\makeatletter
% define new class
\ctikzset{connectors/scale/.initial=1.0}
\ctikzset{connectors/fill/.initial=none}
\ctikzset{connectors/thickness/.initial=none}
% parameters. To have round sockets, 3*height==2*width
\ctikzset{bipoles/iecconn/height/.initial=.2}
\ctikzset{bipoles/iecconn/width/.initial=.3}
% objects
\pgfcircdeclarebipolescaled{connectors}
{
    \anchor{plug center}{\northeast\pgf@y=0pt\divide\pgf@x by 3 }
    % put the node text above and centered
    \anchor{text}{\pgfextracty{\pgf@circ@res@up}{\northeast}
        \pgfpoint{-.5\wd\pgfnodeparttextbox}{
            \dimexpr.5\ht\pgfnodeparttextbox+\pgf@circ@res@up\relax
        }
    }
}
{\ctikzvalof{bipoles/iecconn/height}}%symmetrical
{iecconn}
{\ctikzvalof{bipoles/iecconn/height}}
{\ctikzvalof{bipoles/iecconn/width}}
{
    \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@left/3}{\pgf@circ@res@up}}
    \pgfpatharc{90}{-90}{0.66666\pgf@circ@res@left and \pgf@circ@res@up}
    \pgfusepath{draw}
    \pgfpathrectanglecorners{\pgfpoint{\pgf@circ@res@left/3}{\pgf@circ@res@up/2}}{\pgfpoint{\pgf@circ@res@right}{\pgf@circ@res@down/2}}
    \pgfusepath{draw, fill}
}
\pgfcirc@activate@bipole@simple{l}{iecconn}
\pgfcirc@style@to@style{iecconn}{iec connector}

\pgfdeclareshape{iecsocket}{%
    \savedmacro{\ctikzclass}{\edef\ctikzclass{connectors}}
    \saveddimen{\scaledRlen}{\pgfmathsetlength{\pgf@x}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}}
    \savedanchor\northwest{%
        \pgfmathsetlength{\pgf@circ@scaled@Rlen}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}
        \pgf@y=\ctikzvalof{bipoles/iecconn/height}\pgf@circ@scaled@Rlen
        \pgf@y=.5\pgf@y
        \pgf@x=-\ctikzvalof{bipoles/iecconn/width}\pgf@circ@scaled@Rlen
        \divide\pgf@x by 6
    }
    \anchor{center}{\pgfpointorigin}
    \anchor{right}{\northwest\pgf@y=0pt\pgf@x=-\pgf@x}
    \anchor{east}{\northwest\pgf@y=0pt\pgf@x=-\pgf@x}
    \anchor{e}{\northwest\pgf@y=0pt\pgf@x=-\pgf@x}
    \anchor{left}{\northwest\pgf@y=0pt}
    \anchor{west}{\northwest\pgf@y=0pt}
    \anchor{w}{\northwest\pgf@y=0pt}
    \anchor{south}{\northwest\pgf@x=0pt\pgf@y=-\pgf@y}
    \anchor{s}{\northwest\pgf@x=0pt\pgf@y=-\pgf@y}
    \anchor{north}{\northwest\pgf@x=0pt}
    \anchor{n}{\northwest\pgf@x=0pt}
    \anchor{south west}{\northwest\pgf@y=-\pgf@y}
    \anchor{sw}{\northwest\pgf@y=-\pgf@y}
    \anchor{north east}{\northwest\pgf@x=-\pgf@x}
    \anchor{ne}{\northwest\pgf@x=-\pgf@x}
    \anchor{north west}{\northwest}
    \anchor{nw}{\northwest}
    \anchor{south east}{\northwest\pgf@x=-\pgf@x\pgf@y=-\pgf@y}
    \anchor{se}{\northwest\pgf@x=-\pgf@x\pgf@y=-\pgf@y}
    % put the node text above and centered
    \anchor{text}{\pgfextracty{\pgf@circ@res@up}{\northwest}
        \pgfpoint{-.5\wd\pgfnodeparttextbox}{
            \dimexpr.5\ht\pgfnodeparttextbox+\pgf@circ@res@up\relax
        }
    }
    \pgf@circ@draw@component{%
        \pgf@circ@scaled@Rlen=\scaledRlen
        \pgfstartlinewidth=\pgflinewidth
        \northwest
        \pgf@circ@res@up=\pgf@y
        \pgf@circ@res@left=\pgf@x
        \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
        \pgfpathmoveto{\pgfpoint{-\pgf@circ@res@left}{\pgf@circ@res@up}}
        \pgfpatharc{90}{-90}{2\pgf@circ@res@left and \pgf@circ@res@up}
        \pgfusepath{draw}
    }
}

\pgfdeclareshape{iecplug}{%
    \savedmacro{\ctikzclass}{\edef\ctikzclass{connectors}}
    \saveddimen{\scaledRlen}{\pgfmathsetlength{\pgf@x}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}}
    \savedanchor\northwest{%
        \pgfmathsetlength{\pgf@circ@scaled@Rlen}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}
        \pgf@y=\ctikzvalof{bipoles/iecconn/height}\pgf@circ@scaled@Rlen
        \pgf@y=.25\pgf@y
        \pgf@x=-\ctikzvalof{bipoles/iecconn/width}\pgf@circ@scaled@Rlen
        \divide\pgf@x by 3
    }
    \anchor{center}{\pgfpointorigin}
    \anchor{right}{\northwest\pgf@y=0pt\pgf@x=-\pgf@x}
    \anchor{east}{\northwest\pgf@y=0pt\pgf@x=-\pgf@x}
    \anchor{e}{\northwest\pgf@y=0pt\pgf@x=-\pgf@x}
    \anchor{left}{\northwest\pgf@y=0pt}
    \anchor{west}{\northwest\pgf@y=0pt}
    \anchor{w}{\northwest\pgf@y=0pt}
    \anchor{south}{\northwest\pgf@x=0pt\pgf@y=-\pgf@y}
    \anchor{s}{\northwest\pgf@x=0pt\pgf@y=-\pgf@y}
    \anchor{north}{\northwest\pgf@x=0pt}
    \anchor{n}{\northwest\pgf@x=0pt}
    \anchor{south west}{\northwest\pgf@y=-\pgf@y}
    \anchor{sw}{\northwest\pgf@y=-\pgf@y}
    \anchor{north east}{\northwest\pgf@x=-\pgf@x}
    \anchor{ne}{\northwest\pgf@x=-\pgf@x}
    \anchor{north west}{\northwest}
    \anchor{nw}{\northwest}
    \anchor{south east}{\northwest\pgf@x=-\pgf@x\pgf@y=-\pgf@y}
    \anchor{se}{\northwest\pgf@x=-\pgf@x\pgf@y=-\pgf@y}
    % put the node text above and centered
    \anchor{text}{\pgfextracty{\pgf@circ@res@up}{\northwest}
        \pgfpoint{-.5\wd\pgfnodeparttextbox}{
            \dimexpr.5\ht\pgfnodeparttextbox+2\pgf@circ@res@up\relax
        }
    }
    \pgf@circ@draw@component{%
        \pgf@circ@scaled@Rlen=\scaledRlen
        \pgfstartlinewidth=\pgflinewidth
        \northwest
        \pgf@circ@res@up=\pgf@y
        \pgf@circ@res@left=\pgf@x
        \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
        \pgfpathrectanglecorners{\pgfpoint{\pgf@circ@res@left}{\pgf@circ@res@up}}{\pgfpoint{-\pgf@circ@res@left}{-\pgf@circ@res@up}}
        \pgfusepath{draw, fill}
    }
}

\makeatother

```

Rmano, 2022-02-16:
Another possibility is to put the `center` anchor on the left in the socket --- if you rotate it to put it in the other direction it will work without the need for the explicit anchor. Maybe I'll do that. But in the plug is a bit more difficult... hmmm... because then you have to rotate it.
Maybe the correct thing is to keep the plug as it is, and move the center anchor on the left just for the socket.

Rmano, 2022-02-17:
> First of all, very good start! 

Thanks. 

>Concerning the anchors, I am not sure how exactly those work as default. I guess the center anchor is placed at the specified position, if not specified otherwise? 

Yes. Moreover, for simple shapes (without border anchors) is where the wire ends if you do `(start) -- (end) node[shape]{}`, and the default rotating point  --- so on second thought, for simplicity of usage, clearly the `center` anchor must be on the left for the socket and for the plug, so it's usage is symmetric with grounds, `vcc`/`vee` and the `bnc` connector. 

For the path-style component, there is no doubt, the center must be in the center and on the same line of the left/right connection, otherwise, it will be a mess. 

> Concerning labels, the socket should have the label slightly to the side where the wire is connecting, and also the plug should have its label more to the side where the wire is connecting. 

The problem here is that you can specify just one text per node and just one position. So while I see clearly the thing for socket, it's more difficult for the plug --- the wire can exit to the left or to the right, and people will not rotate it... because the shape is symmetric.  Maybe we could take the freedom to change the shape a bit? Having the plug slightly rounded in the opposite part of the wire seems elegant to me.

> The combination of both would have labels non-overlapping, on each side. Is it possible to have two labels for the connector? One at the socket-side and one at the plug-side? I can’t try modifying myself at the moment, sorry.

No, as I said you can have just one text per node. But you can add Ti*k*Z-level labels, and you can use the geographic anchors for extra nodes, so you can do it. I will try to add examples in the next try.

n3odon, 2022-02-17:
Great! Seems like you have a plan to fix all of it.

Maybe rounding the plug on one side is a good idea (even though not fully according to the standard). 

I also tried the code you posted earlier, in a way I would find natural to use them.
[image]
The issue with the anchors is the main thing I notice. Maybe the plug can also be slightly thinner, but just a bit. It could also just stay like it is. The colors are just to distinguish the different elements.

For the labels, maybe using the other available anchors is the best solution here. The optimal position may in deed vary from usage to usage.

Rmano, 2022-02-17:
With the new version you have this: 

[image]

notice that when you place a node, the node is placed at the anchor selected, but the point is never automatically moved, this is why you have the overlapping lines. All the advantages of path-style components are that you can forget about that. 
And BTW, the node is placed *after* the path is stroke, as you can see in the last line --- the black line is there under the component.

Rmano, 2022-02-17:
The new version is this: 

```latex
\makeatletter
% define new class
\ctikzset{connectors/scale/.initial=1.0}
\ctikzset{connectors/fill/.initial=none}
\ctikzset{connectors/thickness/.initial=none}
% parameters. To have round sockets, 3*height==2*width
\ctikzset{bipoles/iecconn/height/.initial=.2}
\ctikzset{bipoles/iecconn/width/.initial=.3}
% objects
\pgfcircdeclarebipolescaled{connectors}
{
    \anchor{plug center}{\northeast\pgf@y=0pt\divide\pgf@x by 3 }
    % put the node text above and centered
    \anchor{text}{\pgfextracty{\pgf@circ@res@up}{\northeast}
        \pgfpoint{-.5\wd\pgfnodeparttextbox}{
            \dimexpr.5\ht\pgfnodeparttextbox+\pgf@circ@res@up\relax
        }
    }
}
{\ctikzvalof{bipoles/iecconn/height}}%symmetrical
{iecconn}
{\ctikzvalof{bipoles/iecconn/height}}
{\ctikzvalof{bipoles/iecconn/width}}
{
    \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@left/3}{\pgf@circ@res@up}}
    \pgfpatharc{90}{-90}{0.66666\pgf@circ@res@left and \pgf@circ@res@up}
    \pgfusepath{draw}
    \pgfpathrectanglecorners{\pgfpoint{\pgf@circ@res@left/3}{\pgf@circ@res@up/2}}{\pgfpoint{\pgf@circ@res@right}{\pgf@circ@res@down/2}}
    \pgfusepath{draw, fill}
}
\pgfcirc@activate@bipole@simple{l}{iecconn}
\pgfcirc@style@to@style{iecconn}{iec connector}

\pgfdeclareshape{iecsocket}{%
    \savedmacro{\ctikzclass}{\edef\ctikzclass{connectors}}
    \saveddimen{\scaledRlen}{\pgfmathsetlength{\pgf@x}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}}
    \savedanchor\northwest{%
        \pgfmathsetlength{\pgf@circ@scaled@Rlen}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}
        \pgf@y=\ctikzvalof{bipoles/iecconn/height}\pgf@circ@scaled@Rlen
        \pgf@y=.5\pgf@y
        \pgf@x=-\ctikzvalof{bipoles/iecconn/width}\pgf@circ@scaled@Rlen
        \divide\pgf@x by 6
    }
    % notice: center is on the left side
    \anchor{center}{\northwest\pgf@y=0pt}
    \anchor{right}{\northwest\pgf@y=0pt\pgf@x=-\pgf@x}
    \anchor{east}{\northwest\pgf@y=0pt\pgf@x=-\pgf@x}
    \anchor{e}{\northwest\pgf@y=0pt\pgf@x=-\pgf@x}
    \anchor{left}{\northwest\pgf@y=0pt}
    \anchor{west}{\northwest\pgf@y=0pt}
    \anchor{w}{\northwest\pgf@y=0pt}
    \anchor{south}{\northwest\pgf@x=0pt\pgf@y=-\pgf@y}
    \anchor{s}{\northwest\pgf@x=0pt\pgf@y=-\pgf@y}
    \anchor{north}{\northwest\pgf@x=0pt}
    \anchor{n}{\northwest\pgf@x=0pt}
    \anchor{south west}{\northwest\pgf@y=-\pgf@y}
    \anchor{sw}{\northwest\pgf@y=-\pgf@y}
    \anchor{north east}{\northwest\pgf@x=-\pgf@x}
    \anchor{ne}{\northwest\pgf@x=-\pgf@x}
    \anchor{north west}{\northwest}
    \anchor{nw}{\northwest}
    \anchor{south east}{\northwest\pgf@x=-\pgf@x\pgf@y=-\pgf@y}
    \anchor{se}{\northwest\pgf@x=-\pgf@x\pgf@y=-\pgf@y}
    % put the node text above and to the left, ignore depth
    \anchor{text}{%
        \pgfextractx{\pgf@circ@res@left}{\northwest}
[... 130 lines cut ...]
            % to avoid problems with the line thickness
            \pgfpathrectanglecorners{\pgfpoint{-2\pgf@circ@res@other}{-2\pgf@circ@res@other}}
                {\pgfpoint{2\pgf@circ@res@other}{2\pgf@circ@res@other}}
            % next the opening to the right
            \pgfpathrectanglecorners{\pgfpoint{-\pgf@circ@res@step}{-\pgf@circ@res@step}}
                {\pgfpoint{2\pgf@circ@res@other}{\pgf@circ@res@step}}
            % do the difference and clip before drawing
            \pgfseteorule
            \pgfusepath{clip}
            \pgfpathcircle{\pgfpointorigin}{\pgf@circ@res@other}
            \pgfusepath{draw}
        \endpgfscope
        % internal circle
        \pgfpathcircle{\pgfpointorigin}{\pgf@circ@res@step}
        \pgf@circ@draworfill
        % and the contact line to the right
        \pgfsetlinewidth{\pgfstartlinewidth}
        \pgfpathmoveto{\pgfpoint{\pgf@circ@res@step}{0pt}}
        \pgfpathlineto{\pgfpoint{\pgf@circ@res@other}{0pt}}
        \pgfusepath{draw}
    }
}

\makeatother
```

n3odon, 2022-02-17:
Hm. I earlier thought that it would be nice to be able to use these without having the need of left or right anchors, or the need of name-references. But thinking about it, you always will have to position them side by side using left and right (apart from using path-style elements). This is a bit unsatisfying, but I guess one can get used to it. But that also means that an optimization of the anchors maybe doesn't change so much, as long as there are useful anchors available. I think this also applies to the labels. Maybe the centered version from the beginning is okay as a default after all.

n3odon, 2022-02-17:
Here is a possible workaround for alignment, both for the label and the path-style with the node-style elements:
[image]

I would prefer having it a bit easier, but it's not very bad.

Rmano, 2022-02-17:
> * The mirror-option doesn't work. But drawing from the opposite side backwards works. Both in-path lines should look the same if mirror would work.

`mirror` mirrors around the horizontal axis. You want `invert` here ;-)

Rmano, 2022-02-17:
> I would prefer having it a bit easier, but it's not very bad.

I find orthogonal coordinates much easier here --- it's the same kind of problem when aligning path-style and node-styles, anyway...

```latex
\begin{circuitikz}[european]
\draw (0,3) coordinate (start) to[iec connector, name=con1, l={\%IX01\quad\%QX04}] ++(2,0) coordinate (stop);
\draw (0,2) coordinate (below) -- (con1.left |- below) node[iecsocket] (s1) {\%IX01};
\draw[red] (s1.right) node[iecplug] {\%QX04} -- (stop |- below);
\end{circuitikz}
```
[image]

you can also align the labels in the same way if you want, using `con1.north west` etc...

Rmano, 2022-02-17:
Well, the only problem I see is that doing the connection separated on the other side is a bit complex (but that would be the same with other components... hmm...)

```
\begin{circuitikz}[]
        \draw (0,1) -- ++(1,0) node[iecplug, xscale=-1](p1){\ctikzflipx{P1}};
        \draw [red] (p1.left)
                node[iecsocket, anchor=right, xscale=-1](s1){\ctikzflipx{S1}}
                (s1.center) -- ++(1,0);
\end{circuitikz}
```

[image]

n3odon, 2022-02-17:
That's no fun. I tried to additionally align it with the path-style element, men the requires too much.
[image]

```
\begin{circuitikz}[european]
\draw (0,3) coordinate (start) to[iec connector, name=con1, l={\%QX04\quad\%IX01}, invert] ++(2,0) coordinate (stop);
\draw (0,2) coordinate (below) -- (con1.right |- below) node[iecplug, xscale=-1, left] (p1) {\ctikzflipx{\%QX04}};
\draw[red] (p1.left) node[iecsocket, right, xscale=-1] {\ctikzflipx{\%IX01}} -- (stop |- below);
\end{circuitikz}
```
Maybe a path-type element with an invisible socket, and one with an invisible plug could be interesting? Do you think that's a possibility?

Rmano, 2022-02-17:
For the alignment, you have to remember that `xscale=-1` (as `invert`) also flips the labels. 

> Maybe a path-type element with an invisible socket, and one with an invisible plug could be interesting? Do you think that's a possibility?

No, because the leading wires will be there anyway.  Probably the best solution is having 2 sockets and 2 plugs, L and R versions, and be done with them. If people want to put them vertically they will have to struggle a bit...

Must go to Real Work™ now!

n3odon, 2022-02-17:
I agree, let's do a L and R version. 

Thanks for all your work! When do you think this can be included in a new version of circuitikz?

Rmano, 2022-02-17:
Well, I have a couple of quite a few biggish things to put in 1.5.0 now, so I am not sure. I have an extra holiday around the end of the month --- I am targeting the end of February.

Rmano, 2022-02-17:
I have to run now, but it would be similar to this: 

[image]

n3odon, 2022-02-17:
> BTW, how do you like to be cited/thanked in the manual? I have your name and email address, but if you prefer I can use @n3odon on GitHub...

You can thank me with my full name, thank you for that.

n3odon, 2022-02-17:
> I have to run now, but it would be similar to this:
> 
> [image]

I think this is a good compromise. I don't see (at the moment) how this could be made easier. I am happy. :)

commit e998a1c6 (2022-02-21): Add IEC-style connectors

Fixes #611

Thanks to Alexander Sauter for suggesting them and helping in the
design: https://github.com/circuitikz/circuitikz/issues/611
closed 2022-02-21 by Rmano