circuitikz/circuitikz #615: [new shapes] nullor and norator

atticus-sullivan, 2022-03-06:
Hi,
I recently stumbled over the need of drawing nullators/norators but I wasn't able to find a cirtuitikz shape for it. For my use case I came up with a 
<details>
  <summary>quick and dirty implementation</summary>

```tex
\makeatletter
% used to process styles for to-path
\def\TikzBipolePath#1#2{\pgf@circ@bipole@path{#1}{#2}}
    % restore size value for bipole definitions
% \pgf@circ@Rlen=\pgfkeysvalueof{/tikz/circuitikz/bipoles/length}
\makeatother

\newlength{\ResUp}
\newlength{\ResDown}
\newlength{\ResLeft}
\newlength{\ResRight}

% norator
\ctikzset{bipoles/norator/height/.initial=.5}   % box height
\ctikzset{bipoles/norator/width/.initial=.5}    % box width

\pgfcircdeclarebipole{}                           % no extra anchors
{\ctikzvalof{bipoles/norator/height}}             % offset to label
{norator}                                         % component name
{\ctikzvalof{bipoles/norator/height}}             % height of box
{\ctikzvalof{bipoles/norator/width}}              % width of box
{                                                 % component symbol drawing ...
  \pgfsetlinewidth{\pgfkeysvalueof{/tikz/circuitikz/bipoles/thickness}\pgfstartlinewidth}
  \pgfextracty{\ResUp}{\northeast}                % coordinates
  \pgfextracty{\ResDown}{\southwest}
  \pgfextractx{\ResLeft}{\southwest}
  \pgfextractx{\ResRight}{\northeast}
  \pgfpathellipse{\pgfpoint{\ResUp/2}{0}}
        {\pgfpoint{0}{\ResUp/2}}
        {\pgfpoint{\ResUp/2}{0}}
  \pgfpathellipse{\pgfpoint{-\ResUp/2}{0}}
        {\pgfpoint{0}{\ResUp/2}}
        {\pgfpoint{\ResUp/2}{0}}
  \pgfusepath{draw}                               % draw it!
}

\def\circlepath#1{\TikzBipolePath{norator}{#1}}
\tikzset{norator/.style = {\circuitikzbasekey, /tikz/to path=\circlepath, l=#1}}

% nullator
\ctikzset{bipoles/nullator/height/.initial=.5}   % box height
\ctikzset{bipoles/nullator/width/.initial=.5}    % box width

\pgfcircdeclarebipole{}                            % no extra anchors
{\ctikzvalof{bipoles/nullator/height}}             % offset to label
{nullator}                                         % component name
{\ctikzvalof{bipoles/nullator/height}}             % height of box
{\ctikzvalof{bipoles/nullator/width}}              % width of box
{                                                 % component symbol drawing ...
  \pgfsetlinewidth{\pgfkeysvalueof{/tikz/circuitikz/bipoles/thickness}\pgfstartlinewidth}
  \pgfextracty{\ResUp}{\northeast}                % coordinates
  \pgfextracty{\ResDown}{\southwest}
  \pgfextractx{\ResLeft}{\southwest}
  \pgfextractx{\ResRight}{\northeast}
  \pgfpathellipse{\pgfpointorigin}
        {\pgfpoint{0}{\ResUp/2}}
        {\pgfpoint{\ResLeft}{0}}
  \pgfusepath{draw}                               % draw it!
}

\def\circlepathII#1{\TikzBipolePath{nullator}{#1}}
\tikzset{nullator/.style = {\circuitikzbasekey, /tikz/to path=\circlepathII, l=#1}}

```
</details>
which basically is the adaptation of [this tex.stackexchange post](https://tex.stackexchange.com/a/292834/206293).

The result is e.g. 
[image]

to be fair, I think the nullor already doesn't look bad. But as far as I know the norator is supposed to have a shape like an infinity-symbol. And since I don't have experience with implementing circuitikz shapes of course there are no fancy keys for scaling, other manual modification or anchors.

This is my reference on how the shapes should look like from my textbook:
[image]

Rmano, 2022-03-06:
Hmmm... https://en.m.wikipedia.org/wiki/Nullor shows quite different symbols. Is there a standard for them? 
I will look tomorrow, but the Wikipedia symbols can be probably obtained by simple tweaks to `esource` and `oosource`.

atticus-sullivan, 2022-03-06:
Well yes I can find both/multiple symbols for both components. The basic shape is always the same, but sometimes the symbol is more like a circle, sometimes more like an ellipsis. Up to now, I only knew it from my textbook (see 2nd picture in my first post). 

Not quite sure how to handle this on first sight I cannot really find a standard or an explanation for the different shaping (e.g. American/European or maybe the circular shapes tend to get too huge the symbol is scaled up).

Rmano, 2022-03-07:
The wikipedia shapes can be obtained by simply styling: 

```latex
\documentclass[border=2.78pt, tikz]{standalone}
\usepackage{circuitikz}
\tikzset{norator/.style={voosource,
		bipoles/oosource/circlesize=0.5,
		bipoles/oosource/circleoffset=0.5},
	nullor/.style={esource, sources/scale=0.5},
}
\begin{document}
	\begin{tikzpicture}
		\draw (0,0) to[norator] ++(2,0) to[nullor] ++(2,0);
	\end{tikzpicture}
\end{document}
```

[image]

atticus-sullivan, 2022-03-07:
Alright, thanks. Of course it would be nice if the shapes are available, but it's absolutely not urgent (after all we have at least two implementations to go with for now) so take your time.

Just correcting a small mistake I made when creating this issue: The first element isn't called `nullor` instead it should be the `nullator` (the `nullor` is already some sort of a composition of `nullator` and `norator`).

Rmano, 2022-03-09:
This could be a first attempt (I am not convinced of the shape of the norator symbol...)

```latex
\documentclass[]{article}
\usepackage[T1]{fontenc}
\usepackage[utf8]{inputenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\usepackage{ctikzmanutils}

\makeatletter
\ctikzset{bipoles/nullator/height/.initial=.30}
\ctikzset{bipoles/nullator/width/.initial=.60}
%% Empty Source
\pgfcircdeclarebipolescaled{sources}
{}
{\ctikzvalof{bipoles/nullator/height}}
{nullator}
{\ctikzvalof{bipoles/nullator/height}}
{\ctikzvalof{bipoles/nullator/width}}
{
    \pgfpointorigin
    \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
    \pgfpathellipse{\pgfpointorigin}{\pgfpoint{0}{\pgf@circ@res@up}}{\pgfpoint{\pgf@circ@res@left}{0}}
    \pgf@circ@draworfill
}
\pgfcirc@activate@bipole@simple{l}{nullator}
\ctikzset{bipoles/norator/height/.initial=.30}
\ctikzset{bipoles/norator/width/.initial=.80}
%% Empty Source
\pgfcircdeclarebipolescaled{sources}
{}
{\ctikzvalof{bipoles/norator/height}}
{norator}
{\ctikzvalof{bipoles/norator/height}}
{\ctikzvalof{bipoles/norator/width}}
{
    \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
    \pgf@circ@res@other=\dimexpr\pgf@circ@res@right-\pgf@circ@res@up\relax
    % right semicircle
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@other}{\pgf@circ@res@up}}
    \pgfpatharc{90}{-90}{\pgf@circ@res@up}
    % connect to left semicircle
    \pgfpathcurveto
    {\pgfpoint{-.3\pgf@circ@res@other}{-\pgf@circ@res@up}}
    {\pgfpoint{.3\pgf@circ@res@other}{\pgf@circ@res@up}}
    {\pgfpoint{-\pgf@circ@res@other}{\pgf@circ@res@up}}
    % left semicircle
    \pgfpatharc{90}{270}{\pgf@circ@res@up}
    % connect to right semicircle
    \pgfpathcurveto
    {\pgfpoint{.3\pgf@circ@res@other}{-\pgf@circ@res@up}}
    {\pgfpoint{-.3\pgf@circ@res@other}{\pgf@circ@res@up}}
    {\pgfpoint{\pgf@circ@res@other}{\pgf@circ@res@up}}
    \pgf@circ@draworfill
}
\pgfcirc@activate@bipole@simple{l}{norator}
\makeatother

\begin{document}

%% override "showexpl" fixed setting
\circuitdescbip*{esource}{Empty voltage source}{}
\circuitdescbip*{nullator}{Nullator element (virtual short circuit)}{}
\circuitdescbip*{norator}{Norator element (forces V and I)}{}

They are in the \texttt{sources} class, but they are not really sources (you have to manually specify voltages and current if you want).

\begin{LTXexample}[varwidth=true, basicstyle=\small\ttfamily]
    \begin{circuitikz}[american]
        \draw (0,2) to[nullator, v=0, i=0] ++(3,0);
        \draw (0,0) to[norator, v=$V$, i=$i$] ++(3,0);
    \end{circuitikz}
\end{LTXexample}

\end{document}
```

[image]

Rmano, 2022-03-09:
And after a bit of struggle with basic trigonometry and the `arc` syntax... I think this one is better:

```latex
\documentclass[]{article}
\usepackage[T1]{fontenc}
\usepackage[utf8]{inputenc}
\usepackage[siunitx, RPvoltages]{circuitikz}
\usepackage{ctikzmanutils}

\makeatletter
\ctikzset{bipoles/nullator/height/.initial=.30}
\ctikzset{bipoles/nullator/width/.initial=.60}
%% Empty Source
\pgfcircdeclarebipolescaled{sources}
{}
{\ctikzvalof{bipoles/nullator/height}}
{nullator}
{\ctikzvalof{bipoles/nullator/height}}
{\ctikzvalof{bipoles/nullator/width}}
{
    \pgfpointorigin
    \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
    \pgfpathellipse{\pgfpointorigin}{\pgfpoint{0}{\pgf@circ@res@up}}{\pgfpoint{\pgf@circ@res@left}{0}}
    \pgf@circ@draworfill
}
\pgfcirc@activate@bipole@simple{l}{nullator}
\ctikzset{bipoles/norator/height/.initial=.30}
\ctikzset{bipoles/norator/width/.initial=.80}
%% Empty Source
\pgfcircdeclarebipolescaled{sources}
{}
{\ctikzvalof{bipoles/norator/height}}
{norator}
{\ctikzvalof{bipoles/norator/height}}
{\ctikzvalof{bipoles/norator/width}}
{
    \pgf@circ@setlinewidth{bipoles}{\pgfstartlinewidth}
    \pgf@circ@res@other=\dimexpr\pgf@circ@res@right-\pgf@circ@res@up\relax
    \pgfmathsetmacro{\@@angle}{atan2(\pgf@circ@res@other,\pgf@circ@res@up)}
    % \typeout{ANGLE\space\@@angle}
    \pgfmathsetlength{\pgf@circ@res@step}{\pgf@circ@res@up*cos(\@@angle)}
    \pgfmathsetlength{\pgf@circ@res@temp}{\pgf@circ@res@up*sin(\@@angle)}
    % right semicircle
    \pgfpathmoveto{\pgfpoint{\pgf@circ@res@other-\pgf@circ@res@step}{\pgf@circ@res@temp}}
    \pgfpatharc{180-\@@angle}{-180+\@@angle}{\pgf@circ@res@up}
    % connect to left semicircle
    \pgfpathlineto
        {\pgfpoint{-\pgf@circ@res@other+\pgf@circ@res@step}{\pgf@circ@res@temp}}
    % left semicircle
    \pgfpatharc{\@@angle}{360-\@@angle}{\pgf@circ@res@up}
    % connect to right semicircle
    \pgfpathlineto
        {\pgfpoint{\pgf@circ@res@other-\pgf@circ@res@step}{\pgf@circ@res@temp}}
    \pgfpathclose
    \pgf@circ@draworfill
}
\pgfcirc@activate@bipole@simple{l}{norator}
\makeatother

\begin{document}

%% override "showexpl" fixed setting
\circuitdescbip*{esource}{Empty voltage source}{}
\circuitdescbip*{nullator}{Nullator element (virtual short circuit)}{}
\circuitdescbip*{norator}{Norator element (forces V and I)}{}

They are in the \texttt{sources} class, but they are not really sources (you have to manually specify voltages and current if you want).

\begin{LTXexample}[varwidth=true, basicstyle=\small\ttfamily]
    \begin{circuitikz}[american]
        \draw (0,2) to[nullator, v=0, i=0] ++(3,0);
        \draw (0,0) to[norator, v=$V$, i=$i$] ++(3,0);
    \end{circuitikz}
\end{LTXexample}

\end{document}
```

[image]

Rmano, 2022-03-09:
Do you think that the relative sizes with respect to the normal sources are ok? I went for more or less retaining the bigger side, but...

atticus-sullivan, 2022-03-10:
Oh looks pretty nice, looking at the picture from my textbook the nullator maybe should be "longer" (there it is as long as the norator). But that's just judging based on my textbook (no standardization document or what so ever).
I asked in my lecture if someone know the reason for the different shapes (maybe there is some difference and in the end both shape types can be implemented with some sort of switch, but for that we need to know the origin/reason behind the difference)

Rmano, 2022-03-10:
I quite like the effect:

[image]

Rmano, 2022-03-10:
This is with the nullator and norator with the same (default) length:

[image]

Rmano, 2022-03-11:
What about this? It seems more balanced w/respect to the standard sources...

[image]

Rmano, 2022-03-11:
@atticus-sullivan  how do you like to be cited in the manual? is "user atticus-sullivan on github" ok?

Rmano, 2022-03-11:
[image]

Rmano, 2022-03-11:
s/this/these/ ;-)

closed 2022-03-12 by Rmano

atticus-sullivan, 2022-03-12:
> @atticus-sullivan how do you like to be cited in the manual? is "user atticus-sullivan on github" ok?

Yes that's ok. Thanks for implementing. (When I get a response on if the different shapes can be classified somehow, I'll write here in this issue)