circuitikz/circuitikz #621: Request for German Mixer Types

myzinsky, 2022-04-07:
This would be the mixer symbol according DIN and EN:

[image]

Howes in Germany also this variant (similar to the boxed mixer, but without circle) is used:

[image]

Would it be possible to integrate both?

Rmano, 2022-04-09:
Hi! These components (like the antenna in #620) are quite easy to draw with the existing ones... I will see if I can add them, anyway, but I am quite busy now...

Rmano, 2022-04-18:
I have added the "no-circle" option for the boxed mixer (and similar components). 

I am not sure about the other symbol --- it does not fit with the blocks; they have one input and one output, and the symbol you shows seems to have three inputs (even if the symbol shows 4). 

[image]

Maybe you can have a mixer of that style painting over a generic fourport (which can easily added as a macro or as a subcircuit):

```latex
\documentclass[border=10pt]{standalone}
\usepackage[T1]{fontenc}
\usepackage{circuitikz}
\begin{document}
\begin{circuitikz}
    \node [fourport](A){};
    \coordinate (lA) at ($(A.west)!0.3!(A.center)$);
    \coordinate (mA) at ($(A.west)!0.6!(A.center)$);
    \coordinate (eA) at ($(A.east)!0.3!(A.center)$);
    \draw (A.port1 -| lA) -- (mA) (A.port4 -| lA) -- (mA) -- (eA);
    \draw (A.port1) -- ++(-1,0);
    \draw (A.port4) -- ++(-1,0);
    \draw (A.east) -- ++(1,0);
\end{circuitikz}
\end{document}
```

[image]

closed 2022-04-18 by Rmano