circuitikz/circuitikz #628: Version 1.5.0 using regexpatch

alceu-git, 2022-04-25:
After the last update, 1.5.0, miktex (version 4.7) started complaining about 'commands being redefined' (like \xshowcmd \xpretocmd \xpatchcmd in the package regexpatch.sty). 
It took me a awhile to find out what was going on (I don't use/include, myself, any (x)patch package). 
anyway, once I identified 'who was probably loading/using regexpatch', circuitikz, I reverted to version 1.4.6 and all problems were gone.
In time, in the LaTeX projects I had trouble with the new version, I'm also using xparse and xstring (those seem to be the ones that, behind the scenes, are loading xpatch themselves... which crashes with the later use of regexpatch).

Rmano, 2022-04-26:
This is the new color management stuff. I'll look into it as soon as I can...

Rmano, 2022-04-26:
I have tried on linux with TeXLive 2021 and TeXLive 2022 and I am not able to reproduce it, so I need help to debug it...

alceu-git, 2022-04-26:
hi. I'm tight on time 'today' (for the next 9 hours or so), later I can try to have a 'Minimal example' reproducing the bug.
Btw, yesterday, one test I did was replacing the file regexpatch with a stub (all declarations removed...) and your package, indeed, complained about not being able to patch some color commands.

Rmano, 2022-04-26:
> hi. I'm tight on time 'today' (for the next 9 hours or so), later I can try to have a 'Minimal example' reproducing the bug.
> Btw, yesterday, one test I did was replacing the file regexpatch with a [...]

Yes, that is the problem --- it should be a minor one, anyway, so I can think to use some logic to avoid the patching altogether if it can create errors. I'll wait for more info. It's strange that I can't reproduce the problem here --- are you sure you do not have a partially updated system (like using packages that are newer than the kernel or something like that?)

Rmano, 2022-04-26:
Ah, found it. The problem is that 

```latex
\documentclass[border=10pt]{standalone}
\usepackage{xpatch}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{tikzpicture}[]
    \draw [red] (0,0) to[R, l=\pgfcircversion] ++(2,0);
\end{tikzpicture}
\end{document}
```

will not work. So you probably load `xpatch` somewhere. Hmmm.... I have to fix it, yes.

alceu-git, 2022-04-26:
:) system update status... the good with miktex is that it is "up to date" (well, sometimes too up to date). 
that's something I can test (and inform you) later on (sorry, I really have to check out, for a while, now)

alceu-git, 2022-04-26:
> Ah, found it. The problem is that
> 
> ```
> \documentclass[border=10pt]{standalone}
> \usepackage{xpatch}
> \usepackage[siunitx, RPvoltages]{circuitikz}
> \begin{document}
> \begin{tikzpicture}[]
>     \draw [red] (0,0) to[R, l=\pgfcircversion] ++(2,0);
> \end{tikzpicture}
> \end{document}
> ```
> 
> will not work. So you probably load `xpatch` somewhere. Hmmm.... I have to fix it, yes.

Oh, this is what baffles me at first, when digging in, I don't load any (x)patch package, and I have tried to find out who was loading xpatch and who was loading regexpatch.
from the log traces (I kept a copy of it failing), xpatch seems to be loaded by either xparse or xstring (might also be biblatex)

(C:\Users\Alceu\AppData\Local\Programs\MiKTeX\tex/latex/l3packag
es/xparse\xparse.sty
(C:\Users\Alceu\AppData\Local\Programs\MiKTeX\tex/latex/l3kernel
\expl3.sty
Package: expl3 2022-04-20 L3 programming layer (loader) 

(C:\Users\Alceu\AppData\Local\Programs\MiKTeX\tex/latex/l3backen
d\l3backend-pdftex.def
File: l3backend-pdftex.def 2022-04-20 L3 backend support: PDF output (pdfTeX)
\l__color_backend_stack_int=\count402
\l__pdf_internal_box=\box50
))
Package: xparse 2022-01-12 L3 Experimental document command parser
)
(C:\Users\Alceu\AppData\Local\Programs\MiKTeX\tex/latex/xpatch\x
patch.sty
Package: xpatch 2020/03/25 v0.3a Extending etoolbox patching commands
)
Package biblatex Info: Trying to load bibliography style 'standard'...
Package biblatex Info: ... file 'standard.bbx' found.
......
......

(C:\Users\Alceu\AppData\Local\Programs\MiKTeX\tex/generic/pgf/fr
ontendlayer/tikz/libraries\tikzlibrarytopaths.code.tex
File: tikzlibrarytopaths.code.tex 2021/05/15 v3.1.9a (3.1.9a)
)))
(C:\Users\Alceu\AppData\Local\Programs\MiKTeX\tex/latex/regexpat
ch\regexpatch.sty
Package: regexpatch 2021/03/21 v0.2f Extending etoolbox patching commands

commit 32e9e548 (2022-04-26): CTAN release 1.5.1 (2022-04-26)

Bug fix release: fixes #628
closed 2022-04-26 by Rmano