circuitikz/circuitikz #631: proximity sensor symbol

anisiobraga, 2022-04-29:
A proximity symbol would be nice to have in Circuitikz, my preferred schematic editor from now on.  Here are examples: https://knowledge.autodesk.com/support/autocad-electrical/learn-explore/caas/CloudHelp/cloudhelp/2019/ENU/AutoCAD-Electrical/files/GUID-1B18023D-241C-4E50-822A-94F873FC258B-htm.html

I've improvised using \ctikzsubcircuitdef, but an embed symbol would be great. An embedded Circuitikz symbol simplify coding and thus usage! 
`
\documentclass[10pt]{standalone}
\usepackage[siunitx,american, RPvoltages]{circuitikz}  

\begin{document}

\ctikzsubcircuitdef{proximeter}{in, out, center}{% reference anchor is -center
coordinate(#1-center)
(#1-center) +(-0.5,0)coordinate (#1-in)  +(0.5,0)coordinate (#1-out) 
(#1-in) to[ecsource] (#1-out)  
(#1-center)++(-0.35,0.15) to ++(0.7,0)
(#1-center)++(-0.35,-0.15) to ++(0.7,0)
% leave the position of the path at the center
(#1-center) 
}

\ctikzsubcircuitactivate{proximeter}
\begin{tikzpicture} 
  \draw[rotate=0] (2,0) \proximeter{PS}{};

  \draw[rotate=90] (0,0) \proximeter{PS}{};
  \draw (PS-in) --++(0,-1) coordinate(PS1) 
        (PS-out) --++(0,1) coordinate(PS2); 
\end{tikzpicture} 

\end{document}
`

Rmano, 2022-04-29:
Ok --- I think it could be useful... @anisiobraga  --- let's plan something:

1. probably just the diamond symbol is needed; the rest is doable by joining it with switches and general boxes
2. the best strategy is to define the diamond as a bipole.

Doubts and opinions:

- names; maybe `proximeter` as bipole-type that will generate `proximetershape` as node-type
- which class? It seems to be used mainly with switches, but maybe `misc` is the better choice
- the internal lines should be the same thickness as the diamond or not?

Rmano, 2022-05-01:
@anisiobraga  but looking around, it seems that the symbol is not a bipole --- it's more a kind of an auxiliary shape used together with a switch.
So I think I have changed my mind and I will build a node-type shape as a base, so that I'll have a `node[proximeter]` and maybe add an `to[inline proximeter, ...]` option.

anisiobraga, 2022-05-01:
Dear Romano,

Thanks for your email and your generous expertise with your usual high professional contribution. 
Since you started contributing to Circuitikz, the package reached such a professional standard that it is now a top recommended Latex schematic drawing package.(:-)

> 
> 
> Ok --- I think it could be useful... @anisiobraga <https://github.com/anisiobraga> --- let's plan something:
> 
> probably just the diamond symbol is needed; the rest is doable by joining it with switches and general boxes
Yes. I’ve used the symbol as dipole and a node shape. Just the fancy diamond with anchors would serves multiple purposes. 
> the best strategy is to define the diamond as a bipole.
As a dipole it's easier to draw a path. However, the symbol is frequently used as a qualifying symbol associate with switches  or standing alone inside boxes without connections, so a node shape companion is desirable. See examples in https://symbols.radicasoftware.com/category/proximity-sensors <https://symbols.radicasoftware.com/category/proximity-sensors> 
> Doubts and opinions:
> 
> names; maybe proximeter as bipole-type that will generate proximetershape as node-type
Yes. These names make an immediate association with the symbol usage.
> which class? It seems to be used mainly with switches, but maybe misc is the better choice
misc class seems to be more appropriate since it can be used with switches and as a dipole in the case of a generic 2-wire primary symbol of a proximate probe (see an example of an PNP sensor below). The probe can be inductive, capacitive, optical, etc.
> the internal lines should be the same thickness as the diamond or not?
I’ve implemented it with different thickness to highlight the external shape, but I understand that this behavior may complicate when scaling things. So, an implementation with the same thickness sounds more robust.
Following is a Latex code  using the symbol as a dipole representing a primary sensor of a generic proximity probe.

%-------------------------------------------------------------------
\documentclass[10pt]{standalone}
\usepackage[usenames]{color} %used for font color

\usepackage{tikz}
\usepackage[siunitx,american, RPvoltages]{circuitikz}
\usetikzlibrary{fit,arrows,shapes,calc,positioning}
\usetikzlibrary{backgrounds}  % drawing on the background after the foreground

% Definitions to help locate nodes. Comment out the second definition when done!
\def\normalcoord(#1){coordinate(#1)} \def\showcoord(#1){node[circle, red, draw, inner sep=1pt, 
	pin={[red, overlay, inner sep=0.5pt, font=\tiny, pin distance=0.1cm, 
		pin edge={red, overlay}]45:#1}](#1){}} 
\let\coord=\normalcoord 
%\let\coord=\showcoord 
\begin{document}

\ctikzsubcircuitdef{proximeter}{in, out, center}{% reference anchor is -center
coordinate(#1-center)
(#1-center) +(-0.5,0)coordinate (#1-in)  +(0.5,0)coordinate (#1-out) 
(#1-in) to[ecsource] (#1-out)  
(#1-center)++(-0.35,0.15) to ++(0.7,0)
(#1-center)++(-0.35,-0.15) to ++(0.7,0)
% leave the position of the path at the center
(#1-center) 
}

\ctikzsubcircuitactivate{proximeter}
\begin{circuitikz}[scale=1, transform shape, cute inductors]
	\ctikzset{resistors/scale=0.7,  capacitors/scale=0.7, inductors/scale=0.8,sources/scale=0.8, amplifiers/scale=1, transistors/scale=0.9}
    \ctikzset{american open voltage=legacy}
    \ctikzset{transistors/arrow pos=end} 

	\ctikzset{amplifiers/plus={\tiny 1},amplifiers/minus={\tiny 2}} 
	% --- Definitions -----------
	\def\dx{2}	\def\dy{2}
	%---------------------------
	%--- 3-wire Discrete Sensor 
\begin{scope}
    \draw (0,0) node[inst amp, noinv input up,  anchor=out, text width=1.25cm, 
                        align=center, font=\tiny] (TX) {Conditioning\\Circuit};
        %--- proximetro
        \draw[scale=0.5, rotate=90] ($(TX.+)!0.5!(TX.-)+(-0.1,0)$) \proximeter{PS}{};
        \draw (PS-out) |- (TX.+) (PS-in) |- (TX.-) ; 

        \draw(TX.out) node[pnp, anchor=B](Q){}(Q.C) to[short,-o] ++(0.5,0)node[above right]{$V_{S}$} \coord(Vs); 

        \draw[color=red!50!gray]  (TX.up) node[below, font=\color{red}]{$+$} to[short] ++(0,0.25) \coord(VCCi) 
                                  (VCCi) to[short,-o] (VCCi-|Vs) node[above right]{$+24V_{DC}$}\coord(VCC); 
        \draw (Q.E)  to[short,-*] (Q.E|-VCC);
		\draw (TX.down) to[short] ++(0,-0.5) \coord(aux)  to[short,-*] (aux-|Q.C)node[ground]{}\coord(gnd)
              (gnd)  to[open] (Q.C) 
              (gnd)  to[short,-o] (gnd-|Vs)node[below right]{$GND$}  
		      (TX.down) node[above, font=\color{blue}] {$-$};

        \draw[very thin] ($(TX.out)+(-0.25,0)$) node[schmitt, fill=white,anchor=in,scale=0.35]{}; 

	\end{scope}

% ---  Background blocks
    \begin{pgfonlayer}{background}  
        \path (VCC) ++(-0.35,0.15) \coord(box1)   (TX.down) ++(-1,-0.75) \coord(box2); 
        \node[fit=(box1) (box2),xshift=-0cm, draw, rectangle, thick,inner sep=10pt, color=teal!50!gray, fill=teal!10, opacity=0.25]{};
        \node at ($(VCC-|TX)+(0.5,0)$)[above, align=center]{PNP Sensor};
        \draw[gray,latex-] ($(TX.-)+(-0.1,-0.1)$) to[out=-90, in=180] ++(0.5,-1)node[right, align=center, font=\tiny ]{Primary Sensor\\ Proximity probe};
   \end{pgfonlayer}
%------------------------------------------------
\end{circuitikz}

\end{document}

anisiobraga, 2022-05-01:
Romano,

The proximeter symbol is used both ways, as a dipole and as a node shape. The proximity transducer probe is the one used as a dipole inside a proximeter discrete sensor. Since the proximity transducer can be of different types e.g. inductive, capacitive optical, etc, when drawing schematic diagrams, a generic  dipole symbol serves best the principle of a contactless sensor. Besides, the diamond symbol is a qualifying symbol used standing alone or in conjunction with mainly switches and other qualifying symbols inside square boxes. The advantage of having an embedded symbol with anchors is the ease to scale and rotate besides flexible combination with other components like switches and transistors.

Best regards,

Anisio R. Braga
COLTEC/UFMG
***@***.***

Rmano, 2022-05-01:
I think that's better `switches` if only for the name as classes, and given that they are used alone or with switches most of the time, that will let them scale accordingly. @anisiobraga, hat do you think about the following?

[image]

You can check them with a recent `circuitikz` adding this after loading `circuitikz`:

```latex
\makeatletter
% proximeter shape, for usage with switches
\ctikzset{proximeter/width/.initial=0.3}
\ctikzset{proximeter/hlines thickness/.initial=0.5}
\ctikzset{proximeter/hlines position/.initial=0.3}
\pgfdeclareshape{proximeter}
{
    \savedmacro{\ctikzclass}{\edef\ctikzclass{switches}}
    \saveddimen{\scaledRlen}{\pgfmathsetlength{\pgf@x}{\ctikzvalof{\ctikzclass/scale}\pgf@circ@Rlen}}
    \savedmacro{\hlinepos}{\edef\hlinepos{\ctikzvalof{proximeter/hlines position}}}
    \savedanchor\northeast{%
        \pgfmathsetlength{\pgf@y}{\ctikzvalof{proximeter/width}*\ctikzvalof{\ctikzclass/scale}*\pgf@circ@Rlen}
        \pgf@x=\pgf@y
    }
    \anchor{center}{\pgfpointorigin}
    % geo anchors based on north-west
    \pgfcirc@northeast@symmetric@geoanchors
    \anchor{text}{%
        \northeast
        \pgf@x=\dimexpr -.5\wd\pgfnodeparttextbox\relax
        \advance\pgf@y by .6\ht\pgfnodeparttextbox\relax
    }
    \anchor{hlines ne}{%
        \northeast
        \pgf@y=\hlinepos\pgf@y
        \advance\pgf@x by -\pgf@y
    }
    \anchor{hlines nw}{%
        \northeast
        \pgf@y=\hlinepos\pgf@y
        \advance\pgf@x by -\pgf@y\pgf@x=-\pgf@x
    }
    \anchor{hlines se}{%
        \northeast
        \pgf@y=\hlinepos\pgf@y
        \advance\pgf@x by -\pgf@y
        \pgf@y=-\pgf@y
    }
    \anchor{hlines sw}{%
        \northeast
        \pgf@y=\hlinepos\pgf@y
        \advance\pgf@x by -\pgf@y\pgf@x=-\pgf@x
        \pgf@y=-\pgf@y
    }
    \anchorborder{
        % \typeout{IN\space X:\the\pgf@x\space Y:\the\pgf@y}
        \pgfmathsetmacro{\@@switchx}{ifthenelse(\pgf@x>0,1,-1)}
        \pgfmathsetmacro{\@@switchy}{ifthenelse(\pgf@y>0,1,-1)}
        \pgfmathsetlength{\pgf@xa}{abs(\pgf@x)}
        \pgfmathsetlength{\pgf@ya}{abs(\pgf@y)}
        \pgfextracty{\pgf@circ@res@up}{\northeast}
        % \typeout{MID\space X:\the\pgf@xa\space Y:\the\pgf@ya\space L:\the\pgf@circ@res@up}
        % \typeout{MID\space SX:\@@switchx\space SY:\@@switchy}
        \pgfpointintersectionoflines
            {\pgfpointorigin}{\pgfqpoint{\pgf@xa}{\pgf@ya}}
            {\pgfqpoint{0pt}{\pgf@circ@res@up}}{\pgfqpoint{\pgf@circ@res@up}{0pt}}
        % \typeout{CROSS \space X:\the\pgf@x\space Y:\the\pgf@y}
        \pgf@x=\@@switchx\pgf@x
        \pgf@y=\@@switchy\pgf@y
    }
    \pgf@circ@draw@component{
        \northeast\pgf@circ@res@temp=\pgf@y
        \pgf@circ@setcolor
        \pgf@circ@setlinewidth{tripoles}{\pgflinewidth}
        \pgfpathmoveto{\pgfpoint{-\pgf@circ@res@temp}{0pt}}
        \pgfpathlineto{\pgfpoint{0pt}{\pgf@circ@res@temp}}
        \pgfpathlineto{\pgfpoint{\pgf@circ@res@temp}{0pt}}
        \pgfpathlineto{\pgfpoint{0pt}{-\pgf@circ@res@temp}}
        \pgfpathclose
        \pgf@circ@draworfill
        \pgfsetlinewidth{\ctikzvalof{proximeter/hlines thickness}*\pgflinewidth}
        \pgfmathsetlength{\pgf@circ@res@up}{\hlinepos*\pgf@circ@res@temp}
        \pgfmathsetlength{\pgf@circ@res@right}{\pgf@circ@res@temp-\pgf@circ@res@up}
        \pgfpathmoveto{\pgfpoint{-\pgf@circ@res@right}{\pgf@circ@res@up}}
        \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}{\pgf@circ@res@up}}
        \pgfpathmoveto{\pgfpoint{-\pgf@circ@res@right}{-\pgf@circ@res@up}}
        \pgfpathlineto{\pgfpoint{\pgf@circ@res@right}{-\pgf@circ@res@up}}
        \pgfusepath{draw}
    }

}
\pgfcirc@port@node@to@path{proximeter}{inline proximeter}
\makeatother
```

...and the example above, that you can cut and paste, is:

```latex
\begin{tikzpicture}
    \tikzset{small up proxi/.style={proximeter, solid,
            circuitikz/switches/scale=0.707,
            circuitikz/proximeter/hlines thickness=1,
            circuitikz/proximeter/hlines position=0.1}}
    \draw (0,0) to[inline proximeter, l=P1] ++(2,0)
        to[ccgsw, name=P2] ++(2,0);
    \draw[dashed] (P2.mid) -- ++(0,0.5)
        node[small up proxi, above](P2p){Fe}
        (P2p.north) ++ (0,0.5); % extend bounding box
\end{tikzpicture}
```

anisiobraga, 2022-05-01:
>         % \typeout{IN\space ***@***.***\space ***@***.***}
>         \pgfmathsetmacro{\@@***@***.***>0,1,-1)}
>         % \typeout{MID\space ***@***.***\space ***@***.***\space ***@***.***@***@***.***}
>         % \typeout{CROSS \space ***@***.***\space ***@***.***}

Romano,

It looks great! I agree that the scaling matching with switches is the way to do it since it facilitates a lot combining different switches and a proximeter probe.

Thanks.

Anisio R. Braga
COLTEC/UFMG
***@***.***

Rmano, 2022-05-01:
I have prepared the PR --- do you think that the default shape is ok (look at the inline version) or is it too big?

closed 2022-05-02 by Rmano

anisiobraga, 2022-10-11:
Romano,

I compared the relative sizes with a resistor shape  and as it is the default size matches with {resistors/scale=0.8}. This is ok, but perhaps to scale with the default size of a resistor would facilitate a global scaling. 

Anisio R. Braga
COLTEC/UFMG
***@***.***

\begin{tikzpicture}
  \ctikzset{resistors/scale=0.8}
    \tikzset{small up proxi/.style={proximeter, solid,
            circuitikz/switches/scale=0.707,
            circuitikz/proximeter/hlines thickness=1,
            circuitikz/proximeter/hlines position=0.1}}
    \draw (0,0) to[R] ++(2,0) to[inline proximeter, l=P1] ++(2,0)
        to[ccgsw, name=P2] ++(2,0);
    \draw[red] (2,0) to[R]  ++(2,0);
    \draw[dashed] (P2.mid) -- ++(0,0.5)
        node[small up proxi, above](P2p){Fe}
        (P2p.north) ++ (0,0.5); % extend bounding box
\end{tikzpicture}

Rmano, 2022-10-12:
Hmm, they are a different class, and they are really "auxiliary shapes," so the opinions about the correct size will be personal... so I really prefer to leave it like it is; scaling it with a style definition is easy, and making changes after-release is a bit against my policy.