circuitikz/circuitikz #641: [New Component] Nullor

erwindenboer, 2022-08-11:
Hello, An earlier [issue ](https://github.com/circuitikz/circuitikz/issues/615) requested the implementation of a nullator and a norator. I was wondering if a small enhancement could be made by combining these two components. The nullator and norator together form a so-called nullor. The image is shown below:

**Show the component and its usage**
[image]

source: https://en.wikipedia.org/wiki/Nullor

The component can easily be made by combining the already made nullator and norator. This can ease the design of complex circuits.
It can also serve educational reasons. A nullor is easily translated to an op-amp. To accentuate this. Maybe an option can be implemented to draw a dashed box around this. This highlights the interaction between a nullor and a norator.

erwindenboer, 2022-08-13:
Hello,

I have done some work and put everything in LaTeX. I hope it can be easily implemented into Circuitikz now.

Here is my MWE:

```
\documentclass[a4paper, titlepage]{article}
\usepackage{a4wide} %smaller borders
\usepackage[T1]{fontenc}
\parindent=0pt
\parskip=4pt plus 6pt minus 2pt
\usepackage[siunitx, RPvoltages]{circuitikz}
\makeatletter
%% Test things here
% defines

% components

% paths
\makeatother

\tikzset{noratorW/.style={voosource,bipoles/oosource/circlesize=0.5,
bipoles/oosource/circleoffset=0.5},
nullatorW/.style={esource, sources/scale=0.5}}

\begin{document}
    \begin{circuitikz}
        \draw (0,0) to[short]    ++(-1,0);
        \draw (0,-1.5) to[short]    ++(-1,0);
        \draw (1, 0) to[short]    ++(1,0);
        \draw (1,-1.5) to[short]    ++(1,0);
        \draw (0,0) to[nullator] ++(0,-1.5);
        \draw (1,0) to[noratorW]  ++(0,-1.5);
        \node[rectangle,draw, minimum width=2cm, minimum height=2cm, dashed] (r) at (0.5,-0.75) {};
    \end{circuitikz}
\end{document}
```
This generates the nullor. Of course, by changing the `noratorW` to `norator` the style of the Norator changes.
[image]

Could you implement this in Circuitikz?

Rmano, 2022-08-15:
Sorry for the late answer, I am on vacation right now with a blissfully rare connection to the Internet ;-). 

I'll look at it, but this seems a bit a perfect match for a subcircuit...

erwindenboer, 2022-08-16:
@Rmano No rush, thanks for the update! Enjoy your holiday!

Rmano, 2022-08-21:
I think that a subcircuit is for now a nice solution: 

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\ctikzsubcircuitdef{nullor}{in up, in down, out up, out down}{%
    coordinate(#1-center)
    node[rectangle, draw, dashed, inner sep=1.2cm]{}
    (#1-center) ++(-1.5,1) coordinate(#1-in up)
    -- ++(1,0) to[nullator, name=#1-nullator, v, i, f] ++(0,-2)
    -- ++(-1,0) coordinate(#1-in down)
    (#1-center) ++(1.5,1) coordinate(#1-out up)
    -- ++(-1,0) to[norator, name=#1-norator, v, i ,f] ++(0,-2)
    -- ++(1,0) coordinate(#1-out down)
}
\ctikzsubcircuitactivate{nullor}
\begin{tikzpicture}[]
    \draw (0,0) node[circ]{} \nullor{my1}{in up};
    \draw (my1-out up) -- ++(1,0) \nullor{my2}{in up};
\end{tikzpicture}
\end{document}
```

[image]

Rmano, 2022-08-21:
...although a nicer solution would be to be able to change the components used in a transformer; that would be more general and useful for a generic double bipole. So I will leave this issue open while I am studying it...

PR #650 (2022-09-07): Generic double bipoles
Fixes #641 

[image]

[image]

[image]
commit 15ecfb23 (2022-09-07): Added generic double bipoles

Fixes: #641
closed 2022-09-07 by Rmano