circuitikz/circuitikz #642: [Bug] [Question] Use of \tikzsetnextfilename with circuitikz

MayeulC, 2022-08-17:
**Describe the bug**
I am writing a relatively long manuscript with quite a few circuitikz circuits in it (you'll be able to cherry-pick some as examples, promised!).
In order to speed-up compilation, and generate standalone figures that can be reused by my colleagues, I'd like to use tikzexternalize.

It works with circuitikz, but I couldn't set a filename using the recommended `\tikzsetnextfilename` that sets the filename of the next tikz environement.

I would like to name those files so that they can be [externalized with only named](https://tex.stackexchange.com/a/570721).

**To Reproduce**
Compile the following with `pdflatex -shell-escape`.
See how the `gen-circuit.pdf` only has the red shape instead of the resistor.

```latex
\documentclass{article}
\usepackage{circuitikz}
\usetikzlibrary{external}

\tikzexternalize[prefix=gen-]
\tikzset{external/only named=true}

\begin{document}

Circuitikz: \pgfcircversion{} (\pgfcircversiondate)\par
Ti\emph{k}Z: \pgfversion{} (\pgfversiondate)\par\bigskip

\tikzsetnextfilename{circuit}

\begin{circuitikz}[]
    \draw (0,0) to[R] ++(2,0);
\end{circuitikz}

\tikz[baseline={([yshift=-.8ex]current bounding box.center)}] \node[circle, inner sep=2pt, minimum size=1em, draw=black, color=black, fill=red]{}; % Observe how only this gets saved to a pdf with the filename

\end{document}
```

**Operating system, LaTeX distribution**
```
This is LuaHBTeX, Version 1.15.0 (TeX Live 2022/Arch Linux)
Development id: 7509
```

**Expected behavior**
The resistor should be present in the generated file

**Workaround**
Writing this issue, I discovered I could instead use a `tikzpicture` environment, where the behaviour is the one I expect. I'm not sure what I'll do. Is there a way to use it with the circuitikz environment?

Rmano, 2022-08-19:
I have not a lot of time now, but have you tried to simply change `begin{circuitikz}...\end{circuitikz}` with `\begin{tikzpicture}...\end{tikzpicture}`? They are completely the same, but they can confuse the `externalize` macro (I think it's in the FAQ in the manual somewhere).

Rmano, 2022-08-24:
I will close this now...

closed 2022-08-24 by Rmano