circuitikz/circuitikz #643: [Feature] Make mosfet or bodydiode dashed independently one of another

Revankaiser, 2022-08-20:
In [this](https://tex.stackexchange.com/questions/653348/drawing-mosfet-bodydiode-dashed) question I made it is stated already what I need, and @Rmano told me to post it here, but I will explain it again.

As the title says, It would be nice to be able to make a mosfet which is dashed and the bodydiode solid, or viceversa, as a way to show current path in the circuit. I need to draw stages of a power circuits, which implies multiple identical circuits but with different current paths, and being able to do it as I propose would make it immensely easier. Currently I'm doing it with different components (a mosfet next to a diode) and it takes a lot.

Thank you.

Rmano, 2022-08-21:
As a temporary workaround, you can patch `\drawbodydiode` to add a hook like this:

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\usepackage{etoolbox}
\def\bodydiodehook{\relax}
\tikzset{bodydiode hook/.store in=\bodydiodehook}
\patchcmd{\drawbodydiode}
    {\pgfscope}{\bodydiodehook\pgfscope}
    {}{\fail}
\begin{document}
\begin{tikzpicture}[]
    \draw (0,0) node (mosfet1) [nigfete,anchor=D,bodydiode] {$Q_1$};
    \draw[densely dashed]  (2,0) node (mosfet1) [nigfete,anchor=D,bodydiode] {$Q_2$};
    \draw (0,-2) node (mosfet1) [nigfete,anchor=D,bodydiode, bodydiode hook={\pgfsetdash{{1pt}{1pt}}{0pt}}] {$Q_3$};
    \draw[densely dashed]  (2,-2) node (mosfet1) [nigfete,anchor=D,bodydiode, bodydiode hook={\pgfsetdash{{1pt}{0pt}}{0pt}}] {$Q_4$};
\end{tikzpicture}
\end{document}
```

[image]

Revankaiser, 2022-08-21:
This does the trick thank you

Rmano, 2022-08-22:
Yes, but it is a hack. I'd like to add something more "formal" like the color/thickness/dash pattern I have for the circle...

PR #651 (2022-09-08): Add bodydiode styling
See https://tex.stackexchange.com/questions/653348/drawing-mosfet-bodydiode-dashed
Fixes #643 

[image]
closed 2022-09-08 by Rmano