circuitikz/circuitikz #645: [Feature] option to remove excess wire from the op-amp

erwindenboer, 2022-08-24:
**Is your feature request related to a problem? Please describe.**
Hello, I want to use circuitikz for a converting program. For this, I need to change something in the lay-out of components. The component is the op-amp. For now, when I use the op-amp I automatically get the excess wires from  the output/+ input/ - input.

**Describe the solution you'd like**
Could you add an option which removes the excess wires coming from the output/+ input/ - input of the op-amp?

**Describe alternatives you've considered**
I have considered accounting the excess wires from the op-amp in my converting program. This means that I need to scale the op-amp. That's a shame in my opinion, because the op-amp provided by circuitikz is a nice formatted component.

**Additional context**
op-amp as is currently by the manual:
[image]
The change when the requested feature is activated
[image]

Rmano, 2022-08-25:
Yes, having an option to change the length of the pins would be nice (like in newer components). Unfortunately,  those are quite old ones and it is not easy to patch them, and I have little time now. But I promise I'll look into it. 

For now, you can always use a mux/demux component and create a pinless amplifier (but you need to add the symbols plus and minus "by hand").

Rmano, 2022-08-25:
Anyway, there is a possibility. The pin length is set by the difference between the key `tripoles/op amp/width` (total size, default 1.7) and `tripoles/op amp/port width` (triangle size). So you can do this: 

```latex
\documentclass[border=10pt]{standalone}
\usepackage[siunitx, RPvoltages]{circuitikz}
\begin{document}
\begin{tikzpicture}[]
    \draw (0,0) node[op amp](A){}
        (A.out) node[red,circ]{} (A.+) node[blue, circ]{};
    \ctikzset{tripoles/op amp/port width=1, 
        tripoles/op amp/width=1.19, % 1.7*0.7
    }
    \draw (3,0) node[op amp](A){}
        (A.out) node[red,circ]{} (A.+) node[blue, circ]{};
    % too small!
\end{tikzpicture}
\end{document}
```

[image]

...to have your op-amp without pins (they are still drawn, but well, they have zero-size). Notice that this is a very old part of the package, so I really had to dig it out. You can find those "internal parameters" looking here: https://github.com/circuitikz/circuitikz/blob/dbc62363d185a63a6d09deae034959fea7e59f98/tex/pgfcirctripoles.tex#L5839-L5845

Rmano, 2022-08-25:
...so it would be very difficult to change things. Maybe I can provide a documented key to "squeeze out" all the pins in all the amplifiers together, or simply document the current behavior.

erwindenboer, 2022-08-25:
@Rmano Thanks for your suggestions! I really appreciate your effort to implement this option. For now, the code you have provided is sufficient for me. So please don't rush anything and put your priorities to more important business first ;). In the future, a more slick option can be implemented when the time is right.

closed 2022-08-27 by Rmano